lvgl/lvgl #6319: About the Release Date of New Versions

IAMMX, 2024-06-01:
### Introduce the problem

v9.1 It's been two months since the release. When is v9.2 planned?

kisvegabor, 2024-06-07:
A lot of fixes and features were added indeed, however there are a few open PR that I would like to merge before v9.2. Namely:
- https://github.com/lvgl/lvgl/pull/6333
- https://github.com/lvgl/lvgl/pull/6332
- https://github.com/lvgl/lvgl/pull/6341
- https://github.com/lvgl/lvgl/pull/6068
- https://github.com/lvgl/lvgl/pull/6298

faxe1008, 2024-06-10:
@kisvegabor just to add to the list (and possibly get some reviews :^) ) I also have this PR here for I1 color format support https://github.com/lvgl/lvgl/pull/6345

This would help us use the many intree monochrome displays.

kisvegabor, 2024-07-19:
Getting closer. @liamHowatt made some progress with https://github.com/lvgl/lvgl/pull/6068

IAMMX, 2024-07-20:
> Getting closer. @liamHowatt made some progress with #6068

👍

kisvegabor, 2024-08-15:
Closing in favor of #6570

closed 2024-08-15 by kisvegabor