lvgl/lvgl #6325: ESP32 SPIFFS

gauix4, 2024-06-02:
### Introduce the problem

No documentation or example for using SPIFFS on ESP32 S3

### Proposal

it would be nice to make an example of the use of SPIFFS on ESP32 S3 many people can't do this and there is no documentation on this subject
I can't either
thanks in advance :)

becseya, 2024-06-04:
SPIFF is [compatible with POSIX](https://docs.espressif.com/projects/esp-idf/en/release-v4.2/esp32/api-reference/storage/spiffs.html?highlight=posix#application-example) which is supported in LVGL. But I see value adding an explicit example. What do you think? @kisvegabor 

@gauix4 What would be the use-case? Do you want to store images in the internal flash?

gauix4, 2024-06-04:
@becseya 
Yes, this would be to store images but not in JPEG but rather in binary lvgl or compressed binary, this is much more interesting because the volume is much smaller than JPEG.

Thank you very much for answering anyway ! :)

kisvegabor, 2024-06-05:
It was already asked by many people how to use LVGL's File System on ESP. So an example and a more detailed docs would be really nice.

gauix4, 2024-06-12:
@kisvegabor  hoo yess
I would also like to use it

kisvegabor, 2024-06-14:
See #6367 🙂

kisvegabor, 2024-06-20:
If it's not working by following the instruction in  https://github.com/lvgl/lvgl/pull/6367 please comment in the Pull request.

closed 2024-06-20 by kisvegabor

gauix4, 2024-09-16:
[image]

jgauchia, 2024-09-16:
Hi @gauix4 

I'm using binary images and SPIFFS in my project since LVGL 8.3.

My programming skills are limited but maybe these steps can help you:

-  Check partitions of your board, I suposed that you have a SPIFFS partition with available space

- Create a folder in your project called **data** and put **bin** files on it

[image]

- Upload data to ESP32 file system 
(This metod or **pio run --target uploadfs**)

[image]

- lv_conf.h (using POSIX)

```
/** Setting a default driver letter allows skipping the driver prefix in filepaths. */
#define LV_FS_DEFAULT_DRIVE_LETTER 'F'

/** API for fopen, fread, etc. */
#define LV_USE_FS_STDIO 0
#if LV_USE_FS_STDIO
    #define LV_FS_STDIO_LETTER '\0'     /**< Set an upper cased letter on which the drive will accessible (e.g. 'A') */
    #define LV_FS_STDIO_PATH ""         /**< Set the working directory. File/directory paths will be appended to it. */
    #define LV_FS_STDIO_CACHE_SIZE 0    /**< >0 to cache this number of bytes in lv_fs_read() */
#endif

/** API for open, read, etc. */
#define LV_USE_FS_POSIX 1
#if LV_USE_FS_POSIX
    #define LV_FS_POSIX_LETTER 'F'     /*Set an upper cased letter on which the drive will accessible (e.g. 'A')*/
    #define LV_FS_POSIX_PATH "/spiffs"         /*Set the working directory. File/directory paths will be appended to it.*/
    #define LV_FS_POSIX_CACHE_SIZE 0    /*>0 to cache this number of bytes in lv_fs_read()*/
#endif
```

- Init SPIFFS 

```
  esp_vfs_spiffs_conf_t conf = {
    .base_path = "/spiffs",
    .partition_label = NULL,
    .max_files = 5,
    .format_if_mount_failed = false
  };

  esp_err_t ret = esp_vfs_spiffs_register(&conf);

  if (ret !=ESP_OK)
  {
    if (ret == ESP_FAIL)
      log_e("Failed to mount or format filesystem");
    else if (ret == ESP_ERR_NOT_FOUND)
      log_e("Failed to find SPIFFS partition");
    else
      log_e("Failed to initialize SPIFFS (%s)",esp_err_to_name(ret));
    return ESP_FAIL;
  }

  size_t total = 0, used = 0;
  ret = esp_spiffs_info(NULL, &total, &used);
  if (ret!= ESP_OK)
    log_e("Failed to get SPIFFS partition information (%s)", esp_err_to_name(ret));
  else
    log_i("Partition size: total: %d used: %d", total, used);
  
  return ESP_OK;
```

- Use your bin file in LVGL (example)

```
lv_obj_t *altitImg = lv_img_create(lv_scr_act());
lv_img_set_src(altitImg, "/altit.bin");
```

(Sorry @kisvegabor if I answer this closed Issue)

gauix4, 2024-09-16:
Thank you for your help ! Yes of course I put the binary files in data there is no problem with all that the only problem as I say is really the implementation of the code and the options nothing works! zero documentation nothing and rereading the code is a monumental mess!

[image]

where did you find the minimum documentation to know how to make this library work because frankly I find absolutely nothing and when I find something; since I don't speak English it's only technical words and it doesn't translate them and I don't understand anything frankly the developers really didn't make any effort to detail the functionalities I think they are too caught up in their work and forget that other people will reread the code but given the name of the variables it is very complicated especially when you don't speak English at all

gauix4, 2024-09-17:
@kisvegabor @kisvegabor

I still haven't seen any documentation or examples, it's still a pain, I've been on it for several days and I've been testing things, I still haven't made any progress except being even more annoyed at not finding anything and seeing a lot of people looking too.
sorry if I bet ungrateful that's not the goal it's just that there I'm fed up of not moving forward spending hours to go nowhere each time I think I find something in the end it's absolutely not that I try to reread the code to do tests but nothing conclusive at all

gauix4, 2024-09-17:
[image]

[image]

[image]
[image]
[image]

[image]

I am also unable to activate PNG decoding, it always asks me for a library which cannot be found.