lvgl/lvgl #6334: FPS and Fragmentation

Wlad79, 2024-06-06:
### LVGL version

v8.4

### What happened?

the CPU stabil at 0% FPS remains 33 (Maybe it is frozen) even if LV_DEF_REFR_PERIOD=10 or 1.
Memory Usage 26% and 2 % frag. (can that Fragmentation be complet disabled?)

### How to reproduce?

I have activated system monitor with this in lv_conf.h:
```
/*1: Enable system monitor component*/
#define LV_USE_SYSMON   1
#if LV_USE_SYSMON
    /*Get the idle percentage. E.g. uint32_t my_get_idle(void);*/
    #define LV_SYSMON_GET_IDLE lv_timer_get_idle

    /*1: Show CPU usage and FPS count
     * Requires `LV_USE_SYSMON = 1`*/
    #define LV_USE_PERF_MONITOR 1
    #if LV_USE_PERF_MONITOR
        #define LV_USE_PERF_MONITOR_POS LV_ALIGN_BOTTOM_RIGHT

        /*0: Displays performance data on the screen, 1: Prints performance data using log.*/
        #define LV_USE_PERF_MONITOR_LOG_MODE 0
    #endif

    /*1: Show the used memory and the memory fragmentation
     * Requires `LV_USE_STDLIB_MALLOC = LV_STDLIB_BUILTIN`
     * Requires `LV_USE_SYSMON = 1`*/
    #define LV_USE_MEM_MONITOR 1
    #if LV_USE_MEM_MONITOR
        #define LV_USE_MEM_MONITOR_POS LV_ALIGN_BOTTOM_LEFT
    #endif

#endif /*LV_USE_SYSMON*/
```
set in lv_conf.h LV_DEF_REFR_PERIOD to 10 or 20 or 1.

Wlad79, 2024-06-17:
Sorry for late response but here is it:
https://github.com/lvgl/lvgl/assets/54247746/2eadaf58-f2fc-4f98-a044-80f1bc8a4956

kisvegabor, 2024-06-19:
Very strange! Are you sure that your `lv_conf.h` is used really compiled? What if you add this in it `#error "TEST"`

Wlad79, 2024-06-20:
I get then 145 errors in the project

kisvegabor, 2024-06-20:
Okay, thank! 

Can you debug what is happening [here](https://github.com/lvgl/lvgl/blob/release/v8.4/src/core/lv_refr.c#L379-L416)? Just print a few variables to see where the problem is.

Wlad79, 2024-06-21:
px_num at begining is 0x0005DC00, with fps_limit and fps are 0. second trigger at line 379 has px_num 0xCCC and fps_num and fps have 0 ( screen label shows '?'). next time fps_limit and fps become both 0x21. then px_num will px_num to 0x00004E84 and next time 0 (CPU label shows 100% finuly but '0 % CPU')

kisvegabor, 2024-06-21:
`fps_limit=0x21` means `33`. I wonder why is that if you set `LV_DEF_REFR_PERIOD=10`.

Wlad79, 2024-06-25:
okay if in lv_conf.h will do '#define LV_DISP_DEF_REFR_PERIOD 20', when the label shows "50 FPS". Because of that code:
'''
if(disp_refr->refr_timer) {
            fps_limit = 1000 / disp_refr->refr_timer->period;
}
else {
            fps_limit = 1000 / LV_DISP_DEF_REFR_PERIOD;
}
'''
for first will taken period member value after that will calculate from LV_DISP_DEF_REFR_PERIOD. 

should I close the Request?

Wlad79, 2024-06-25:
okay my mistake. Sorry for inconvinance. I have take lv_conf.h from other branch (not v8.4).

closed 2024-06-25 by Wlad79