lvgl/lvgl #6335: How to use evdev in v9

paulhamsh, 2024-06-06:
### LVGL version

v9.1

### What happened?

Is there any documentation or an example on how to use and configure evdev in v9 (Linux)?

Just adding 'evdev_init();' to working demo code (linux framebuffer port) causes a compilation error.
```
warning: implicit declaration of function ‘evdev_init’ 
```

Are there any examples of working code for v9 anyone can point me to?
Examples I can find seem to refer to .h files I don't have in v9 (#include "lv_drivers/indev/evdev.h") and config files (lv_drivers/lv_drv_conf_templ.h as lv_drv_conf.h)

I am using the Linux framebuffer port and all the demos, installed like this:

```
git clone https://github.com/lvgl/lv_port_linux_frame_buffer.git
cd lv_port_linux_frame_buffer/
git submodule update --init --recursive

mkdir build
cd build 
cmake ..
make -j
```

And ./main works fine.
So then I edited it to try and use evdev.

Updated lv_conf.h to have:

```
#define LV_USE_EVDEV 1
```

And tried this code:
```
#include "lvgl/lvgl.h"
//#include "lvgl/src/drivers/evdev/lv_evdev.h"
//#include "lvgl/src/drivers/lv_drivers.h"
//#include "lvgl/lv_drivers/indev/evdev.h"
#include <unistd.h>
#include <pthread.h>
#include <time.h>

static lv_obj_t *label1;
lv_obj_t *slider1;
lv_obj_t *btn1;

static void btn_event_cb(lv_event_t *e)
{
  lv_event_code_t code = lv_event_get_code(e);
  lv_obj_t *btn = (lv_obj_t *) lv_event_get_target(e);
  if (code == LV_EVENT_CLICKED) {
    static uint8_t cnt = 0;
    cnt++;
    /*Get the first child of the button which is the label and change its text*/
    lv_obj_t *label = lv_obj_get_child(btn, 0);
    lv_label_set_text_fmt(label, "Button: %d", cnt);
  }
}

static void slider_event_cb(lv_event_t *e)
{
  lv_obj_t *slider = (lv_obj_t *) lv_event_get_target(e);
  //lv_obj_t *label = lv_obj_get_child(slider, 0);
  lv_label_set_text_fmt(label1, "%"LV_PRId32, lv_slider_get_value(slider));
  lv_obj_align_to(label1, slider, LV_ALIGN_OUT_TOP_MID, 0, -15);    /*Align top of the slider*/
}

int main(void)
{
    lv_init();

    /*Linux frame buffer device init*/
    lv_display_t * disp = lv_linux_fbdev_create();
    lv_linux_fbdev_set_file(disp, "/dev/fb0");

    evdev_init();

  // Create screen elements

  slider1 = lv_slider_create(lv_screen_active());
  lv_slider_set_value(slider1, 0, LV_ANIM_ON);
  lv_obj_set_pos(slider1, 40, 60);                            /*Set its position*/
  lv_obj_set_size(slider1, 200, 20);                          /*Set its size*/
  lv_obj_add_event_cb(slider1, slider_event_cb, LV_EVENT_VALUE_CHANGED, NULL);

  label1 = lv_label_create(lv_screen_active());
  lv_label_set_text(label1, "0");
  lv_obj_align_to(label1, slider1, LV_ALIGN_OUT_TOP_MID, 0, -15);

  lv_obj_t *label2 = lv_label_create(slider1);                 /*Add a label to the button*/
  lv_label_set_text(label2, "Slider");                         /*Set the labels text*/
  lv_obj_center(label2);

  btn1 = lv_button_create(lv_screen_active());                 /*Add a button the current screen*/
  lv_obj_set_pos(btn1, 40, 120);                               /*Set its position*/
  lv_obj_set_size(btn1, 100, 50);                              /*Set its size*/
  lv_obj_add_event_cb(btn1, btn_event_cb, LV_EVENT_ALL, NULL); /*Assign a callback to the button*/

  lv_obj_t *label3 = lv_label_create(btn1);                    /*Add a label to the button*/
  lv_label_set_text(label3, "Button");                         /*Set the labels text*/
  lv_obj_center(label3);

    /*Handle LVGL tasks*/
    while(1) {
        lv_timer_handler();
        usleep(5000);
    }

    return 0;
}

```

I only added evdev_init(); and it won't compile. I feel I'm missing a .h file from somewhere.

```
lv_port_linux_frame_buffer/main.c:46:5: warning: implicit declaration of function ‘evdev_init’ [-Wimplicit-function-declaration]
   46 |     evdev_init();
      |     ^~~~~~~~~~
[100%] Linking CXX executable /lv_port_linux_frame_buffer/bin/main
/usr/bin/ld: CMakeFiles/main.dir/main.c.o: in function `main':
main.c:(.text+0x1d0): undefined reference to `evdev_init'
```

paulhamsh, 2024-06-06:
There is no evdev_init() any more it seems. Searching the source code I found it only had lv_evdev_create so googled that and found this
 
```
lv_indev_t *mouse = lv_evdev_create(LV_INDEV_TYPE_POINTER, "/dev/input/event0");
lv_indev_set_display(mouse, cursor_disp);
```

closed 2024-06-06 by paulhamsh

kisvegabor, 2024-06-10:
We really should have at least a brief docs about it [here](https://docs.lvgl.io/master/integration/driver/touchpad/index.html). If you had some time to add it, it would be highly appreciated.

paulhamsh, 2024-06-10:
I would be happy to - how do I add to the documentation?

kisvegabor, 2024-06-11:
Thank you! Just create a new file [here](https://github.com/lvgl/lvgl/tree/master/docs/integration/driver/touchpad) (similar to the others) and add an entry in [index.rst](https://github.com/lvgl/lvgl/blob/master/docs/integration/driver/touchpad/index.rst).

paulhamsh, 2024-06-11:
Do I need to clone the repo and then make a push request using git?

kisvegabor, 2024-06-13:
Yes, please. In case you needed it, [here](https://docs.lvgl.io/master/CONTRIBUTING.html#pull-request) you can find a guide about pull requests.

paulhamsh, 2024-06-14:
Ok - submitted my first ever PR! Let me know if it is correct