lvgl/lvgl #6337: Adding border to different sides

fatihozen3, 2024-06-06:
### Introduce the problem

I want to add border to my images. But side does not effect the border, what am I missing ? I expect the line on the images' related side but I see all line on right and bottom...

`    //Write codes img_rect
    img_rect = lv_img_create(screen);
    lv_img_set_src(img_rect, "F:/rectangle.bin");
    lv_img_set_pivot(img_rect, 50, 50);
    lv_img_set_angle(img_rect, 0);
    lv_obj_set_pos(img_rect, 16, 96);
    lv_obj_set_size(img_rect, 180, 140);

    lv_obj_set_style_border_color(img_rect, lv_color_hex(0xffffff), LV_PART_MAIN | LV_STATE_DEFAULT);
    lv_obj_set_style_border_opa(img_rect, 255, LV_PART_MAIN | LV_STATE_DEFAULT);
    lv_obj_set_style_border_width(img_rect, 180, LV_PART_MAIN | LV_STATE_DEFAULT);
    lv_obj_set_style_border_side(img_rect, LV_BORDER_SIDE_LEFT, LV_PART_MAIN | LV_STATE_DEFAULT);`
    
    // another object creation
    //border for this object
    lv_obj_set_style_border_color(img_rect, lv_color_hex(0xffffff), LV_PART_MAIN | LV_STATE_DEFAULT);
    lv_obj_set_style_border_opa(img_rect, 255, LV_PART_MAIN | LV_STATE_DEFAULT);
    lv_obj_set_style_border_width(img_rect, 270, LV_PART_MAIN | LV_STATE_DEFAULT);
    lv_obj_set_style_border_side(img_rect, LV_BORDER_SIDE_RIGHT, LV_PART_MAIN | LV_STATE_DEFAULT);`
    
    // another object creation
    //border for this object
    lv_obj_set_style_border_color(img_rect, lv_color_hex(0xffffff), LV_PART_MAIN | LV_STATE_DEFAULT);
    lv_obj_set_style_border_opa(img_rect, 255, LV_PART_MAIN | LV_STATE_DEFAULT);
    lv_obj_set_style_border_width(img_rect, 138, LV_PART_MAIN | LV_STATE_DEFAULT);
    lv_obj_set_style_border_side(img_rect, LV_BORDER_SIDE_TOP, LV_PART_MAIN | LV_STATE_DEFAULT);`

    // another object creation
    //border for this object
    lv_obj_set_style_border_color(img_rect, lv_color_hex(0xffffff), LV_PART_MAIN | LV_STATE_DEFAULT);
    lv_obj_set_style_border_opa(img_rect, 255, LV_PART_MAIN | LV_STATE_DEFAULT);
    lv_obj_set_style_border_width(img_rect, 360, LV_PART_MAIN | LV_STATE_DEFAULT);
    lv_obj_set_style_border_side(img_rect, LV_BORDER_SIDE_BOTTOM, LV_PART_MAIN | LV_STATE_DEFAULT);`    
     
    // another object creation
    //border for this object
    lv_obj_set_style_border_color(img_rect, lv_color_hex(0xffffff), LV_PART_MAIN | LV_STATE_DEFAULT);
    lv_obj_set_style_border_opa(img_rect, 255, LV_PART_MAIN | LV_STATE_DEFAULT);
    lv_obj_set_style_border_width(img_rect, 120, LV_PART_MAIN | LV_STATE_DEFAULT);
    lv_obj_set_style_border_side(img_rect, LV_BORDER_SIDE_FULL, LV_PART_MAIN | LV_STATE_DEFAULT);`
            
     // another object creation
    //border for this object
    lv_obj_set_style_border_color(img_rect, lv_color_hex(0xffffff), LV_PART_MAIN | LV_STATE_DEFAULT);
    lv_obj_set_style_border_opa(img_rect, 255, LV_PART_MAIN | LV_STATE_DEFAULT);
    lv_obj_set_style_border_width(img_rect, 240, LV_PART_MAIN | LV_STATE_DEFAULT);
    lv_obj_set_style_border_side(img_rect, LV_BORDER_SIDE_INTERNAL, LV_PART_MAIN | LV_STATE_DEFAULT);`

liamHowatt, 2024-06-08:
I got the borders to work by removing `lv_obj_set_size`. Use `lv_image_set_scale` instead. However, the borders will act like the image hasn't been scaled.

It seems like any use of `lv_obj_set_size` on an image ruins the borders, even if the size is being set to the default image size.

kisvegabor, 2024-06-12:
It's working for me with this code:
[image]

```c
  lv_obj_t * img_rect = lv_img_create(lv_screen_active());
  lv_img_set_src(img_rect, &img_benchmark_cogwheel_argb);
  lv_img_set_pivot(img_rect, 50, 50);
  lv_img_set_angle(img_rect, 0);
  lv_obj_set_pos(img_rect, 16, 96);
  lv_obj_set_size(img_rect, 180, 140);

  //Add a background to better see the bounding box
  lv_obj_set_style_bg_opa(img_rect, LV_OPA_50, LV_PART_MAIN | LV_STATE_DEFAULT);
  lv_obj_set_style_bg_color(img_rect, lv_color_hex(0xff0000), LV_PART_MAIN | LV_STATE_DEFAULT);

  lv_obj_set_style_border_color(img_rect, lv_color_hex(0x0000ff), LV_PART_MAIN | LV_STATE_DEFAULT);
  lv_obj_set_style_border_opa(img_rect, 255, LV_PART_MAIN | LV_STATE_DEFAULT);
  lv_obj_set_style_border_width(img_rect, 10, LV_PART_MAIN | LV_STATE_DEFAULT);  //Use a smaller border width
  lv_obj_set_style_border_side(img_rect, LV_BORDER_SIDE_RIGHT, LV_PART_MAIN | LV_STATE_DEFAULT);
```
In your code you have used > 100 px border with. Is it intentional?

closed 2024-07-08 by lvgl-bot