lvgl/lvgl #6347: X11 driver not correctly working (or I am doing it wrong)

mbalmer, 2024-06-09:
### LVGL version

v9.1.0

### What happened?

The X11 driver used for standalone desktop mode does not work correctly (or I am doing it wrong). Depending on the initial resolution, it does either work, segfault, or the function does not even return:

lvgl is initialized as follows:

```c
lv_init();
lv_display_t *disp = lv_x11_window_create("myname", width, height);
lv_x11_inputs_create(disp, NULL);
```

Depending on the resolution in `width` and `height` different things happen:

* 100x64: works as expected
* 256x347: segfaults later in program when calling lv_obj_align()
* 800x600: lv_x11_window_create() does not return

Here ist the code to draw the screen:

```c
lv_obj_set_style_bg_color(lv_screen_active(), lv_color_hex(0x003a57), LV_PART_MAIN);
lv_obj_t *label = lv_label_create(lv_screen_active());
lv_label_set_text(label, "Hello world");
lv_obj_set_style_text_color(lv_screen_active(), lv_color_hex(0xffffff), LV_PART_MAIN);
lv_obj_align(label, LV_ALIGN_CENTER, 0, 0);
```

### How to reproduce?

```c
lv_init();
lv_display_t *disp = lv_x11_window_create("myname", width, height);
lv_x11_inputs_create(disp, NULL);
lv_obj_set_style_bg_color(lv_screen_active(), lv_color_hex(0x003a57), LV_PART_MAIN);
lv_obj_t *label = lv_label_create(lv_screen_active());
lv_label_set_text(label, "Hello world");
lv_obj_set_style_text_color(lv_screen_active(), lv_color_hex(0xffffff), LV_PART_MAIN);
lv_obj_align(label, LV_ALIGN_CENTER, 0, 0);
```

mbalmer, 2024-06-09:
When I set LV_USE_STDLIB_MALLOC to LV_STDLIB_CLIB instead of LV_STDLIB_BULTIN everything works.  So could it be memory fragmentation?  At one point it tries to allocate 92000 bytes in lv_mem.c, which fails when using LV_STDLIB_BUILTIN for LV_USE_STDLIB_MALLOC

kisvegabor, 2024-06-12:
I think [here](https://github.com/lvgl/lvgl/blob/master/src/drivers/x11/lv_x11_display.c#L381-L382) we should always use normal `malloc`. That's what we are doing in the [SDL driver](https://github.com/lvgl/lvgl/blob/master/src/drivers/sdl/lv_sdl_window.c#L400) too.

mbalmer, 2024-06-12:
Wirks like a charm we standard malloc. Guess we can close this.

kisvegabor, 2024-06-17:
Great! I've fixed it here #6384.

PR #6384 (2024-06-26): fix(x11): use normal malloc for frame buffer allocation
fixes #6347

### Description of the feature or fix

A clear and concise description of what the bug or new feature is.

### Notes
- Update the [Documentation](https://github.com/lvgl/lvgl/tree/master/docs) if needed.
- Add [Examples](https://github.com/lvgl/lvgl/tree/master/examples) if relevant.
- Add [Tests](https://github.com/lvgl/lvgl/blob/master/tests/README.md) if applicable.
- If you added new options to `lv_conf_template.h` run [lv_conf_internal_gen.py](https://github.com/lvgl/lvgl/blob/master/scripts/lv_conf_internal_gen.py) and  [... cut]
closed 2024-06-26 by FASTSHIFT