lvgl/lvgl #6348: Funky code in the indev pointer driver for handling rotation

kdschlosser, 2024-06-09:
### Introduce the problem

```c
if(disp->rotation == LV_DISPLAY_ROTATION_180 || disp->rotation == LV_DISPLAY_ROTATION_270) {
    data->point.x = disp->hor_res - data->point.x - 1;
    data->point.y = disp->ver_res - data->point.y - 1;    // 270° rotation doesn't need to have the y axis flipped

if(disp->rotation == LV_DISPLAY_ROTATION_90 || disp->rotation == LV_DISPLAY_ROTATION_270) {
    int32_t tmp = data->point.y;
    data->point.y = data->point.x;
    data->point.x = disp->ver_res - tmp - 1;     // 270° rotation has the y axis flipped back to the way it was
```

### Proposal

There is additional work that is being done that doesn't need to be done when the rotation is 270°. This is a waste of resources in my opinion. It also messes with the readability of the code and makes it a little bit clouded as to what is going on.

I think that something like the code below is clearer to understand what is going on and it eliminates the extra work that is being done. 

```c
if (disp->rotation == LV_DISPLAY_ROTATION_90 || disp->rotation == LV_DISPLAY_ROTATION_180) {
    data->point.y = disp->ver_res - data->point.y - 1;
}

if (disp->rotation == LV_DISPLAY_ROTATION_180 || disp->rotation == LV_DISPLAY_ROTATION_270) {
    data->point.x = disp->hor_res - data->point.x - 1;
}

if (disp->rotation == LV_DISPLAY_ROTATION_90 || disp->rotation == LV_DISPLAY_ROTATION_270) {
    int32_t tmp = data->point.y;
    data->point.y = data->point.x;
    data->point.x = tmp;
}
```

You can easily understand that the y axis needs to be flipped for 90 and 180 degree rotations and the x axis needs to be flipped for 180 and 270 degree rotations and you know the x and y axis need to be swapped for 90 and 270 degree rotations.

 90  = y axis flipped, x and y axis swapped
180 = x axis flipped and y axis flipped
270 = x axis flipped, x and y axis swapped

using a switch/case for each rotation is by far the easiest to read but I believe they are more expensive than condition evaluation. How much more IDK. Would it be worth the expense gain the readability? Maybe someone with a higher understanding of C can comment on this.

faxe1008, 2024-06-09:
I think there is also an off-by-one error in there [https://github.com/lvgl/lvgl/issues/6312](https://github.com/lvgl/lvgl/issues/6312)

kisvegabor, 2024-06-10:
For me both have (the current and the proposed) have the same complexity. Anyway, I'm ok with merging an update for this.

kdschlosser, 2024-06-10:
The way it is does work correctly.\ It's not that it's broken. You have the y axis being flipped one way and then back which is rather strange to do. I was having an issue with the rotation of the indev and it turned out to be an issue with the display and not the touch panel and the MADCTL but when I saw this code I had to actually write it all out to know what it was doing. 

I personally think that use switch/case is the best way to go and have each angle handled in it's own case. That would be by far the easiest to read but it would be a larger code footprint.

When I opened this topic I wasn't sure if there was an issue or not and it turned out there isn't an issue with it. I think the same thing happened with @faxe1008 when he/she looked at it for some reason it just doesn't look right. However it is in fact correct and it does work the way it should.

kdschlosser, 2024-06-10:
The code does look odd that's for sure. It's been in the LVGL core for a long while now and has not had any issues. If it's not broke don't fix it. Maybe some comments should be added to explain what is going on. That would be a good thing to do and it wouldn't change anything.

kisvegabor, 2024-06-11:
Adding comments is safer for sure. If you think it's worth the time feel free to open a PR. :slightly_smiling_face:

kdschlosser, 2024-06-11:
It's funny that there are 2 people at the exact same time that sees code as strange/cryptic. At least I am not the only one, LOL.
How does the saying go?? One person, it could be a user issue. More than one person, it's a code issue.

I found it hard to read is all, not that it was wrong.

it can't hurt anything to add the comments. I will add the comments and submit a PR.

closed 2024-07-03 by lvgl-bot