lvgl/lvgl #6350: Online Font Converter fonts not displaying correctly

GSGDaren, 2024-06-09:
### LVGL version

8.3

### What happened?

Using the recently updated online fort converter to create fonts is producing output that does not render.  Font files created by the previous tool work fine in the same project.  Tried multiple different source fonts, sizes, and settings.  Pouring through the output files everything appears the same except the .line_height and the .base_line ranges.

For fonts that work, we get output in ranges like:

lv_font_montserrat_20.c
```
    .line_height = 22,          /*The maximum line height required by the font*/
    .base_line = 4,             /*Baseline measured from the bottom of the line*/
```
lv_font_inter_bold_22.c (custom-created font using the older tool that works fine in the project)
```
    .line_height = 26,          /*The maximum line height required by the font*/
    .base_line = 5,             /*Baseline measured from the bottom of the line*/
```
etc.

These all have relatively low base_line values and increase to approximately the pixel height.

New fonts, that don't render, have ranges like this:

lv_font_inter_semi_28.c
```
    .line_height = 109,          /*The maximum line height required by the font*/
    .base_line = 80,             /*Baseline measured from the bottom of the line*/
```

lv_font_nunito_bold_28.c
```
    .line_height = 93,          /*The maximum line height required by the font*/
    .base_line = 64,             /*Baseline measured from the bottom of the line*/
```

lv_font_internew_bold_22.c - A remake of the inter22 font created with the old tool.
```
    .line_height = 84,          /*The maximum line height required by the font*/
    .base_line = 62,             /*Baseline measured from the bottom of the line*/
```

The absolute difference is similar (approx. font size), but the starting values are very different.

Not sure if this is the cause of the fonts not rendering, but it's the only discernable difference I see in the produced files.

### How to reproduce?

Use the LVGL online font converter

AlexanderMakhov, 2024-06-10:
I also faced the problem with new font converter. I noticed that parameter --lcd-v added to call options in the header of each converted font file. I didn't find any way to omit this parameter online so I had to install font converter locally and call it without --lcd-v. And it's work fine - just like before. I'm not very good in lvgl or fonts and maybe it's not what you are writing about but hope it'll help you somehow

jcontrerasf, 2024-06-11:
I'm facing the same problem. It appears to be related to the subpixel option because changing `subpx` to `.subpx = LV_FONT_SUBPX_NONE` allowed the font to render, but it appears very stretched.

CasualXiao, 2024-06-11:
I also encountered the same problem, the new version of the font converter generated more font data than the original one. When 9.0 was first released, the font conversion interface redirected to version v8, but now it is no longer available

CasualXiao, 2024-06-11:
> 我也遇到了同样的问题，新版本的字体转换器生成的字体数据比原来的要多。9.0 首次发布时，字体转换界面重定向到 v8 版本，但现在不再可用

I also encountered the same problem, the new version of the font converter generated more font data than the original one. When 9.0 was first released, the font conversion interface could redirect to v8 version, but it is no longer available now

GSGDaren, 2024-06-11:
There is definitely more data per glyph with the new tool than the online version that was made for 8.x

Running the font converter on my local machine produces per-glyph data more closely in size to the old online version 8 converter, but still no rendering.

Here is the public general font descriptor of the same source font

This one **works** and was created with the old online tool:

```
/*Initialize a public general font descriptor*/
#if LVGL_VERSION_MAJOR >= 8
const lv_font_t lv_font_inter_bold_22 = {
#else
lv_font_t lv_font_inter_bold_22 = {
#endif
    .get_glyph_dsc = lv_font_get_glyph_dsc_fmt_txt,    /*Function pointer to get glyph's data*/
    .get_glyph_bitmap = lv_font_get_bitmap_fmt_txt,    /*Function pointer to get glyph's bitmap*/
    .line_height = 26,          /*The maximum line height required by the font*/
    .base_line = 5,             /*Baseline measured from the bottom of the line*/
#if !(LVGL_VERSION_MAJOR == 6 && LVGL_VERSION_MINOR == 0)
    .subpx = LV_FONT_SUBPX_NONE,
#endif
#if LV_VERSION_CHECK(7, 4, 0) || LVGL_VERSION_MAJOR >= 8
    .underline_position = -4,
    .underline_thickness = 2,
#endif
    .dsc = &font_dsc,          /*The custom font data. Will be accessed by `get_glyph_bitmap/dsc` */
#if LV_VERSION_CHECK(8, 2, 0) || LVGL_VERSION_MAJOR >= 9
    .fallback = NULL,
#endif
    .user_data = NULL,
};
```

This one is the generated descriptor from the new online version that **does not work**:

```
/*Initialize a public general font descriptor*/
#if LVGL_VERSION_MAJOR >= 8
const lv_font_t lv_font_inter2_bold_22 = {
#else
lv_font_t lv_font_inter2_bold_22 = {
#endif
    .get_glyph_dsc = lv_font_get_glyph_dsc_fmt_txt,    /*Function pointer to get glyph's data*/
    .get_glyph_bitmap = lv_font_get_bitmap_fmt_txt,    /*Function pointer to get glyph's bitmap*/
    .line_height = 84,          /*The maximum line height required by the font*/
    .base_line = 62,             /*Baseline measured from the bottom of the line*/
#if !(LVGL_VERSION_MAJOR == 6 && LVGL_VERSION_MINOR == 0)
    .subpx = LV_FONT_SUBPX_VER,
#endif
#if LV_VERSION_CHECK(7, 4, 0) || LVGL_VERSION_MAJOR >= 8
    .underline_position = -4,
    .underline_thickness = 2,
#endif
    .dsc = &font_dsc,          /*The custom font data. Will be accessed by `get_glyph_bitmap/dsc` */
#if LV_VERSION_CHECK(8, 2, 0) || LVGL_VERSION_MAJOR >= 9
    .fallback = &lv_font_montserrat_22,
#endif
    .user_data = NULL,
};
```

And here is a converted font using the local tool for the same font - **doesn't work**:

```
/*Initialize a public general font descriptor*/
#if LV_VERSION_CHECK(8, 0, 0)
const lv_font_t lv_font_intr_bold_22 = {
#else
lv_font_t lv_font_intr_bold_22 = {
#endif
    .get_glyph_dsc = lv_font_get_glyph_dsc_fmt_txt,    /*Function pointer to get glyph's data*/
    .get_glyph_bitmap = lv_font_get_bitmap_fmt_txt,    /*Function pointer to get glyph's bitmap*/
    .line_height = 26,          /*The maximum line height required by the font*/
    .base_line = 5,             /*Baseline measured from the bottom of the line*/
#if !(LVGL_VERSION_MAJOR == 6 && LVGL_VERSION_MINOR == 0)
    .subpx = LV_FONT_SUBPX_NONE,
#endif
#if LV_VERSION_CHECK(7, 4, 0) || LVGL_VERSION_MAJOR >= 8
    .underline_position = -4,
    .underline_thickness = 2,
#endif
    .dsc = &font_dsc           /*The custom font data. Will be accessed by `get_glyph_bitmap/dsc` */
};
```

CasualXiao, 2024-06-11:
I tried using lv'font_conv for font creation and it was able to display normally,

GSGDaren, 2024-06-11:
Interestingly enough I was looking through an old post about LVGL fonts ([https://forum.lvgl.io/t/problems-using-lv-font-conv-conversion/4296](https://forum.lvgl.io/t/problems-using-lv-font-conv-conversion/4296)) and saw there were rending issues depending on the bpp setting.  Just now recreated a font using lv_font_conv with a bpp setting of 1 and the font renders.   Fonts are a blackhole.

kisvegabor, 2024-06-11:
You are right. `.subpx = LV_FONT_SUBPX_VER,` should be `.subpx = LV_FONT_SUBPX_NONE`.  And it's really caused by `--lcd-v` which shouldn't be passed by default. 

cc @richardgazdik

richardgazdik, 2024-06-11:
Just removed the `lcd_v` parameter from the form. Let's see if that solves this issue.

GSGDaren, 2024-06-11:
Just produced a 2bpp font using the online converter (sans lcd_v) and it works like a charm!

DrNeuroSurg, 2024-06-12:
Now, there's a new error ?

What's wrong ?

[image]

CasualXiao, 2024-06-12:
[image]

Using ttf text format can generate fonts

DrNeuroSurg, 2024-06-12:
Ok - works now (had to empty browser cache ..) !!!

kisvegabor, 2024-06-13:
Cool, closing the issue then. :slightly_smiling_face:

closed 2024-06-28 by kisvegabor