lvgl/lvgl #6351: simple container of radius one shows mixed color

jihunluv, 2024-06-10:
### LVGL version

lvgl 8.3.11

### What happened?

closned lv_port_pc_vscode with 8.3.11 tag

Created simple testcase where the container has radius 1,
[image]

Expected result is the container's color is white. Instead, the container color is pink (like mixed with background)
When the radius is 2, the color of container is white.

### How to reproduce?

static void user_image_demo1()
{
  lv_obj_t *scr = lv_obj_create(NULL);
  lv_obj_set_size(scr, LV_PCT(100), LV_PCT(100));
  lv_obj_set_pos(scr, 0, 0);
  lv_obj_set_style_bg_color(scr, lv_color_hex(0xFF0000), LV_PART_MAIN|LV_STATE_DEFAULT);
  lv_obj_t *container = lv_obj_create(scr);
  lv_obj_set_style_radius(container, 1, LV_PART_MAIN);
  lv_obj_set_style_border_width(container, 0, LV_PART_MAIN);
  lv_obj_set_size(container, 491, 271);
  lv_obj_set_pos(container, 19, 20);
  lv_scr_load(scr);
}

kisvegabor, 2024-06-11:
I've tested in the Eclipse simulator and it works for me with v8.3 and v8.4 too. 

If you are still at the beginning of your project can you upgrade to LVGL v9?

jihunluv, 2024-06-17:
Sorry about incomplete bug report.
I've tested this with v9 already and it seems OK with that version.
But with V8.3.11 tag and below is the configuration I've used.
Makefile

-LV_DRIVER          := X11
-#LV_DRIVER          := SDL2
+#LV_DRIVER          := X11
+LV_DRIVER          := SDL2

-LV_DRIVER_USE       := USE_SDL
+LV_DRIVER_USE       := USE_SDL_GPU

It seems the problem is related to SDL_GPU configuration.

kisvegabor, 2024-06-19:
Happy to here that it's working in v9. Can you continue your project with v9?

The SDL render implementation of v8 is quite complex and I won' be able to look into it in the ear future. :slightly_frowning_face:

jihunluv, 2024-06-20:
Thank you for the feedback, V9 is not a viable option for me because of  migration issues. Perhaps we can withhold this issue for sometime.

closed 2024-06-20 by jihunluv