lvgl/lvgl #6357: Tabview scrolling issue

Goran58, 2024-06-12:
### LVGL version

v9.1

### What happened?

If you continue to scroll left when you have reached the last tab, you jump back to the second-to-last tab. Scrolling from the last to the second-to-last tab is always done without animation. This worked until version 9.0. 

### How to reproduce?

The error can be easily reproduced on, for example, LVGL's demo page.

kisvegabor, 2024-06-12:
Hi,

On my side it's working well with `master`. Could you also try with `master`?
[image]

Goran58, 2024-06-12:
Hi,
Sorry that I was a bit unclear, but by the demo page I actually meant the widget examples, see below. And there it also goes wrong for master but not for version <= 9. Possibly, the error depends on which input device is used. I am using evdev on a Linux computer.
[image]

Goran58, 2024-06-14:
Have now tried with master. Same problem, see below.

https://github.com/lvgl/lvgl/assets/6393885/bb33d861-55c1-455d-ad74-b6c4635a060c

faruq-zhao, 2024-06-14:
Hi.
I successfully reproduced this error in the `master` branch.
Please see the video below for specific performance:
[Screencast from 2024-06-15 02-46-21.webm](https://github.com/lvgl/lvgl/assets/49693069/9b58bf7a-dd8f-4dc8-a415-46df4d3b37fa)

This error seems to be introduced by PR #6230.
https://github.com/kisvegabor/lvgl_upstream/blob/269dd90211225fca8e56311d3120acf1ba51f0b3/src/indev/lv_indev_scroll.c#L683
When you continue to slide on the last tab,`no_more_end_snap` is always false.
@kisvegabor Do you have time to check it out? Thank you.

niklasf, 2024-06-17:
The same patch is also breaking scrolling on the last tile of tile views: https://docs.lvgl.io/master/examples.html#tileview

kisvegabor, 2024-06-19:
I've just fixed it in #6393.
Please test it.

Goran58, 2024-06-20:
The fix in #6393 seems to work, at least in the tabview case. 
Thanks a lot.

PR #6393 (2024-07-04): fix(scroll): fix jumping on scroll end
fixes #6357

### Description of the feature or fix

A clear and concise description of what the bug or new feature is.

### Notes
- Update the [Documentation](https://github.com/lvgl/lvgl/tree/master/docs) if needed.
- Add [Examples](https://github.com/lvgl/lvgl/tree/master/examples) if relevant.
- Add [Tests](https://github.com/lvgl/lvgl/blob/master/tests/README.md) if applicable.
- If you added new options to `lv_conf_template.h` run [lv_conf_internal_gen.py](https://github.com/lvgl/lvgl/blob/master/scripts/lv_conf_internal_gen.py) and update [Kconfig]( [... cut]
closed 2024-07-04 by FASTSHIFT