lvgl/lvgl #6360: The calendar component date in v8.3 version can only be displayed up to 2023.

CasualXiao, 2024-06-12:
### LVGL version

v8.3.11

### What happened?

I found that the date can only be displayed up to 2023 when using the calendar component in v8.3 version.
I use this function lv_calendar_set_showed_date (calendar, 2024, 6); It will be displayed directly from 1901. The year dropdown list can only display up to 2023
LV_USE_CALENDAR_HEADER_DROPDOWN =1

### How to reproduce?

lv_obj_clean(lv_scr_act()); // 清屏

    lv_obj_t *scr = lv_scr_act();
    lv_obj_set_size(scr, 320, 240);

    lv_obj_t *calendar = lv_calendar_create(scr);
    lv_obj_set_size(calendar, 220, 220);
    lv_obj_align(calendar, LV_ALIGN_CENTER, 0, 0);
    lv_obj_add_event_cb(calendar, event_handler, LV_EVENT_ALL, NULL);

    lv_calendar_set_today_date(calendar, 2023, 06, 23);
    // lv_calendar_set_showed_date(calendar, 2021, 02);

    /*Highlight a few days*/
    static lv_calendar_date_t highlighted_days[3]; /*Only its pointer will be saved so should be static*/
    highlighted_days[0].year = 2023;
    highlighted_days[0].month = 01;
    highlighted_days[0].day = 01;

    highlighted_days[1].year = 2023;
    highlighted_days[1].month = 06;
    highlighted_days[1].day = 11;

    highlighted_days[2].year = 2023;
    highlighted_days[2].month = 06;
    highlighted_days[2].day = 30;

    lv_calendar_set_highlighted_dates(calendar, highlighted_days, 3);

#if LV_USE_CALENDAR_HEADER_DROPDOWN
    lv_calendar_header_dropdown_create(calendar);
#elif LV_USE_CALENDAR_HEADER_ARROW
    lv_calendar_header_arrow_create(calendar);
#endif
    lv_calendar_set_showed_date(calendar, 2024, 6);

C47D, 2024-06-12:
That's supposed to be fixed, let me check that version received the fix.

CasualXiao, 2024-06-13:
[image]
[image]

I added the year to 2025 and changed the max year to 2025 to display it normally,

C47D, 2024-06-13:
Yep, that was the initial "fix" we did, later on we decided to enable the user to add a custom year list.

C47D, 2024-07-01:
Any comments on this @kisvegabor?

kisvegabor, 2024-07-09:
[Custom year list](https://github.com/lvgl/lvgl/blob/release/v8.4/src/extra/widgets/calendar/lv_calendar_header_dropdown.h#L39-L46) is the recommended way to instead of editing LVGL's source code (which makes updating harder).

@CasualXiao could you try it out?

CasualXiao, 2024-07-09:
> [Custom year list](https://github.com/lvgl/lvgl/blob/release/v8.4/src/extra/widgets/calendar/lv_calendar_header_dropdown.h#L39-L46) is the recommended way to instead of editing LVGL's source code (which makes updating harder).
> 
> @CasualXiao could you try it out?

Okay, thank you. I saw them in 8.4。

kisvegabor, 2024-07-10:
As the issue should be fixed I close it.

If it's still not working as expected please comment here and reopen the issue.

closed 2024-07-10 by kisvegabor