lvgl/lvgl #6365: Make the NuttX drivers even simpler

kisvegabor, 2024-06-12:
### Introduce the problem

I'm preparing to the NuttX conference on Friday. LVGL's NuttX integration is quite complete, except that the users need to create an `uvloop` themselves. 

### Proposal

I wonder if [this](https://github.com/apache/nuttx-apps/blob/master/examples/lvgldemo/lvgldemo.c#L69C1-L87C2) can be part of `lv_nuttx_libuv.c`. If the user wants something custom, he can just write use it's own implementation and simply not call the built-in `lv_nuttx_uv_loop`.

kisvegabor, 2024-06-12:
cc @XuNeo

XuNeo, 2024-06-13:
I see no problem to move to `lv_nuttx_libuv.c`.

This part can also have a dedicated function like `lv_nuttx_run()`

```c
#ifdef CONFIG_LV_USE_NUTTX_LIBUV
  lv_nuttx_uv_loop(&ui_loop, &result);
#else
  while (1)
    {
      uint32_t idle;
      idle = lv_timer_handler();

      /* Minimum sleep of 1ms */

      idle = idle ? idle : 1;
      usleep(idle * 1000);
    }
#endif
```

liamHowatt, 2024-06-14:
> users need to create an `uvloop` themselves.

I think you already know, but `libuv` integration is optional. It just [uses the normal sleep loop if it's disabled](https://github.com/apache/nuttx-apps/blob/00f98947786892eaabf60b2e61fad553aee1c36c/examples/lvgldemo/lvgldemo.c#L148-L161). It's [not enabled by default in the sim](https://github.com/apache/nuttx/blob/dc651e090eb366f9abb18c48be1dbb92b7084bdc/boards/sim/sim/sim/configs/lvgl_fb/defconfig#L38-L39).

liamHowatt, 2024-06-30:
Not stale.

PR #6371 (2024-07-04): feat(nuttx): add lv_nuttx_run
### Description of the feature or fix

Fixes #6365

I don't know why but after updating LVGL to the latest master with `sim:lvgl_fb` `make` fails:
```
LD:  nuttx
/usr/bin/ld: nuttx.rel: in function `nx_start_application':
/home/kisvegabor/projects/lvgl/nuttx-workspace/nuttx/sched/init/nx_bringup.c:348: undefined reference to `lvgldemo_main'
/usr/bin/ld: sim_x11eventloop.o: in function `sim_x11events':
/home/kisvegabor/projects/lvgl/nuttx-workspace/nuttx/arch/sim/src/sim/posix/sim_x11eventloop.c:139: undefined reference to `sim_buttonevent'
/usr/bin/ld: /home/kisve [... cut]
closed 2024-07-04 by kisvegabor