lvgl/lvgl #6366: lv_image_set_scale and LV_IMAGE_ALIGN_TILE cannot be effective at the same time

C-Ackerman, 2024-06-13:
### LVGL version

v9.0.0

### What happened?

lv_image_set_scale(bgimg, LV_SCALE_NONE * 0.5);and lv_image_set_align(bgimg, LV_IMAGE_ALIGN_TILE);cannot be effective at the same time。

[image]

### How to reproduce?

`
void lv_example_style_6(void)
{

    lv_obj_t * native = lv_obj_create(lv_screen_active());
    lv_obj_set_size(native, 300, 300);
    static lv_style_t style;
    lv_style_init(&style);
    lv_style_set_border_width(&style, 10);
    lv_style_set_border_color(&style, lv_palette_lighten(LV_PALETTE_BLUE, 3));
    lv_obj_add_style(native, &style, 0);

    lv_obj_t * bgimg = lv_image_create(native);
    lv_image_set_src(bgimg, "/data/app/logo.png");
    lv_obj_set_size(bgimg, 300, 300);
    lv_image_set_scale(bgimg, LV_SCALE_NONE * 0.5);
    lv_image_set_align(bgimg, LV_IMAGE_ALIGN_TILE);
}

`

kisvegabor, 2024-06-17:
The expected behavior, so I haven't implemented this case:
- With 50% scale make all the images smaller and have twice as much tiles?
- If so what to do if the images are rotated? Shall we place the tiles on bounding box of the rotated image or use the original image's bounding box?

C-Ackerman, 2024-06-18:
With 50% scale make all the images smaller and have twice as much tiles?
：yes，This problem is to implement background-size and background-repeat.
Reference https://github.com/lvgl/lvgl/issues/4733
If so what to do if the images are rotated? Shall we place the tiles on bounding box of the rotated image or use the original image's bounding box?
：Used in scenes where the background image has no rotation or borders.

kisvegabor, 2024-06-19:
Building up the tiled image from the scaled images and ignoring rotation can work. 

Do you have time to create a Pull Request for it?

closed 2024-07-08 by C-Ackerman