lvgl/lvgl #6368: Gridnav does not send LV_EVENT_FOCUSED

di-jtrumpower, 2024-06-13:
### LVGL version

v9.1.0-214-g59277de9f

### What happened?

Gridnav does not send LV_EVENT_FOCUSED, I found an issue linked in the forum from a year ago with example code:

https://forum.lvgl.io/t/trigger-lv-event-focused-with-grid-navigation/12371

I dug into the code and found that focus events are only sent from lv_indev (touchscreen/pointers). Style changes work fine which is strange. I am assuming the way that focus events are sent are different than how styling is set. Adding in the send event in the key handler for gridnav seems like the correct answer but wasn't sure if there was  reason it  wasn't done or I am missing something obvious.

kisvegabor, 2024-06-17:
Hi,

I've looked into it and I agree that manually sending the events [here](https://github.com/lvgl/lvgl/blob/master/src/others/gridnav/lv_gridnav.c#L231-L232) should work. Could send a pull request for it?

di-jtrumpower, 2024-06-17:
Yes I will send a PR shortly

kisvegabor, 2024-07-04:
Fixed by https://github.com/lvgl/lvgl/pull/6385

closed 2024-07-04 by kisvegabor