lvgl/lvgl #6373: Smooth font, Smooth display of text during rendering

Staok, 2024-06-14:
### Introduce the problem

Smooth font display function can be added during rendering.
Some fonts with lower BPP (BPP 1 font and export with compression , there is no change in size) may appear significantly jagged. However, once the BPP of the font is increased, the exported font file will be large.

[image]

### Proposal

Either optimize the font export compression algorithm to significantly reduce the size, or smooth the font during rendering

kisvegabor, 2024-06-17:
Hi,

Is bpp = 2 is already too large for you use case? What area you size constraints? Do you add a lot of characters into the fonts and it makes them very large?

Staok, 2024-06-20:
@kisvegabor yes, there are many characters used, for supporting multiple languages, such as China, Japan, and South Korea,
and just use bpp1, the 24 size font file is 10MB, even for linux device is very large

[image]

and the lvgl ui proj is using squareLine, so cannot direct use .ttf font file in program for now

Staok, 2024-06-21:
https://forum.squareline.io/t/can-squareline-direct-use-ttf-font-file-in-ui-program/3405
and squareLine planning to use .ttf font file in the future, case closed

closed 2024-06-21 by Staok