lvgl/lvgl #6374: Canvas examples

AZ-Renesas, 2024-06-14:
### LVGL version

9.1.1

### What happened?

The canvas examples are showing a black screen on EK-RA8D1.

### How to reproduce?

Try to run lv_example_canvas_1.c

kisvegabor, 2024-06-14:
Thanks, we will check it out! 

cc @becseya

becseya, 2024-06-18:
It seems to be a Dave2D bug, that requires deeper investigation.

In the meantime you can use `#define LV_USE_DRAW_DAVE2D 0`, @AZ-Renesas.

AZ-Renesas, 2024-06-18:
Hello,
Thanks for looking into this ?

Does this mean that I cannot implement something like on canvas examples ?
https://docs.lvgl.io/master/widgets/canvas.html

How does setting LV_USE_DRAW_DAVE2D to 0 help me to use all canvas functions ?

becseya, 2024-06-19:
Setting `LV_USE_DRAW_DAVE2D` to `0` instead of `1` in `src/lv_conf.h` will allow you run canvas examples and develop your application using canvas while we are working on the issue.

AZ-Renesas, 2024-06-20:
Thanks, after setting it to 0 I can implement a canvas. Please let us know for any updates.

kisvegabor, 2024-07-09:
Not stale

kisvegabor, 2024-07-10:
Sorry that it took so long. After a long debugging session I found the single number to change to resolve the issue. :slightly_smiling_face: 

See https://github.com/lvgl/lvgl/pull/6498

PR #6498 (2024-07-15): fix(dave2d): fix rendering to canvas with dave2d
fixes #6374

### Description of the feature or fix

FYI @nicusorcitu @anaGrad @cristian-stoica @FASTSHIFT 

### Notes
- Update the [Documentation](https://github.com/lvgl/lvgl/tree/master/docs) if needed.
- Add [Examples](https://github.com/lvgl/lvgl/tree/master/examples) if relevant.
- Add [Tests](https://github.com/lvgl/lvgl/blob/master/tests/README.md) if applicable.
- If you added new options to `lv_conf_template.h` run [lv_conf_internal_gen.py](https://github.com/lvgl/lvgl/blob/master/scripts/lv_conf_internal_gen.py) and update [Kconfig](ht [... cut]
closed 2024-07-15 by XuNeo