lvgl/lvgl #6387: SSD1306 Display with ESP32 using SPI

VigneshVicky97, 2024-06-18:
### Introduce the problem

I want to connect SSD1306 (128X64) display with ESP32 using SPI communication. The LVGL already has I2C communication for this particular Display. I have modified that code to communicate using SPI. But, I am not able to get the output. I don't know what I am wrong. I will attach my code . Can someone please help me to get correct output?

```
/*
 * SPDX-FileCopyrightText: 2021-2022 Espressif Systems (Shanghai) CO LTD
 *
 * SPDX-License-Identifier: CC0-1.0
 */

/* INCLUDES ------------------------------------------------------------------*/

#include "driver/gpio.h"
#include "driver/spi_master.h"

#include <stdio.h>
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "esp_timer.h"
#include "esp_lcd_panel_io.h"
#include "esp_lcd_panel_ops.h"
#include "driver/i2c.h"
#include "esp_err.h"
#include "esp_log.h"
#include "lvgl.h"
#include "esp_lvgl_port.h"
//#include "spi_oled_example_main.h"

#include "esp_lcd_panel_vendor.h"

static const char *TAG = "SSD1306_SPI";

/* MACROS --------------------------------------------------------------------*/
// Using SPI2 in the example
#define LCD_HOST  SPI2_HOST

////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////
//////////////////// Please update the following configuration according to your LCD spec //////////////////////////////
////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////
#define EXAMPLE_LCD_PIXEL_CLOCK_HZ     (2 * 1000 * 1000)

#define GPIO_SCLK                      18
#define GPIO_MOSI                      23
#define GPIO_DC                        27
#define GPIO_RESET                     32
#define GPIO_CS                        5

#define EXAMPLE_LCD_H_RES              128
#define EXAMPLE_LCD_V_RES              64

#define EXAMPLE_LCD_CMD_BITS           8
#define EXAMPLE_LCD_PARAM_BITS         8

extern void example_lvgl_demo_ui(lv_disp_t *disp);

/* The LVGL port component calls esp_lcd_panel_draw_bitmap API for send data to the screen. There must be called
lvgl_port_flush_ready(disp) after each transaction to display. The best way is to use on_color_trans_done
callback from esp_lcd IO config structure. In IDF 5.1 and higher, it is solved inside LVGL port component. */
static bool notify_lvgl_flush_ready(esp_lcd_panel_io_handle_t panel_io, esp_lcd_panel_io_event_data_t *edata, void *user_ctx)
{
    lv_disp_t * disp = (lv_disp_t *)user_ctx;
    lvgl_port_flush_ready(disp);
    return false;
}

void app_main(void)
{
	/* SPI Bus Initialisation --------------------------------------------------------------------*/
	esp_err_t ret;

//	gpio_reset_pin( GPIO_CS );
//	gpio_set_direction( GPIO_CS, GPIO_MODE_OUTPUT );
//	gpio_set_level( GPIO_CS, 0 );
//
//	gpio_reset_pin( GPIO_DC );
//	gpio_set_direction( GPIO_DC, GPIO_MODE_OUTPUT );
//	gpio_set_level( GPIO_DC, 0 );
//
//	if ( GPIO_RESET >= 0 ) {
//		gpio_reset_pin( GPIO_RESET );
//		gpio_set_direction( GPIO_RESET, GPIO_MODE_OUTPUT );
//		gpio_set_level( GPIO_RESET, 0 );
//		vTaskDelay( pdMS_TO_TICKS( 100 ) );
//		gpio_set_level( GPIO_RESET, 1 );
//	}

	spi_bus_config_t spi_bus_config = {
		.mosi_io_num = GPIO_MOSI,
		.miso_io_num = -1,
		.sclk_io_num = GPIO_SCLK,
		.quadwp_io_num = -1,
		.quadhd_io_num = -1,
//		.max_transfer_sz = 0,
		.max_transfer_sz = EXAMPLE_LCD_H_RES * 80 * sizeof(uint16_t),
		.flags = 0
	};

	ESP_LOGI(TAG, "SPI HOST_ID=%d", LCD_HOST);
	ret = spi_bus_initialize( LCD_HOST, &spi_bus_config, SPI_DMA_CH_AUTO );
	ESP_LOGI(TAG, "spi_bus_initialize=%d",ret);
	assert(ret==ESP_OK);
	/*  --------------------------------------------------------------------*/

	/* Panel IO Initialisation --------------------------------------------------------------------*/
	ESP_LOGI(TAG, "Install panel IO");
	esp_lcd_panel_io_handle_t io_handle = NULL;
	esp_lcd_panel_io_spi_config_t io_config = {
		.dc_gpio_num = GPIO_DC,
		.cs_gpio_num = GPIO_CS,
		.pclk_hz = EXAMPLE_LCD_PIXEL_CLOCK_HZ,
		.lcd_cmd_bits = EXAMPLE_LCD_CMD_BITS,
		.lcd_param_bits = EXAMPLE_LCD_PARAM_BITS,
		.spi_mode = 0,
		.trans_queue_depth = 10,
	};
	// Attach the LCD to the SPI bus
	ESP_ERROR_CHECK(esp_lcd_new_panel_io_spi((esp_lcd_spi_bus_handle_t)LCD_HOST, &io_config, &io_handle));
    /* --------------------------------------------------------------------*/

    /* SSD1306 Panel Driver Initialisation --------------------------------------------------------------------*/
    ESP_LOGI(TAG, "Install SSD1306 panel driver");
    esp_lcd_panel_handle_t panel_handle = NULL;
    esp_lcd_panel_dev_config_t panel_config = {
        .bits_per_pixel = 1,
        .reset_gpio_num = GPIO_RESET,
    };

    ESP_ERROR_CHECK(esp_lcd_new_panel_ssd1306(io_handle, &panel_config, &panel_handle));
    ESP_ERROR_CHECK(esp_lcd_panel_reset(panel_handle));
    ESP_ERROR_CHECK(esp_lcd_panel_init(panel_handle));
    ESP_ERROR_CHECK(esp_lcd_panel_disp_on_off(panel_handle, true));
    /* --------------------------------------------------------------------*/

    /* LVGL Library Initialisation --------------------------------------------------------------------*/
    ESP_LOGI(TAG, "Initialize LVGL");
    const lvgl_port_cfg_t lvgl_cfg = ESP_LVGL_PORT_INIT_CONFIG();
    lvgl_port_init(&lvgl_cfg);

    const lvgl_port_display_cfg_t disp_cfg = {
        .io_handle = io_handle,
        .panel_handle = panel_handle,
        .buffer_size = EXAMPLE_LCD_H_RES * EXAMPLE_LCD_V_RES,
        .double_buffer = true,
        .hres = EXAMPLE_LCD_H_RES,
        .vres = EXAMPLE_LCD_V_RES,
        .monochrome = true,
        .rotation = {
            .swap_xy = false,
            .mirror_x = false,
            .mirror_y = false,
        }
    };
    lv_disp_t * disp = lvgl_port_add_disp(&disp_cfg);
    /* Register done callback for IO */
    const esp_lcd_panel_io_callbacks_t cbs = {
        .on_color_trans_done = notify_lvgl_flush_ready,
    };
    esp_lcd_panel_io_register_event_callbacks(io_handle, &cbs, disp);

    /* Rotation of the screen */
    lv_disp_set_rotation(disp, LV_DISP_ROT_NONE);

    ESP_LOGI(TAG, "Display LVGL Scroll Text");
    example_lvgl_demo_ui(disp);
}
```

suda-morris, 2024-06-19:
https://github.com/espressif/esp-idf/issues/10635 may be related

VigneshVicky97, 2024-06-19:
> [espressif/esp-idf#10635](https://github.com/espressif/esp-idf/issues/10635) may be related

This is the exact issue I am also facing. Even after following the explanation, I am not able to solve that issue. Previously, I got that Noisy output in the screen. But, Now I don't get any output. How did you solve that problem? What are the settings which you have used?

liyang5945, 2024-06-19:
Man, you can't port lvgl without the screen driver working well, you should use a mature driver library. I wrote an example for you using a mature driver library, and it works well.
the driver form here:https://github.com/nopnop2002/esp-idf-ssd1306

```
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "esp_log.h"

#include "ssd1306.h"

#include "lvgl.h"

#define DISP_HOR_RES 128
#define DISP_VER_RES 64

static SSD1306_t dev;

static void disp_flush(lv_disp_drv_t *disp_drv, const lv_area_t *area, lv_color_t *color_p) {
    
    int32_t x;
    int32_t y;
    lv_color_t *colorPtr = color_p;
    ssd1306_clear_screen(&dev, false);
    for (y = area->y1; y <= area->y2; y++) {
        for (x = area->x1; x <= area->x2; x++) {
            /*Put a pixel to the display. For example:*/
            /*put_px(x, y, *color_p)*/
            if (colorPtr->full) {
                _ssd1306_pixel(&dev, x, y, 0);
            } else {
                _ssd1306_pixel(&dev, x, y, 1);
            }
            colorPtr++;
        }
    }
    ssd1306_show_buffer(&dev);
    lv_disp_flush_ready(disp_drv);
}

void lv_port_disp_init(void) {
    static lv_disp_draw_buf_t draw_buf_dsc_1;
    static lv_color_t buf_1[DISP_HOR_RES * DISP_VER_RES];                          /*A buffer for 10 rows*/
    lv_disp_draw_buf_init(&draw_buf_dsc_1, buf_1, NULL, DISP_HOR_RES * DISP_VER_RES);   /*Initialize the display buffer*/
    /*-----------------------------------
     * Register the display in LVGL
     *----------------------------------*/
    
    static lv_disp_drv_t disp_drv;                         /*Descriptor of a display driver*/
    lv_disp_drv_init(&disp_drv);                    /*Basic initialization*/
    
    /*Set up the functions to access to your display*/
    
    /*Set the resolution of the display*/
    disp_drv.hor_res = DISP_HOR_RES;
    disp_drv.ver_res = DISP_VER_RES;
    
    /*Used to copy the buffer's content to the display*/
    disp_drv.flush_cb = disp_flush;
    
    /*Set a display buffer*/
    disp_drv.draw_buf = &draw_buf_dsc_1;
    
    lv_disp_drv_register(&disp_drv);
}

 void app_main(void) {

#if CONFIG_SPI_INTERFACE
    spi_master_init(&dev, CONFIG_MOSI_GPIO, CONFIG_SCLK_GPIO, CONFIG_CS_GPIO, CONFIG_DC_GPIO, CONFIG_RESET_GPIO);
#endif // CONFIG_SPI_INTERFACE

#if CONFIG_SSD1306_128x64
    ssd1306_init(&dev, 128, 64);
#endif // CONFIG_SSD1306_128x64
    
    ssd1306_clear_screen(&dev, false);
    ssd1306_contrast(&dev, 0xff);
    ssd1306_display_text_x3(&dev, 0, "Hello", 5, false);
    vTaskDelay(500);
    
    _ssd1306_pixel(&dev, 100, 50, 0); //test draw a pixel
    ssd1306_show_buffer(&dev);
    
    vTaskDelay(500);
    
    //test draw a rect
    ssd1306_clear_screen(&dev, false);
    for (int y = 0; y < 20; y++) {
        for (int x = 0; x < 50; x++) {
            _ssd1306_pixel(&dev, x, y, 0);
        }
    }
    ssd1306_show_buffer(&dev);
    vTaskDelay(500);

    lv_init();
    lv_port_disp_init();
//
    lv_obj_t *scr = lv_scr_act();
    lv_obj_remove_style_all(scr);
    lv_obj_set_style_bg_opa(lv_scr_act(), LV_OPA_COVER, 0);
    lv_obj_set_style_bg_color(lv_scr_act(), lv_color_white(), 0);
    lv_obj_t *label = lv_label_create(lv_scr_act());
    
    lv_obj_center(label);
    lv_obj_set_style_text_color(label,lv_color_black(),0);
    lv_label_set_text(label, "hello lvgl");
    
    while (true) {
        lv_timer_handler();
        vTaskDelay(10);
    }
    
}

```
[image]

VigneshVicky97, 2024-06-19:
> Man, you can't port lvgl without the screen driver working well, you should use a mature driver library. I wrote an example for you using a mature driver library, and it works well. the driver from here:https://github.com/nopnop2002/esp-idf-ssd1306
> 
> ```
> #include "freertos/FreeRTOS.h"
> #include "freertos/task.h"
> #include "esp_log.h"
> 
> #include "ssd1306.h"
> 
> #include "lvgl.h"
> #include "demos/lv_demos.h"
> 
> 
> #define DISP_HOR_RES 128
> #define DISP_VER_RES 64
> 
> 
> static SSD1306_t dev;
> 
> 
> static void disp_flush(lv_disp_drv_t *disp_drv, const lv_area_t *area, lv_color_t *color_p) {
>     
>     int32_t x;
>     int32_t y;
>     lv_color_t *colorPtr = color_p;
>     for (y = area->y1; y <= area->y2; y++) {
>         for (x = area->x1; x <= area->x2; x++) {
>             /*Put a pixel to the display. For example:*/
>             /*put_px(x, y, *color_p)*/
>             bool pixel = colorPtr->full;
>             _ssd1306_pixel(&dev, x, y, !pixel);
>             colorPtr++;
>         }
>     }
>     ssd1306_show_buffer(&dev);
>     lv_disp_flush_ready(disp_drv);
> }
> 
> 
> void lv_port_disp_init(void) {
>     static lv_disp_draw_buf_t draw_buf_dsc_1;
>     static lv_color_t buf_1[DISP_HOR_RES * DISP_VER_RES];                          /*A buffer for 10 rows*/
>     lv_disp_draw_buf_init(&draw_buf_dsc_1, buf_1, NULL, DISP_HOR_RES * DISP_VER_RES);   /*Initialize the display buffer*/
>     /*-----------------------------------
>      * Register the display in LVGL
>      *----------------------------------*/
>     
>     static lv_disp_drv_t disp_drv;                         /*Descriptor of a display driver*/
>     lv_disp_drv_init(&disp_drv);                    /*Basic initialization*/
>     
>     /*Set up the functions to access to your display*/
>     
>     /*Set the resolution of the display*/
>     disp_drv.hor_res = DISP_HOR_RES;
>     disp_drv.ver_res = DISP_VER_RES;
>     
>     /*Used to copy the buffer's content to the display*/
>     disp_drv.flush_cb = disp_flush;
>     
>     /*Set a display buffer*/
>     disp_drv.draw_buf = &draw_buf_dsc_1;
>     
>     lv_disp_drv_register(&disp_drv);
> }
> 
> void app_main(void) {
> 
> #if CONFIG_SPI_INTERFACE
>     spi_master_init(&dev, CONFIG_MOSI_GPIO, CONFIG_SCLK_GPIO, CONFIG_CS_GPIO, CONFIG_DC_GPIO, CONFIG_RESET_GPIO);
> #endif // CONFIG_SPI_INTERFACE
> 
> 
> #if CONFIG_SSD1306_128x64
>     ssd1306_init(&dev, 128, 64);
> #endif // CONFIG_SSD1306_128x64
>     
>     ssd1306_clear_screen(&dev, false);
>     ssd1306_contrast(&dev, 0xff);
>     ssd1306_display_text_x3(&dev, 0, "Hello", 5, false);
>     vTaskDelay(500);
>     
>     _ssd1306_pixel(&dev, 100, 50, 0); //test draw a pixel
>     ssd1306_show_buffer(&dev);
>     
>     vTaskDelay(500);
>     
>     //test draw a rect
>     ssd1306_clear_screen(&dev, false);
>     for (int y = 0; y < 20; y++) {
>         for (int x = 0; x < 50; x++) {
>             _ssd1306_pixel(&dev, x, y, 0);
>         }
>     }
>     ssd1306_show_buffer(&dev);
>     vTaskDelay(500);
>     
>     lv_init();
>     lv_port_disp_init();
> //
>     lv_obj_t *scr = lv_scr_act();
>     lv_obj_remove_style_all(scr);
>     lv_obj_set_style_bg_opa(lv_scr_act(), LV_OPA_COVER, 0);
>     lv_obj_set_style_bg_color(lv_scr_act(), lv_color_white(), 0);
>     
>     lv_obj_t *label = lv_label_create(lv_scr_act());
>     lv_obj_center(label);
>     lv_obj_set_style_text_color(label, lv_color_black(), 0);
>     lv_label_set_text(label, "hello lvgl");
> //lv_demo_benchmark();
>     
>     while (true) {
>         lv_timer_handler();
>         vTaskDelay(10);
>     }
>     
> }
> ```
> 
> ![20240619_212142](https://private-user-images.githubusercontent.com/23631026/341090445-6907d222-e571-422b-8090-00f6a36a661e.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTg4MDU0NDYsIm5iZiI6MTcxODgwNTE0NiwicGF0aCI6Ii8yMzYzMTAyNi8zNDEwOTA0NDUtNjkwN2QyMjItZTU3MS00MjJiLTgwOTAtMDBmNmEzNmE2NjFlLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA2MTklMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNjE5VDEzNTIyNlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTI3OGUxNTkzMTY0ZmVlN2I5MTYzYzc0NmRmYTUwY2NlYzZkOTBkOTA4ZDViNTU1ZmMxNDBhZGJjZjg1MDExYWImWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.Ck9i-yHc_jmcOXyuB9CN_afLaQCLHByuCZAezEMWU2w)

Thankyou for your reply. I will use this code in my device and get you back once I got the output.

liyang5945, 2024-06-19:
I forgot to mention that you should change LV_COLOR_DEPTH to 1 and change LV_USE_THEME_BASIC to 0

VigneshVicky97, 2024-06-19:
> I forgot to mention that you should change lv_conf.h : LV_COLOR_DEPTH to 1 and change LV_USE_THEME_BASIC to 0，I use the lvlg 8.4.0

Sure. I will change those configurations. Once again Thanks.

VigneshVicky97, 2024-06-21:
Hi @liyang5945  ,
I used your code and changed the configurations as you said. But, the letters are not clearly displaying. It looks like the words from  the pen that runs out of ink. I don't know why it's showing like that. And also, there is the shadow or print of previously displayed contents on the screen. Do you also experienced like that ?

hey @suda-morris ,

```
        io_config.flags.dc_low_on_data = 0;
	io_config.flags.cs_high_active = 0;

```

Even after changing these two lines, I am not able to get the output. As the issue you mentioned, shows the exact problem in my code. You have added these two flags for connecting SSD1306 with SPI specific purpose only. Even after changing those flags in relation with the Datasheet, I am not able to get the output. Can you please suggest some solution based on your experience?

closed 2024-07-15 by lvgl-bot