lvgl/lvgl #6389: NuttX + FB device (SAMA5D2) - rendering problem/tearing

TimJTi, 2024-06-18:
### LVGL version

9.1.0

### What happened?

Having moved from LVGL V8 to V9 I am seeing issues with rendering basic graphics to an 800x480 framebuffer using NuttX.

I have created a very simple screen with just a spinner, but the leading/trailing edges of the spinner "tear" with horizontal lines much of the time.

https://github.com/lvgl/lvgl/assets/56726697/2e24582b-11a1-49e2-8078-c3567bbabb77

I have played around with every rendering option I can find (via Kconfig) but it is stubbornly consistent, and occurs for ARGB888 and RGB565. I am at a loss and am thinking it might be a problem with the LVGL+NuttX framebuffer implementation?

Neither V8 nor V9 use double buffering as far as I can tell, but is looks like it is almost like the rendering being done to the buffer at the same time as it is being written to the LCD. Or

LVGL demos suffer the same issue, but they didn't with V8 so I don't think its my (custom) board nor the NuttX RTOS itself (as opposed to the NuttX LVGL port)?

I have run out of ideas but am very happy to try things - I really don't want to go back to LVGL V8 given all the function name changes etc as that is a PITA!!

TimJTi, 2024-06-18:
For completeness here is (nearly) the same spinner in LVGL V8.3.3:

https://github.com/lvgl/lvgl/assets/56726697/323e11e4-60b0-42d3-8a2f-47b7e655f85b

liyang5945, 2024-06-19:
check your  lv_port_disp_init function, has a lv_display_set_buffers... function call, wha't the last param? if LV_DISPLAY_RENDER_MODE_FULL or LV_DISPLAY_RENDER_MODE_DIRECT change it to LV_DISPLAY_RENDER_MODE_PARTIAL

jianglianfang, 2024-06-20:
It happend to me when I enbled CONFIG_FB_UPDATE with single framebuffer mode. U can try this https://github.com/lvgl/lvgl/pull/6397. it may help you.

TimJTi, 2024-06-20:
> check your lv_port_disp_init function, has a lv_display_set_buffers... function call, wha't the last param? if LV_DISPLAY_RENDER_MODE_FULL or LV_DISPLAY_RENDER_MODE_DIRECT change it to LV_DISPLAY_RENDER_MODE_PARTIAL

@liyang5945 - if I change this, the display does not update at all unfortunately. The spinner is drawn once but there's then no animation of it.

TimJTi, 2024-06-20:
> It happend to me when I enbled CONFIG_FB_UPDATE with single framebuffer mode. U can try this #6397. it may help you.

@jianglianfang The SAMA5D2 NuttX configurations do not set CONFIG_FB_UPDATE as it is not necessary. If I enable it, the app crashes at the FBIO_UPDATE ioctl code as the fb->vtable entries NULL.

Since LVGL V8 works OK, in my case, it is not to do with FB_UPDATE but is likely more of a mismatch between the LVGL V9 framebuffer driver implementation and the specifics of the SAM5D2 LCDC peripheral and the NuttX driver for it :-(

kisvegabor, 2024-06-20:
Maybe @XuNeo has some ideas.

TimJTi, 2024-07-07:
> We need some feedback on this issue.
> 
> Now we mark this as "stale" because there was no activity here for 14 days.
> 
> Remove the "stale" label or comment else this will be closed in 7 days.

Well, Mr Bot: it's not sorted, no one seems to have any ideas, and although I don't have any thoughts I am very willing to try anything suggested. For now, I have to stick with V8.x

Perhaps you can apply your bot-powers to making technical suggestions?

XuNeo, 2024-07-08:
Hi Tim,

Sorry I have no clue what's wrong. I though it was screen tearing, but it looks more like a dirty area issue. Maybe you could try to enable `LV_USE_REFR_DEBUG` to see how LVGL does the refresh?

TimJTi, 2024-07-08:
@XuNeo 
Hi - thank you for replying. I always wondered what "Draw random coloured rectangles over the redrawn area" meant! 

The redrawn areas in LVGL 9 are much smaller than in V8, and I don't think they are centred correctly in V9 compared to V8?

What determines the redraw area and centre - is it related to the widget or something derived from lv_conf?

Rather difficult to capture, but here's what V9 does.

https://github.com/lvgl/lvgl/assets/56726697/85b5b007-b22f-44f4-a873-9b3fe1757c9d

XuNeo, 2024-07-09:
> The redrawn areas in LVGL 9 are much smaller than in V8, and I don't think they are centred correctly in V9 compared to V8?

These are imporvements compared with v8, if v8 is not reshing only the changed areas. 

For now, my best guess is the screen tearing. You can verify it by enabling double buffer. I'm not familar with SAM MCU, if you have limited RAM, maybe you can try to lower the resolution.

TimJTi, 2024-07-09:
v8 refreshes in pretty much the same way but the random rectangles are a little larger and seen to cover the leading edge of the spinner better. My hunch is that there is something to do with the interplay of a full framebuffer, smaller dirty areas.

I have tried RGB565 just in case, and it is the same.

Plenty of RAM, so no need to reduce resolution - but since V8 works and V9 doesn't it has to still be an issue with V9 in combination with NuttX, but could be unique to the SAMA5. Nuttx uses a simple full framebuffer, and that does not change in anyway when I switch between v8 and v9 - it is still /dev/fb0 with no NuttX config changes at all. The V9 demo apps suffer the same as well.

I did actually try and make it work with a double buffer (by allowing more memory for 2 framebuffers, and setting virtual x and y offsets etc) and it it did run but had some issues (it was a hack) - but this rendering problem was still there. 

One day I may find time to investigate this more but for now, V9 is not a possibility for me when V8 works fine. Big shame.

XuNeo, 2024-07-09:
The tearing normally happen when the same memory is changing while screen is refreshing. Since LVGL's trace is fully ready with nuttx, you may try to add trace point to the driver, where it starts and finishes refreshing screen. Check the buffer address against the one LVGL is rendering. Any overlap is a sign of tearing.

For single buffer case, if the screen is a command LCD(the one with integerated controller and GRAM), you can avoid the tearing by time spliting the buffer for LVGL and driver.

For single buffer with video mode LCD(with controller locates on MCU and refreshing screen all the time), it's not possible to avoid tearing.

TimJTi, 2024-07-09:
> The tearing normally happen when the same memory is changing while screen is refreshing. Since LVGL's trace is fully ready with nuttx, you may try to add trace point to the driver, where it starts [...]

Thanks - I will try that and see what I find.

> For single buffer case, if the screen is a command LCD(the one with integerated controller and GRAM), you can avoid the tearing by time spliting the buffer for LVGL and driver.
>
> For single buffer [...]

The SAMA5 has an integrated LCD controller (actually a complex and powerful one but, for now, NuttX only supports the base layer).

But since LVGL V8 does not suffer the tearing issue I still do not understand why LVGL V9 does, unless there's something very different with the V9 NuttX fb interface that just *has* to be like it is and therefore a problem and unusable?

If the answer is for me to write my own LCD driver interface for the SAMA5 I will, but I want to be 100% that is the only way to get it to work and that it isn't just a bug in the current LVGL NuttX framebuffer driver.

TimJTi, 2024-07-11:
@XuNeo @kisvegabor 
Brief update. As my LVGL V8 screens get more complex, I am seeing FPS drop very slow and CPU usage rise very high. Using the "Draw random coloured rectangles over the redrawn area" feature it seems that the entrie screen is refreshed most of the time rather than just the dirt areas:
- with just a spinner, the leading and trailing edges of the spinner are the only parts refreshed
- with 8 duplicate custom "areas" - like a custom widget (similar to a Squareline component I guess) composed of labels, and bars, arranged using grid layout, and a meter, it is diabolical.

Looking at the NuttX framebuffer driver, specifically with SAMA5 arch in mind, it seems very dumb and is just a a full framebuffer that LVGL writes to with no real ability to do an "update area" as driver by CONFIG_FB_UPDATE (as mentioned by @jianglianfang earlier) and no double buffering.

When I first used the SAMA5 it was "bare metal" with a traditional LVGL LCD driver switching framebuffers and worked well, so I know it is not LVGL and not SAMA5....so I am back full circle to "blaming" the very basic FB support in NuttX for this processor.

I will switch back to LVGL V9 and the simple spinner and try and fix/improve the NuttX support and report back.

Thank you to those here who have tried to help - although there was no magic answer it has helped me sit back and work through where the likely issue is.

TimJTi, 2024-07-11:
Update from today's work - I will keep doing this, just in case now or in the future anyone else encounters similar issues.

- Found a comment from @XuNeo I think it was that if I clone from the LVGL repo to the NuttX lvgl folder I can easily switch between LVGL versions
- LVGL "master" results in nothing displayed on the LCD. I think there are comments relating to  recent NuttX related commits here that would explain this so am going with 9.1 release for now.
- I added an FB_UPDATE function for the SAMA5 to allow areas to be redrawn rather than only ever relying on a whole FB refresh. Had fun with that and have found that the display is updated even if I just `return OK` from the updatearea function...which means, I think, that because the FB memory used by the NuttX SAMA5 code is directly written to by LVGL (i.e. no intermediate buffer) then the are update is redundant and, worse, fights!
- Using the same area update code with LVGL V8 and I still get horrendous CPU usage and low frame rates - it is like V8 is not clever in working out the minimal dirty areas to redraw - there seems little point trying to get that improved as V9 onwards has to be the way to go, to keep current.
- I cherry picked the @jianglianfang commit relating to the invalid area joins but it is not a magic fix.
- I need to now investigate the suggestion from @XuNeo to trace what's going on; I am already using a jlink debugger but there might be information from this once I work out what to so.
- I am part way through getting double buffering implemented, but not successful so far
- I am also trying to dig out old Atmel softpack-related code I had working to see how that updated the frame buffers

Or, I am totally missing the point here and the fix is simple and stupid but alluding me!!!!

XuNeo, 2024-07-12:
Hi,

> The SAMA5 has an integrated LCD controller (actually a complex and powerful one but, for now, NuttX only supports the base layer).

If MCU has integerated LCD controller, it's more likely it will refresh the screen at fixed rate, no matter if LVGL is drawing or not. 
That explains the flicking you saw. The intermediate result LVGL is drawing, is sent to screen. Normally LVGL will clear the background(the flicking rectangle), then drawing the arc.

For single buffer configuration, you may need to use `LV_DISPLAY_RENDER_MODE_PARTIAL` that [liyang5945](https://github.com/liyang5945) suggested at beginning. Unfortunately, it's not supported in nuttx port. Partial mode allows to drawing intermediate result to a standalone buffer before memcpy-ing to FB buffer.

TimJTi, 2024-07-12:
Today's work.
Based on the LVGL Linux FB code, I added code to do partial redraw for `LV_DISPLAY_RENDER_MODE_PARTIAL` . It has fixed the "tearing" but has 2 issues nevertheless and I would appreciate any suggestions.

1. The leading edge of the spinner shows a black box - the dirty/redrawn area perhaps. I tried the "update the joined invalid area instead of the last one" fix but it made the whole screen go crazy.
2. Top edge of the screen is being trashed.
    - If I bypass the `flush_cb()` completely the top edge is not trashed (but obviously the spinner doesn't spin; it is just the first drawn spinner that is there, stationary)
    - The same is true if I bypass the actually memcpy statement (I tried both memcpy and lv_memcpy)  of the flush_cb and just call `lv_display_flush_ready(disp)`. Again, no spinning spinner but the top of the screen is trashed.

It feels like I am very close to having code that would enhance LVGL V9 with NuttX and a processor that has an LCDC using  memory-mapped DMA framebuffer....and will gladly dumbit a PR once it's is sorted :-)

https://github.com/user-attachments/assets/0211e7aa-da31-4248-a6c5-04e3db9beacc

TimJTi, 2024-07-12:
The code (with `fbdev_join_areas() `inactive here:
```
    /* Join the areas to update */
    //lv_memzero(&final_inv_area, sizeof(final_inv_area));
    //fbdev_join_inv_areas(disp, &final_inv_area);

    final_inv_area = *area;

    int32_t w = lv_area_get_width(&final_inv_area);
    uint32_t px_size = lv_color_format_get_size(lv_display_get_color_format(disp));
    uint32_t color_pos = (final_inv_area.x1 + dsc->pinfo.xoffset) * px_size + final_inv_area.y1 * dsc->pinfo.stride;
    uint32_t fb_pos = color_pos + yoffset * dsc->pinfo.stride;

    uint8_t * fbp = (uint8_t *)dsc->mem;
    int32_t y;
    if (disp->render_mode == LV_DISPLAY_RENDER_MODE_DIRECT) {
        for(y = final_inv_area.y1; y <= final_inv_area.y2; y++) {
            lv_memcpy(&fbp[fb_pos], &color_p[color_pos], w * px_size);
            fb_pos += dsc->pinfo.stride;
            color_pos += dsc->pinfo.stride;
        }
    }
    else {
        for(y = final_inv_area.y1; y <= final_inv_area.y2; y++) {
            lv_memcpy(&fbp[fb_pos], color_p, w * px_size);
            fb_pos += dsc->pinfo.stride;
            color_p += w * px_size;
        }
    }

```

TimJTi, 2024-07-12:
>     2. Top edge of the screen is being trashed.
I have just worked out the reason for this and have done a quick and dirt fix to prove it but would appreciate ideas on the RIGHT way to fix this.

NuttX Kconfig for this processor needs to you to enter the FB size and address for it. LVGL uses that address for the dsc->mem address. That is the address of the buffer supplied to the flush_cb as well as of the frame buffer - it is set in `lv_nuttx_fbdev_set_file()` as part of the init.

So the frame buffer itself gets written to twice.

If I manually assigned a (reserved) area of memory for it, I can copy the buffer to the actual framebuffer properly and the nasties at the top disappear. But I still get the black box leading the spinner.

Perhaps this is where the NuttX/LVGL `Use Custom Nuttx init API` comes in?

Thank you for your patience and I hope someone can make good suggestions on how to fix this all nicely :-)

TimJTi, 2024-07-18:
My quick fix wasn't quite right.
For completeness here is my conclusion.

The SAMA5D2 LCDC uses framebuffers and as @XuNeo said this is used continuously to update the actual LCD with data. My quick fix of putting code into the LVGL NuttX FB code still allowed multiple writes to the LCDC FB memory as the image was rendered, hence the continued artifacts noted recently.

The SAMA5 NuttX support had no FB_UPDATE functions, otherwise the (indirect) suggestion from @jianglianfang  might have worked.

I have now added this function to the NuttX SAMA5D2 LCDC driver, and moved my quick fix code from the LVGL FB driver to there (and fixed it up better), and it seems to be behaving.

I have - for now - added (in my NuttX fork) a new Kconfig entry to define a specific buffer address that is supplied as the `pinfo->fbmem` address, which is what LVGL can now render to, and the FPIO_UPDATE call allows the NuttX code to memcpy the rendered data to the "real" framebuffer.

I am still suspicious that this may still foul up the display, and I am still digging into the DMA aspects of it more deeply by reference back to the original Atmel (now Microchip of course) "softpack" drivers, but I do now understand it is a NuttX issue not an LVGL issue so I will close this here for now.

I will submit a NuttX draft PR once I am happier as the method I am using can undoubtedly be bettered.

TimJTi, 2024-07-19:
@kisvegabor @XuNeo 
Reopening.

Feedback from the [PR](https://github.com/apache/nuttx/pull/12733) I submitted to NuttX to "fix" this, once I had "concluded" it was specific to the SAMA5D2, makes me, again, strongly suspect that the NuttX framebuffer driver is flawed after all. As best as I can tell, the driver allows LVGL to render directly to the framebuffer - if so, then no wonder it has artefacts.

I perhaps had the right idea when I looked at the LVGL Linux FB driver, which does a memcpy from the draw buffger to the framebuffer - but when I did that I was, of course, simply copying the framebuffer to itself!

A closed look at l`v_linux_fb.c` shows that it opens the framebuffer device, works out the display size, and mmaps it but, crucially, then goes on to malloc its own draw_buffers:

**NuttX FB driver**

```
    if((ret = fbdev_get_pinfo(dsc->fd, &dsc->pinfo)) < 0) {
        goto errout;
    }

    dsc->mem = mmap(NULL, dsc->pinfo.fblen, PROT_READ | PROT_WRITE,
                    MAP_SHARED | MAP_FILE, dsc->fd, 0);
    if(dsc->mem == MAP_FAILED) {
        LV_LOG_ERROR("ioctl(FBIOGET_PLANEINFO) failed: %d", errno);
        ret = -errno;
        goto errout;
    }

    uint32_t w = dsc->vinfo.xres;
    uint32_t h = dsc->vinfo.yres;
    uint32_t stride = dsc->pinfo.stride;
    uint32_t data_size = h * stride;
    lv_draw_buf_init(&dsc->buf1, w, h, LV_COLOR_FORMAT_NATIVE, stride, dsc->mem, data_size);
```
Then the flush_cb - um - actually does nothing as LVGL has rendered direct to the mapped framebuffer

**Linux FB driver**

```
    /* Map the device to memory*/
    dsc->fbp = (char *)mmap(0, dsc->screensize, PROT_READ | PROT_WRITE, MAP_SHARED, dsc->fbfd, 0);
```

then (essentially)

```
    uint8_t * draw_buf = NULL;
    draw_buf = malloc(draw_buf_size);
...
    lv_display_set_buffers(disp, draw_buf, draw_buf_2, draw_buf_size, LV_LINUX_FBDEV_RENDER_MODE);

```
And the flush_cb does a memcpy from the draw buffer to the mapped framebuffer.

Have I missed something or have I finally homed in on the root cause?

TimJTi, 2024-07-19:
> Have I missed something or have I finally homed in on the root cause?
What I have probably missed - so far - is `lv_draw_buf_init` which the linux driver doesn't use, but NuttX does - and that, I think, is setting up the draw buffer.

I am investigating still...and going around in circles, still!!

TimJTi, 2024-07-19:
Having promised myself I would leave this alone until someone who knows what they're doing cam along (PLEASE!!!), I have created a version of the lv_nuttx_fbdev.c that is more closely aligned with the linux one.

It's [here](https://github.com/TimJTi/lvgl.git)

I have two flavours of flushing - the entire frame buffer, or just the changed area. Here is what happens:

1. LVGL no longer renders direct to the FB which has to be a good thing
2. The tearing has gone
3. It behaves OK if I refresh the entire framebuffer (all 800x400 x 32bpp of it) but frame rate is horrid - 13 FPS and 72% CPU!
4. If I refresh just the dirty areas - as defined by the `area` parameter of the `flush_cb `call, it behaves EXACTLY as I reported [above](https://github.com/lvgl/lvgl/issues/6389#issuecomment-2225845612) with NO fps/cpu stat updates, and the forward edge of the spinner is a black box.
5. Using the code as-was I get the tearing as originally reported which is - I am sure - because LVGL is rendering direct to the framebuffer: I get 33fps and 2% CPU.

FYI - if I use the LVGL master branch I get a black display so I have to use the 9.1 release.

If someone could PLEASE PLEASE PLEASE help as I have no more time to play around with what I genuinely think is a defect in V9 and it's interaction with NuttX. I can try suggestions easily, but I can't spend time prodding and poking and hoping any more.

I will stick with my quick and dirty hack to the SAMA5 kernel driver in NuttX, which is in the wrong place, and  will never be merged into NuttX, quite rightly.

TimJTi, 2024-07-20:
Perhaps the answer is `lv_disp_flush_is_last(&disp_drv)` ?

TimJTi, 2024-07-22:
@XuNeo @kisvegabor 
> Perhaps the answer is `lv_disp_flush_is_last(&disp_drv)` ?

I was right...but wrong...

`lv_disp_flush_is_last(&disp_drv)` is called at the start of the flush, but **_shouldn't be_**. It means all but the last areas that are rendered are SKIPPED and only the LAST one is copied to the framebuffer. Hence no FPS update and the black rectangle area at the leading edge of the spinner.

I have this working now, with no artefacts, with the NuttX /dev/fb0 device on a SAMA5D2, and get 50fps and 0%  (i.e. < 1% I guess) cpu usage.

I am not 100% sure of why there's some other code in this driver but will read up on submission rules etc and do a PR soon.

TimJTi, 2024-07-24:
Now an ongoing PR [here](https://github.com/lvgl/lvgl/pull/6561)

closed 2024-07-24 by TimJTi