lvgl/lvgl #6390: label text does not show in transparent background display

docterling, 2024-06-19:
### LVGL version

v9.1.0

### What happened?

### 1.in v8.3.5 i draw some text on a transp display i can see it,but no in v9.1.0
### 2.in v9.1.0,in init func i had already called lv_canvas_set_draw_buf to set draw buf,but in draw text func i must call it again or else it will not trigger the disp_flush callback

### How to reproduce?

### in v8.3.5
**1.init a display and a canvas**
```c
lv_init();
lv_disp_draw_buf_init(&lvgl_draw_buf_dsc,lvgl_disp_buf, NULL, lvgl_buf_size/4);
lv_disp_drv_init(&lvgl_disp_drv);
lvgl_disp_drv.draw_buf = &lvgl_draw_buf_dsc;
lvgl_disp_drv.flush_cb = disp_flush;
lvgl_disp_drv.hor_res = lvgl_disp_w;
lvgl_disp_drv.ver_res = lvgl_disp_h;
lvgl_disp_drv.antialiasing = 1;
lvgl_disp_drv.full_refresh = 1;
lvgl_disp_drv.screen_transp = 1;
lvgl_disp = lv_disp_drv_register(&lvgl_disp_drv);
lv_disp_set_bg_opa(lv_disp_get_default(),LV_OPA_0);

canvas = lv_canvas_create(lv_scr_act());
lv_canvas_set_buffer(canvas, canvasbuf, lvgl_disp_w, lvgl_disp_h, LV_IMG_CF_TRUE_COLOR_ALPHA);
lv_obj_set_align(canvas, LV_ALIGN_TOP_LEFT);
lv_canvas_fill_bg(canvas, lv_color_white(), LV_OPA_0);
```
**2.draw some text**
```c
lv_draw_label_dsc_t label_dsc;
lv_draw_label_dsc_init(&label_dsc);
label_dsc.color = lv_color_black();
label_dsc.font = g_lvgl_font;
lv_canvas_draw_text(canvas, x, y+text_h*num, lvgl_disp_w, &label_dsc, "hello world");
```
### in v9.1.0
**1.init a display and a canvas**
```c
lv_init();
lvgl_disp = lv_display_create(lvgl_disp_w,lvgl_disp_h);
lv_display_set_flush_cb(lvgl_disp, disp_flush);
lv_display_set_buffers(lvgl_disp, lvgl_disp_buf, NULL, lvgl_buf_size, LV_DISPLAY_RENDER_MODE_DIRECT);
lv_display_set_antialiasing(lvgl_disp, true);
lv_obj_set_style_bg_opa(lv_screen_active(), LV_OPA_0, LV_PART_MAIN);

canvas = lv_canvas_create(lv_screen_active());
lv_canvas_set_draw_buf(canvas, &canvasbuf);
lv_obj_set_align(canvas, LV_ALIGN_TOP_LEFT);
lv_canvas_fill_bg(canvas, lv_color_white(), LV_OPA_0);
```
**2.draw some text**
```c
lv_draw_label_dsc_t label_dsc;
lv_layer_t layer;
lv_area_t coords;
lv_canvas_set_draw_buf(canvas, &canvasbuf);
lv_canvas_init_layer(canvas, &layer);
lv_draw_label_dsc_init(&label_dsc);
label_dsc.color = lv_color_black();
label_dsc.font = g_lvgl_font;
label_dsc.opa = LV_OPA_COVER;
label_dsc.align = LV_TEXT_ALIGN_LEFT;
label_dsc.decor = LV_TEXT_DECOR_NONE;
label_dsc.blend_mode = LV_BLEND_MODE_NORMAL;
label_dsc.text = "hello world";
coords.x1 = x;
coords.y1 = y + text_h*num;
coords.x2 = x + (2 + strlen("hello world")) * text_h;
coords.y2 = coords.y1 + text_h;
lv_draw_label(&layer, &label_dsc, &coords);
lv_canvas_finish_layer(canvas, &layer);
```

kisvegabor, 2024-06-21:
How do you create `canvasbuf`? Is it a static or global variable?

docterling, 2024-06-23:
canvasbuf it's a class member,in init calloc for canvasbuf.data
draw_label is a member func.

kisvegabor, 2024-06-28:
Please provide a self sufficient canvas only code snippet that I can try in a simulator on PC.

docterling, 2024-07-01:
you can try this with v9.1.0,but first you must impl the disp_flush
ui.h
```
#ifndef ___UI_H__
#define ___UI_H__

extern "C"{
#include "lv_conf.h"
#include "lvgl.h"
}

typedef struct {
    const char * name;  /* The name of the font file */
    lv_font_t * font;   /* point to lvgl font */
    lv_freetype_font_render_mode_t mode;
    uint16_t weight;    /* font size */
    uint16_t style;     /* font style */
} lv_ft_info_t;

class CLvglUI
{
public:
	CLvglUI();
	~CLvglUI();
	int init();
	void deinit();
	void draw_osd(char *info);	
private:
	void clear_bg();
public:	
	int lvgl_disp_w;
	int lvgl_disp_h;
	int canvas_w;
	int canvas_h;
        int text_h;
	lv_ft_info_t arial_font_info;
	lv_color_t *lvgl_disp_buf;
	lv_draw_buf_t canvasbuf;
	lv_obj_t *canvas;
	int lvgl_buf_size;
	int canvas_size;
	bool lvgl_inited;
private:
	lv_disp_t *lvgl_disp;
};
```
ui.cpp
```
#include <sys/time.h>
#include <stdlib.h>
#include <stdio.h>
#include "ui.h"

extern "C" 
{
extern lv_font_t *g_lvgl_font;//define in src/font/lv_font.c

uint32_t custom_tick_get(void)
{
    struct timeval tv_now;
    gettimeofday(&tv_now, NULL);
    uint64_t time_ms;
    time_ms = (tv_now.tv_sec * 1000000 + tv_now.tv_usec) / 1000;
    return time_ms;
}

void custom_delay_cb(uint32_t ms)
{
	if(ms > 0)
		usleep(ms*1000);
[... 193 lines cut ...]
    }	
    return NULL;
}

void procSig(int signo)
{
	g_quit = 1;
}

int main(int argc,char *argv[])
{
    pthread_t pid;
    int i = 0;
    signal(SIGINT,  procSig);
    pthread_create(&pid, NULL, ui_thread,NULL);
    while(!g_quit )
    {
        usleep(10*1000);
        i++;
        if(500 == i)
            g_lvglui.draw_osd("Hello World");
    }
}
```
[arial.zip](https://github.com/user-attachments/files/16049071/arial.zip)

kisvegabor, 2024-07-17:
I've simplified your code a little bit and it work on my end with the latest `master`:
```c
lv_obj_t * canvas;

void draw_osd(char *info)
{
	int x;
	int y;
	x = 32;
	y = 32;

	lv_draw_label_dsc_t label_dsc;
	lv_layer_t layer;
	lv_area_t coords;
        lv_canvas_init_layer(canvas, &layer);
	lv_draw_label_dsc_init(&label_dsc);

	label_dsc.color = lv_color_black();
	label_dsc.opa = LV_OPA_COVER;
	label_dsc.align = LV_TEXT_ALIGN_LEFT;
	label_dsc.decor = LV_TEXT_DECOR_NONE;
	label_dsc.blend_mode = LV_BLEND_MODE_NORMAL;
	label_dsc.text_local = 1;
	label_dsc.text = info;
	coords.x1 = x;
	coords.y1 = y;
	coords.x2 = x + (2 + strlen(info)) * 28;
	coords.y2 = coords.y1 + 28;
	lv_draw_label(&layer, &label_dsc, &coords);
	lv_canvas_finish_layer(canvas, &layer);
}

...

 int32_t canvas_w = 200;
  int32_t canvas_h = 200;

  int32_t canvas_size = _LV_DRAW_BUF_SIZE(canvas_w,canvas_h, LV_COLOR_FORMAT_ARGB8888);
  lv_draw_buf_t canvasbuf;
  canvasbuf.data = (uint8_t *)calloc(1,canvas_size);
  canvasbuf.header.magic = LV_IMAGE_HEADER_MAGIC;
  canvasbuf.header.cf = LV_COLOR_FORMAT_ARGB8888;
  canvasbuf.header.flags = LV_IMAGE_FLAGS_MODIFIABLE;
  canvasbuf.header.w = canvas_w;
  canvasbuf.header.h = canvas_h;
  canvasbuf.header.stride = _LV_DRAW_BUF_STRIDE(canvas_w, LV_COLOR_FORMAT_ARGB8888);
  canvasbuf.header.reserved_2 = 0;
  canvasbuf.data_size = canvas_size;
  canvasbuf.unaligned_data = canvasbuf.data;

  canvas = lv_canvas_create(lv_screen_active());
  lv_canvas_set_draw_buf(canvas, &canvasbuf);
  lv_obj_set_align(canvas, LV_ALIGN_TOP_LEFT);
  lv_canvas_fill_bg(canvas, lv_color_white(), LV_OPA_0);

  draw_osd("Hello");
```

docterling, 2024-07-18:
I've changed back to v8.4.0
and thank you

closed 2024-07-18 by docterling