lvgl/lvgl #6391: It sucks in draw_buf_flush function

Jamie793, 2024-06-19:
### Introduce the problem

I am porting LVGL 9.1 with an LCD driver for ST7735. Currently, I'm encountering a problem with draw_buf_flush. During debugging, I found that the issue occurs within the lv_task_handler function. It is stuck in the following code segment:
```c
while (layer->draw_task_head) {
    lv_draw_dispatch_wait_for_request();
    lv_draw_dispatch();
}
```

kisvegabor, 2024-06-21:
Hi,

A few questions:
- Have you set an `OS` with `LV_USE_OS`?
- Do you use a GPU?
- With which UI/demo do you test?

Jamie793, 2024-06-21:
> Hi,
> 
> A few questions:
> 
> * Have you set an `OS` with `LV_USE_OS`?
> * Do you use a GPU?
> * With which UI/demo do you test?

Thank you for your reply.
Instead of using LVGL's OS, I ported the FreeRTOS kernel myself. 
My MCU lacks a GPU and has only 128K ROM and 16K RAM, with a DMA that isn't DMA2. 
I used the test code that I wrote myself.

```c
lv_obj_t *label = lv_label_create(lv_scr_act(), NULL); /*Add a label to the button*/
lv_label_set_text(label, "Testing");                   /*Set the labels text*/
lv_obj_set_pos(label, 30, 30);
lv_obj_set_style_local_text_color(label, LV_LABEL_PART_MAIN, LV_STATE_DEFAULT, lv_color_hex(0xFF0000));

// Systick interrupt
void SysTick_Handler(void)
{
    lv_tick_inc(1);
    delay_decrement();
}

// Main loop
while(1){
    lv_task_handler();
}
```

kisvegabor, 2024-06-21:
16kB RAM is very small. What is the size of `LV_MEM_SIZE`? It can easily happen that there was an out of memory issue somewhere.

Please enable all LV_USE_ASSERTs in lv_conf.h

Jamie793, 2024-06-21:
> LV_USE_ASSERTs

The size of LV_MEM_SIZE is (8u * 1024) and have set LV_USE_ASSERT_MEM to 1.
It didn't break in assert when I debugged.

kisvegabor, 2024-06-25:
Sorry, but I don't know what could be the problem. I haven't heard similar issues from other, so I assume it's related to the small memory somehow. :slightly_frowning_face:

kisvegabor, 2024-06-28:
Can it #6414 related?

closed 2024-07-24 by lvgl-bot

Jamie793, 2024-08-06:
> Sorry, but I don't know what could be the problem. I haven't heard similar issues from other, so I assume it's related to the small memory somehow. 🙁

It is similar after changing to 64K RAM and 256K ROM.

kisvegabor, 2024-08-07:
My guess is that it's the RAM not the ROM.

Jamie793, 2024-08-07:
> My guess is that it's the RAM not the ROM.

Yep， But my RAM is 64K and show me 50% RAM left after compiling.

Jamie793, 2024-08-09:
> My guess is that it's the RAM not the ROM.

I used FreeRTOS and set OS macro to RTOS. But it can't too. stuck in "draw_buf_flush" function

kisvegabor, 2024-08-12:
Does it work without `LV_OS_NONE`? 

There is an open issue about FreeRTOS: you need to call `lv_display_create` and `lv_timer_handler` in the same thread.

Jamie793, 2024-08-13:
> Does it work without `LV_OS_NONE`?
> 
> There is an open issue about FreeRTOS: you need to call `lv_display_create` and `lv_timer_handler` in the same thread.

When I debug, I find that It doesn't jump into the condition.
[image]

Jamie793, 2024-08-14:
I turn on "LV_USE_DRAW_SW". The task isn't run after creating the task.
```c
lv_result_t lv_thread_init(lv_thread_t * pxThread, lv_thread_prio_t xSchedPriority,
                           void (*pvStartRoutine)(void *), size_t usStackSize,
                           void * xAttr)
{
    pxThread->xTaskArg = xAttr;
    pxThread->pvStartRoutine = pvStartRoutine;

    BaseType_t xTaskCreateStatus = xTaskCreate(
                                       prvRunThread,
                                       pcTASK_NAME,
                                       (uint16_t)usStackSize,
                                       (void *)pxThread,
                                       tskIDLE_PRIORITY + xSchedPriority,
                                       &pxThread->xTaskHandle);

    /* Ensure that the FreeRTOS task was successfully created. */
    if(xTaskCreateStatus != pdPASS) {
        LV_LOG_ERROR("xTaskCreate failed!");
        return LV_RESULT_INVALID;
    }

    return LV_RESULT_OK;
}
```

kisvegabor, 2024-08-15:
In this case I recommend debugging what's happening [in this functions](https://github.com/lvgl/lvgl/blob/master/src/draw/sw/lv_draw_sw.c#L394).

WjunqiangS, 2025-09-25:
Have you solve it ? i have met the same problem as you