lvgl/lvgl #6395: Alternative solutions for the implementation of lv_obj_is_valid()

bhspyder, 2024-06-19:
### Introduce the problem

The function lv_obj_is_valid() as it is currently implemented searches every object (and it's chidren recursively) in every screen on every display looking for a match. This does not scale well on big projects, especially if all of the screens and objects are created on bootup and exist for lifetime of the application.

I ran into this performance bottleneck while using the theme support provided by SquareLine studio generated code (I understand SquareLine is not affiliated with LVGL any more so this is more of an aside) which uses lv_obj_is_valid() to check for valid objects in it's theme style linked list.

Are there any alternative solutions that can be considered to make this function more performant?

liamHowatt, 2024-06-24:
Nice to be evaluating this implementation. Thanks for your input.

> If the object was deleted and the block of feed memory was reused by another lv_obj_t starting at the same offset, it would have a valid class_p and be considered valid.

Yes, that's even a weakness of the current implementation. I was able to exploit it in 2 minutes.
```c
static void is_valid_test(void)
{
    lv_obj_t * active_screen = lv_screen_active();

    lv_obj_t * obj = lv_obj_create(active_screen);
    LV_LOG_USER("%d", (int) lv_obj_is_valid(obj));
    lv_obj_delete(obj);
    LV_LOG_USER("%d", (int) lv_obj_is_valid(obj));
    lv_obj_create(active_screen);
    LV_LOG_USER("%d", (int) lv_obj_is_valid(obj));
}
```
It prints `1`, `0`, `1`, which is incorrect. It's exactly what you described. The memory block got reused by the allocator and there's nothing to distinguish it from the one that replaced it. I think your class solution would be susceptible too.

Take a look at obj ID. (`LV_USE_OBJ_ID`). The builtin implementation for assigning object IDs is to increment a monotonic counter for each created object: [source](https://github.com/lvgl/lvgl/blob/21a6a8bb87a89ded4644d8356abafc43e3fa4696/src/core/lv_obj_id_builtin.c#L45). We could use the ID system to improve `lv_obj_is_valid`. But it would still have the performance issue you're talking about, because it still has to traverse the obj tree.

```c
static void is_valid_test_using_obj_id(void)
{
    lv_obj_t * active_screen = lv_screen_active();

    lv_obj_t * obj = lv_obj_create(active_screen);
    void * obj_id = lv_obj_get_id(obj);
    LV_LOG_USER("%p", lv_obj_get_child_by_id(NULL, obj_id));
    lv_obj_delete(obj);
    LV_LOG_USER("%p", lv_obj_get_child_by_id(NULL, obj_id));
    lv_obj_create(active_screen);
    LV_LOG_USER("%p", lv_obj_get_child_by_id(NULL, obj_id));
}
```

Prints `0x556a78603cd0`, `(nil)`, `(nil)`, which is correct. Only checks one screen. Need to loop over all displays+screens to check those.

### Proposal

Maybe there could be a long bit array of every obj ID and the bit indicates whether the obj is alive? 8000 objs would be a 1kb array. I think that's a bad idea. Just brainstorming.

### Summary

The current `lv_obj_is_valid` is slow and unreliable. Using `class_p` is a solution for performance but not reliability. Using obj ID is a solution for reliability but not performance.

kisvegabor, 2024-06-25:
In Liam's example `lv_obj_is_valid` is actually working well. The stored pointer really points to a valid widget. It might not be widget that you want, but it's valid. It's similar to:
```
obj1 = lv_button_create(scr);
obj2 = lv_button_create(scr);
lv_obj_is_valid(obj1);  //true
lv_obj_is_valid(obj2);  //true
obj1 = obj2 ;
lv_obj_is_valid(obj1);  //true
```
How could `lv_obj_is_valid()` know if the pointer is the widget you mean? 

(We could speed it up by storing some magic number in the widgets which is cleared on delete.) 

However if we want to `NULL` a variable on delete we can add a function like
```c
lv_obj_add_event_cb(obj, my_delete_cb, LV_EVENT_DELETE, &obj); //In the callback *obj = NULL;
```

So I think it's 2 different use cases:
1. Is this the widget that I mean? We don't know that, but we can tell is it's a valid widget. 
2. Automatically NULL my variable when the widget is deleted.

niklasf, 2024-06-25:
I think most current issues with invalid object pointers come from mixed ownership. Screens are owned by the application logic. Other objects are owned by their parent object, but the application might want to retain valid references to them.

A common solution to guarantee that references at least keep pointing to valid memory is reference counting. It could also make bindings to languages with automatic destructors (like C++ and Rust) more natural. Has that been considered for lvgl?

liamHowatt, 2024-06-25:
Okay I think I understand the use case of `lv_obj_is_valid`. For @bhspyder's SquareLine example, there is a collection of obj pointers that is being pruned using `lv_obj_is_valid`? I can appreciate that. If it's pruning the list of obj pointers that are certainly invalid, that makes sense (although, how does it know the styles apply to the valid ones?). But for [this](https://github.com/lvgl/lvgl/blob/cbea72dc01b90766b7ddef32172776468fc11988/src/others/ime/lv_ime_pinyin.c#L595) I think it's risky to call the delete function on something that might not be `kb`.

We can drop the discussion about a function that checks if your obj is still alive. I understand now. kisvegabor gave the solution to the alive problem anyways (a pointer to the obj pointer given to a delete callback).

For performance of `lv_obj_is_valid`, kisvegabor's magic number idea seems solid to me. Would a magic 32 bits be unique enough in the memory soup? I think so. Do we want to open a PR for it?

RE reference counting:
Is an object usable after its parent removes it from its array of children? I'm not sure whether ref counting would be helpful for obj, unless the refcount saves it from being deleted, and then the user assigns it to a new parent.

xlbao123, 2024-06-26:
I'm a bit curious, in what scenarios must this api be used? It looks like this is only for debugging, and it seems like doing this in the system's memory management module would be better.

kisvegabor, 2024-06-27:
@niklasf 
> A common solution to guarantee that references at least keep pointing to valid memory is reference counting. It could also make bindings to languages with automatic destructors (like C++ and Rust) more natural. Has that been considered for lvgl?

Imagine that you have a complex UI and you store pointer for the screen and label somewhere. With reference counting when the screen is deleted The label won't be deleted as it's referenced somewhere in the user's code. I think it would be counter intuitive.

> But for [this](https://github.com/lvgl/lvgl/blob/cbea72dc01b90766b7ddef32172776468fc11988/src/others/ime/lv_ime_pinyin.c#L595) I think it's risky to call the delete function on something that might not be kb.

Here adding a DELETED event to `pinyin_ime->kb` would be more appropriate and NULL `pinyin_ime->kb` in it.

> Would a magic 32 bits be unique enough in the memory soup?

It will fail once in every 4 billion cases :slightly_smiling_face: 

---

All things considered I think we can add
- `lv_obj_null_on_delete(lv_obj_t ** obj)`
- magic number to `lv_obj_is_valid()`

niklasf, 2024-06-27:
A dangling object is not very useful, but it would guarantee valid memory that belongs to the exact object. So all operations on it would still have well-defined and predictable behavior. You could reliably and efficiently check if the exact same object is indeed dangling without any external tracking, reliably get the text that had been entered into the textarea, etc., or even reattach the object at some point (like dangling nodes into a browser's DOM tree).

That said, I am also reasonable happy with using DELETED events to null managed pointers. Together with null checks that also guarantees well-defined predictable behavior, i.e., crashes that are easy to debug.

I would not (exclusively) rely on magic numbers, as all the statistics break down when a (potentially even malicious) user can influence the memory soup, even just by normal inputs.

liamHowatt, 2024-06-27:
Can I please challenge someone to come up with an example use case of `lv_obj_is_valid` so I can understand its true utility? Even in the SquareLine case, after it prunes the list, can it really make any assumptions about what's left over in the list? Can't any use of `lv_obj_is_valid` be enhanced in performance and reliability by always using `lv_obj_null_on_delete`?

Something to watch out for with `lv_obj_null_on_delete` is where the pointer lives. The `lv_obj_t **` passed to `lv_obj_null_on_delete` must not point to an element in an `lv_array` because references to array members become invalid if the array resizes. That's the only big gotcha I can think of.

kisvegabor, 2024-06-28:
> ... reference counting ...

Even if we said that it's a good design I can't really imagine how to implement it in C without too many boiler plate on the user's side.
 
> I would not (exclusively) rely on magic numbers, as all the statistics break down when a (potentially even malicious) user can influence the memory soup, for example, simply by making normal inputs.

We might be able to find something in between. Slower than a magic number but faster then the current solution. However, if the pointer being tested is a memory garbage we cannot do anything with it. We cannot get its class or parent, as it might immediately result in an address exception. So all I could figure out is starting from the other was around: check if someone knows this pointer. 

> Can I please challenge someone to come up with an example use case of lv_obj_is_valid so I can understand its true utility?

From my side it's only for debugging. It already helped a lot to find issues when referencing deleted widgets. It might not be perfect, but if it catches 99 errors from 100, it's definitely better than nothing.

I can't really comment about the SquareLine part. The related SquareLine feature was implemented when I left. I remember that I was recommending a different solution to the whole theming issue but I don't know what the result ended up being. Maybe @leidto or someone else from the SquareLine team can tell more about it. 

> The lv_obj_t ** passed to lv_obj_null_on_delete must not point to an element in an lv_array because references to array members become invalid if the array resizes. That's the only big gotcha I can think of.

:+1:

closed 2024-07-24 by lvgl-bot