lvgl/lvgl #6396: toutch freeze gui

Red-Owl, 2024-06-19:
### LVGL version

8.4 and 8.3.1

### What happened?

after a random period of 2/3 hours touching it will freeze
serial. prind add after for testing
tft non freeze https://github.com/lovyan03/LovyanGFX/issues/580

[platformio]
default_envs = wt32-sc01-plus

[env:wt32-sc01-plus]
platform = espressif32 @ 6.7.0
board = esp32-s3-devkitm-1
framework = arduino
upload_speed = 115200
lib_deps =
   mathieucarbou/ESP Async WebServer@^2.10.5
   lvgl/lvgl@^8.3.1
   lovyan03/LovyanGFX@1.1.16
   bblanchon/ArduinoJson@^6.21.2
   paulstoffregen/Time@^1.6.1
   fbiego/ESP32Time@^2.0.0
lib_extra_dirs =
	lib/async_web_server
	lib/ArduinoMD5
build_flags =
   -DBOARD_HAS_PSRAM
monitor_filters = esp32_exception_decoder

### How to reproduce?

uint16_t touchX, touchY;
void my_touchpad_read_gatto(lv_indev_drv_t *indev_driver, lv_indev_data_t *data){
  
  boolean touched = tft.getTouch(&touchX, &touchY,0);
  // tft.getCursorX();
  //  tft.getCursorY();
  Serial.print("Toro");

  if (!touched){
    data->state = LV_INDEV_STATE_REL;
    Serial.print("Toro1");
  }else{
    data->state = LV_INDEV_STATE_PR;
    data->point.x = touchX;
    data->point.y = touchY;
    nanna_nanni = 0;
    nanna_exec = true;
    touch++;
    Serial.print("Toro2");
  }

  Serial.println(touch);

}

void setup_screen_lvgl(){
  tft.init();
  tft.initDMA();
  tft.startWrite();
  tft.setBrightness(70);
  lv_init();
  lv_disp_draw_buf_init(&draw_buf, disp_draw_buf, disp_draw_buf2, screenWidth * SCR);/* Initialize the display */
  lv_disp_drv_init(&disp_drv);
  disp_drv.hor_res = screenWidth;/* Change the following line to your display resolution */
  disp_drv.ver_res = screenHeight;
  disp_drv.flush_cb = my_disp_flush;
  disp_drv.draw_buf = &draw_buf;
  lv_disp_t * disp = lv_disp_drv_register(&disp_drv);
  /* Initialize the input device driver */
  static lv_indev_drv_t indev_drv;
  lv_indev_drv_init(&indev_drv);
  indev_drv.type = LV_INDEV_TYPE_POINTER;
  indev_drv.read_cb = my_touchpad_read_gatto;
  lv_indev_drv_register(&indev_drv);
  lv_color_t c = lv_color_make(125, 125, 255);
  lv_disp_set_bg_color(disp,c);
}

kisvegabor, 2024-06-20:
It's unlikely that  it's an LVGL issue as rendering still works. 

Which `Serial.print`s do you see in `my_touchpad_read_gatto`?

Red-Owl, 2024-06-20:
yes works !! test for 4 hours .... touch and crash

kisvegabor, 2024-06-21:
Could you answer this?
> Which Serial.prints do you see in my_touchpad_read_gatto?

closed 2024-07-14 by lvgl-bot