lvgl/lvgl #6407: Assertion on lv_obj_send_event when using lv_screen_load_anim

andersonrs66, 2024-06-21:
### LVGL version

v9.1.0-253-ga2883e30f

### What happened?

There is a NULL object assertion happening when using screen transitions with auto_del == true

### How to reproduce?

If you generate any event on the screen during the fade animation, this assertion will be triggered:

[Error] (1.181, +1181)   lv_obj_send_event: Asserted at expression: lv_obj_has_class(obj, (&lv_obj_class)) == true (Incompatible object type.) lv_obj_event.c:49

```
/* using SDL simulator */
int main(int argc, char **argv)
{
    (void)argc; /*Unused*/
    (void)argv; /*Unused*/

    /*Initialize LVGL*/
    lv_init();

    /*Initialize the HAL (display, input devices, tick) for LVGL*/
    hal_init(800, 480);

    lv_obj_t *scr = lv_obj_create(NULL);
    // change bg color to see fade animation
    lv_obj_set_style_bg_color(scr, lv_palette_main(LV_PALETTE_BLUE_GREY), 0);
    lv_screen_load_anim(scr, LV_SCR_LOAD_ANIM_FADE_OUT, 1000, 0, true);

    while (1)
    {
        static int count = 0;
        /* Periodically call the lv_task handler.
         * It could be done in a timer interrupt or an OS task too.*/
        lv_timer_handler();
        usleep(5 * 1000);

    }

    return 0;
}
```

kisvegabor, 2024-06-24:
Hi,

Thank you for reporting this issue, however I can't see where does the attached code send any events during the transition. 

Could you clarify it?

andersonrs66, 2024-06-24:
Hello,

The code itself doesn't send any event. However, cursor movement or a click event will generate an event and cause this error.

sorry for not being clear on how to reproduce the error. This is the sequence I did to reproduce:

1. Run the example code on the simulator
2. During the first screen transition, move the cursor or click anywhere on the screen
3. The error message will appear in the terminal
4. The simulator will freeze.

I also noticed that the bug only occurs on the first screen transition. Subsequent transitions pause cursor movement and do not generate this error.

niklasf, 2024-06-24:
I'll mention it here just in case: https://github.com/lvgl/lvgl/commit/eff59f7bba4486b3a2da97b1078eee0465d3cc8d fixes very similar symptoms, so it might be worth trying with the very latest lvgl master.

andersonrs66, 2024-06-24:
Great! @niklasf 
tested here and works fine, thanks!

closed 2024-06-24 by andersonrs66