lvgl/lvgl #6410: Draw arc not working if LV_DRAW_SDL is enabled.

sliang951753, 2024-06-22:
### LVGL version

9.1.0

### What happened?

compile LVGL with SDL display driver and enable the LV_DRAW_SDL option.
run the lv_example_arc_1() or other demos which contains arc drawing.
The arc will not be draw on the screen.

### How to reproduce?

Here's the lv_conf.h 

`/**
 * @file lv_conf.h
 * Configuration file for v9.1.0
 */

/*
 * Copy this file as `lv_conf.h`
 * 1. simply next to the `lvgl` folder
 * 2. or any other places and
 *    - define `LV_CONF_INCLUDE_SIMPLE`
 *    - add the path as include path
 */

/* clang-format off */
#if 1 /*Set it to "1" to enable content*/

#ifndef LV_CONF_H
#define LV_CONF_H

/*If you need to include anything here, do it inside the `__ASSEMBLY__` guard */
#if  0 && defined(__ASSEMBLY__)
#include "my_include.h"
#endif

/*====================
   COLOR SETTINGS
 *====================*/

/*Color depth: 8 (A8), 16 (RGB565), 24 (RGB888), 32 (XRGB8888)*/
#define LV_COLOR_DEPTH 16

/*=========================
   STDLIB WRAPPER SETTINGS
 *=========================*/

/* Possible values
 * - LV_STDLIB_BUILTIN:     LVGL's built in implementation
 * - LV_STDLIB_CLIB:        Standard C functions, like malloc, strlen, etc
 * - LV_STDLIB_MICROPYTHON: MicroPython implementation
 * - LV_STDLIB_RTTHREAD:    RT-Thread implementation
 * - LV_STDLIB_CUSTOM:      Implement the functions externally
 */
#define LV_USE_STDLIB_MALLOC    LV_STDLIB_BUILTIN
#define LV_USE_STDLIB_STRING    LV_STDLIB_BUILTIN
#define LV_USE_STDLIB_SPRINTF   LV_STDLIB_BUILTIN

#define LV_STDINT_INCLUDE       <stdint.h>
#define LV_STDDEF_INCLUDE       <stddef.h>
#define LV_STDBOOL_INCLUDE      <stdbool.h>
#define LV_INTTYPES_INCLUDE     <inttypes.h>
#define LV_LIMITS_INCLUDE       <limits.h>
#define LV_STDARG_INCLUDE       <stdarg.h>

#if LV_USE_STDLIB_MALLOC == LV_STDLIB_BUILTIN
    /*Size of the memory available for `lv_malloc()` in bytes (>= 2kB)*/
    #define LV_MEM_SIZE (512 * 1024U)          /*[bytes]*/
[... 922 lines cut ...]
    #define LV_DEMO_MUSIC_ROUND     0
    #define LV_DEMO_MUSIC_LARGE     0
    #define LV_DEMO_MUSIC_AUTO_PLAY 0
#endif

/*Flex layout demo*/
#define LV_USE_DEMO_FLEX_LAYOUT     0

/*Smart-phone like multi-language demo*/
#define LV_USE_DEMO_MULTILANG       0

/*Widget transformation demo*/
#define LV_USE_DEMO_TRANSFORM       0

/*Demonstrate scroll settings*/
#define LV_USE_DEMO_SCROLL          0

/*Vector graphic demo*/
#define LV_USE_DEMO_VECTOR_GRAPHIC  0
/*--END OF LV_CONF_H--*/

#endif /*LV_CONF_H*/

#endif /*End of "Content enable"*/
`

kisvegabor, 2024-06-25:
It seems it's really not supported yet, however it's straightforward to add. Could you try adding it yourself similarly to [these] (https://github.com/lvgl/lvgl/blob/master/src/draw/sdl/lv_draw_sdl.c#L218-L225)?

kisvegabor, 2024-06-28:
I've stated to work on it in https://github.com/lvgl/lvgl/pull/6437

PR #6437 (2024-08-26): feat(sdl_render): support all draw task types
Fixes #6410, 

### Description of the feature or fix

I'm working on adding support to all draw task types, but I'm facing a cache related issue. The current code crashes in `lv_example_buttonmatrix_1()` if [this line](https://github.com/lvgl/lvgl/pull/6437/files#diff-966ce7c60f64975670503d16064d42d2bf308ef75dbdd658bb96f4c1fcff782aR462) is enabled:
```
==34516==ERROR: AddressSanitizer: heap-use-after-free on address 0x60600001e100 at pc 0x559d6a5c9864 bp 0x7ffeda3e96c0 sp 0x7ffeda3e96b0
READ of size 4 at 0x60600001e100 thread T0
    #0 0x559d6a5c98 [... cut]
closed 2024-08-26 by kisvegabor