lvgl/lvgl #6411: How to use rounder_cb in LVGL9.1

jie326513988, 2024-06-22:
### Introduce the problem

I have an 8-color fully reflective screen. I need to update a whole row each time. The X axis needs to start from 0. The update of some widgets will not start from 0.
But there seems to be no rounder_cb related function available in LVGL9.1?

Acknowledgements

liamHowatt, 2024-06-26:
Have you considered using `LV_DISPLAY_RENDER_MODE_DIRECT` and keeping track of the affected lines in your `flush_cb` until `lv_display_flush_is_last()` is `true` and then drawing those full lines?

kisvegabor, 2024-06-27:
You need the `LV_EVENT_INVALIDATE_AREA` event of the display. See the description [here](https://docs.lvgl.io/master/porting/display.html#events).

jie326513988, 2024-06-30:
> 您需要显示的事件。请参阅[此处](https://docs.lvgl.io/master/porting/display.html#events)的说明。`LV_EVENT_INVALIDATE_AREA`

OK, thank you very much

closed 2024-06-30 by jie326513988