lvgl/lvgl #6413: lv_event.c void function have a return code

Red-Owl, 2024-06-23:
### LVGL version

9

### What happened?

void * lv_event_get_target(lv_event_t * e)
{
    return e->original_target;
}

### How to reproduce?

void function have a return code  (multiple function)
lv_event.c

C47D, 2024-06-23:
The function returns a "generic" pointer. Is this causing any warnings when compiling?

Red-Owl, 2024-06-23:
this create error on 
https://docs.lvgl.io/master/widgets/dropdown.html

    lv_event_code_t code = lv_event_get_code(e);
    lv_obj_t * obj = lv_event_get_target(e); // obj not exists

C47D, 2024-06-25:
@kisvegabor When trying to replicate this "issue" I noticed the following behavior in `lv_example_dropdown_3`, is the selected item text not changing expected?

[Screencast from 24-06-24 19:37:14.webm](https://github.com/lvgl/lvgl/assets/11037705/52fc0bf7-c0df-4cf2-81cd-8346a98339de)

C47D, 2024-06-25:
> this create error on https://docs.lvgl.io/master/widgets/dropdown.html
> 
> ```
> lv_event_code_t code = lv_event_get_code(e);
> lv_obj_t * obj = lv_event_get_target(e); // obj not exists
> ```

What error are you seeing? I was able to compile and run all three dropdown examples with no errors. Also, which example are you refering to?

XuNeo, 2024-06-26:
You need to disable warning of `fpermissive ` or use pure C code.

There are lots of conversion between `void *` pointer in lvgl, thus it won't fix.

Red-Owl, 2024-06-26:
```
workaround to 

char COMMAND_BUF[32];               //SC4 buffer caratteri di COMMAND_LIST sc4
lv_obj_t * COMMAND_LIST;            //SC4 lista dropdown comandi
const char *US_LIST = "\n"
                      "settime\n"
                      "setip\n"
                      "setsubnet\n"
                      "setgw\n"
                      "setpdns\n"
                      "setch\n"
                      "setssid\n"
                      "setpwd\n"
                      "setlang\n"
                      "secount\n"
                      "duty\n"
                      "frequency\n"
                      "save";
static void us_command_list_event(lv_event_t * e)
{
    lv_event_code_t code = lv_event_get_code(e);

    if(code == LV_EVENT_VALUE_CHANGED) {
        COMMAND_BUF[32] = {};
        lv_dropdown_get_selected_str(COMMAND_LIST, COMMAND_BUF, sizeof(COMMAND_BUF));
        Serial.print(COMMAND_BUF);
    }
}

void us_command_list(lv_obj_t * cnt)
{

    /*Create a normal drop down list*/
    COMMAND_LIST = lv_dropdown_create(cnt);
    lv_dropdown_set_options(COMMAND_LIST, US_LIST);

    lv_obj_set_pos(COMMAND_LIST, 5, 20);   // Set its position 
    lv_obj_set_size(COMMAND_LIST, 140, 40);   // Set its size 

    lv_obj_add_event_cb(COMMAND_LIST, us_command_list_event, LV_EVENT_ALL, NULL);
}

```

kisvegabor, 2024-06-28:
@C47D 
> When trying to replicate this "issue" I noticed the following behavior in lv_example_dropdown_3, is the selected item text not changing expected?

Now it's really true that it's feature not a bug. :smile: This example show the effect of `lv_dropdown_set_text(dropdown, "Menu");`

> You need to disable warning of fpermissive  or use pure C code.
There are lots of conversion between void * pointer in lvgl, thus it won't fix.

You can use `lv_event_get_target_obj` too.

closed 2024-07-23 by lvgl-bot