lvgl/lvgl #6414: lv_timer_handler hangs when running on separate thread on ESP32 using FREERTOS using Arduino framework

Jvsjvs, 2024-06-23:
### LVGL version

9.1.0

### What happened?

When I call lv_timer_handler from the main loop (with or without a mutex protection everything works.

When I create a separate thread using xTaskCreatePinnedToCore (on either core) with or without a mutex protection and this task calls the method lv_timer_handler this method never returns.

### How to reproduce?

//This is the task definition:
void ProcessDisplay(void *arg)
{
    ESP_LOGD(TAG, "Starting LVGL task");

    uint32_t task_delay_ms = LVGL_PORT_TASK_MAX_DELAY_MS;
    while (1) {
        lv_mutex_lock(&lvgl_mutex);
        task_delay_ms = lv_timer_handler(); 
        lv_mutex_unlock(&lvgl_mutex);
        if (task_delay_ms > LVGL_PORT_TASK_MAX_DELAY_MS) {
            task_delay_ms = LVGL_PORT_TASK_MAX_DELAY_MS;
        } else if (task_delay_ms < LVGL_PORT_TASK_MIN_DELAY_MS) {
            task_delay_ms = LVGL_PORT_TASK_MIN_DELAY_MS;
        }
        vTaskDelay(pdMS_TO_TICKS(task_delay_ms));
    }
}

// Mutex is created like this:
lv_mutex_init(&lvgl_mutex);                         // Make lvgl thread safe

// This is how the task is created
BaseType_t ret = xTaskCreatePinnedToCore(ProcessDisplay, "lvgl", LVGL_PORT_TASK_STACK_SIZE, NULL,
                                                                               LVGL_PORT_TASK_PRIORITY, &lvgl_task_handle, core_id);
// where stack size is 6k bytes, the piority is 2

kisvegabor, 2024-06-28:
> Should I not set this LV_USE_OS to LV_OS_FREERTOS when using ESP32 on framwork Arduino which is running on top of the FreeRTOS???

It should work with Arduino + ESP too. Please add this in `lv_freertos.c` right after the includes:
```c

#undef USE_FREERTOS_TASK_NOTIFY
#define USE_FREERTOS_TASK_NOTIFY 0 //Try with 1 too
```

Jvsjvs, 2024-06-28:
Thanks for your suggestions, I tried them out with these results:

Adding the 2 lines with USE_FREERTOS_TASK_NOTIFY 0 --> many build issues, FreeRTOS does not compile successfully (see below some build output).

Adding the 2 lines with USE_FREERTOS_TASK_NOTIFY 1 --> exact same behavior as without these 2 lines, the system hangs.

So unfortunately this does not solve the issue, do you have other ideas I could try?

----------- Example of build errors ----------

Compiling .pio/build/esp32-s3-devkitm-1/lib158/lvgl/osal/lv_freertos.c.o
In file included from .pio/libdeps/esp32-s3-devkitm-1/lvgl/src/osal/lv_freertos.h:29,
                 from .pio/libdeps/esp32-s3-devkitm-1/lvgl/src/osal/lv_os.h:30,
                 from .pio/libdeps/esp32-s3-devkitm-1/lvgl/src/osal/lv_freertos.c:15:
.pio/libdeps/esp32-s3-devkitm-1/lvgl/src/osal/lv_freertos.c: In function 'lv_thread_sync_wait':
.pio/libdeps/esp32-s3-devkitm-1/lvgl/src/osal/lv_freertos.c:211:26: error: 'lv_thread_sync_t' {aka 'struct <anonymous>'} has no member named 'xSyncMutex'
     xSemaphoreTake(pxCond->xSyncMutex, portMAX_DELAY);
                          ^~
/home/john/.platformio/packages/framework-arduinoespressif32/tools/sdk/esp32s3/include/freertos/include/freertos/semphr.h:312:76: note: in definition of macro 'xSemaphoreTake'
 #define xSemaphoreTake( xSemaphore, xBlockTime )    xQueueSemaphoreTake( ( xSemaphore ), ( xBlockTime ) )
                                                                            ^~~~~~~~~~
.pio/libdeps/esp32-s3-devkitm-1/lvgl/src/osal/lv_freertos.c:213:18: error: 'lv_thread_sync_t' {aka 'struct <anonymous>'} has no member named 'xSyncSignal'
     while(!pxCond->xSyncSignal) {
                  ^~
.pio/libdeps/esp32-s3-devkitm-1/lvgl/src/osal/lv_freertos.c:219:61: error: 'lv_thread_sync_t' {aka 'struct <anonymous>'} has no member named 'ulWaitingThreads'
         ulLocalWaitingThreads = Atomic_Increment_u32(&pxCond->ulWaitingThreads);

Jvsjvs, 2024-07-04:
Hi,

Thanks for your time thus far on this topic, is there any chance that it can be resolved, did you get any more insights based on the suggested experiment?

It is hanging on: lv_thread_sync_wait(&_draw_info.sync); in the lv_draw.c

WIth: "#define LV_USE_OS   LV_OS_NONE" it works but I see that then the lv_mutex_init, lv_mutex_lock etc are defined as "empty" doing nothing. Hence I can't protect LVGL against multi-threading access.

The result is that it is not possible to use LVGL in a safe way in a multi-threading environment unless this "hang" is resolved.

Any suggestions I can try?

Thanks,

John

Jvsjvs, 2024-07-05:
This is a stack trace of when it is hanging in lv_result_t lv_thread_sync_wait(lv_thread_sync_t * pxCond)
it seems to be waiting to get notified which never happens.

[image]

mdkendall, 2024-07-06:
I am also seeing this issue.

LVGL 9.1 running on STM32 under FreeRTOS, using code based very closely on the example in the docs:
[https://docs.lvgl.io/master/integration/driver/display/lcd_stm32_guide.html](https://docs.lvgl.io/master/integration/driver/display/lcd_stm32_guide.html)

There is plenty of memory, and no task stacks have overflowed.

After initialisation, I call `lv_task_handler` in a loop inside a task. The very first time I call it, it never returns. The stack trace shows it gets as far as `lv_draw_dispatch_wait_for_request` in `lv_draw.c`, which calls `lv_thread_sync_wait` in `lv_freertos.c`, which pends forever on a notification which never comes.

[image]

If I change `USE_FREERTOS_TASK_NOTIFY` from `1` to `0`, the behaviour is the same except `lv_thread_sync_wait` pends on a semaphore which is never given.

Jvsjvs, 2024-07-06:
I have looked at the LVGL sources and **found the problem** and **the solution.**

The problem occurs when the lv_init() is called on a different thread than the lv_timer_handler() and in the lv-conf-h the FreeRTOS is enabled using: "#define LV_USE_OS   LV_OS_FREERTOS". 

The **problem** is that:
- The first thread calls lv_init which calls lv_draw_init which calls lv_thread_sync_init which initializes pxCond as the thread to be notified using: "pxCond->xTaskToNotify = xTaskGetCurrentTaskHandle();"
- This initialization then store the the first thread in pxCond
- If you then call lc_timer_handler on a second thread it runs into lv_thread_sync_wait() which is waiting for the notification coming from: "lv_thread_sync_signal(lv_thread_sync_t * pxCond)"
- Now the: "lv_thread_sync_signal(lv_thread_sync_t * pxCond)" signals the wrong thread (thread 1 instead of thread 2) and hence thread2 is blocked indefinitely.

The **solution** (without changing LVGL) is to make sure that  the lv_init() is called on the thread which will eventually also call lv_timer_hander. **This has to be added to the documentation of LVGL**

Another solution (by chaning LVGL) is e.g. to only initialize the pxCond in the first call to either: "lv_thread_sync_signal(lv_thread_sync_t * pxCond)" or: "lv_thread_sync_wait".

So for now I solved it by:
1) I have a SetupDisplay function which (in my case) initializes ESP32_Display_Panel or any other display driver
2) Allocates the display buffers
3 Creates the thread that will  do the ProcessDisplay calls in a loop (which will call: "lv_timer_handler call").
4) This ProcessDisplay function starts with initializing LVGL and then run into the infinite loop.

The result is that lv_init and lv_timer_handler are called on the same (dedicated LVGL) thread, this works :-)

My code for your reference:

```
/**
 * @brief Initializes and sets up the display panel and LVGL.
 * 
 * This function initializes the display panel.  It creates a display object, sets up the display buffer,
 * and creates the LVGL task which will do the deferred initialization of LVGL.
 * 
 * @note This function should be called before using any LVGL functions or rendering on the display.
 */
void SetupDisplay()
{
    // Initialize the display panel
    panel = new ESP_Panel();
    panel->init();
    panel->begin();

    drawBuffer = heap_caps_malloc(LV_BUF_SIZE, MALLOC_CAP_DMA | MALLOC_CAP_INTERNAL);

    // Setup the LVGL task which will iniitialize LVGL
    BaseType_t core_id = (LVGL_PORT_TASK_CORE < 0) ? tskNO_AFFINITY : LVGL_PORT_TASK_CORE;
    BaseType_t ret = xTaskCreatePinnedToCore(ProcessDisplay, "lvgl", LVGL_PORT_TASK_STACK_SIZE, NULL,
                                             LVGL_PORT_TASK_PRIORITY, &lvgl_task_handle, core_id);
}

```

The above is called within the Arduino Setup function on the initial thread.

The ProcessDisplay looks like:

```
/**
 * @brief Function to process the display using LVGL library.
 *
 * This function is responsible for initializing and handling the LVGL task.
 * It initializes LVGL such that this library is intizliased on the correct LVGL task
 * It locks the LVGL mutex, calls the LVGL timer handler,
 * and unlocks the mutex. It also adjusts the task delay based on the timer handler's return value.
 * The task delay is then used to delay the task execution using vTaskDelay function.
 *
 * @param arg Pointer to the argument (not used in this function).
 */
void ProcessDisplay(void *arg)
{
    ESP_LOGD(TAG, "Starting LVGL task");

    SetupDisplayOnLVGLThread();

    uint32_t task_delay_ms = LVGL_PORT_TASK_MAX_DELAY_MS;    
    while (1) {
        lv_lock();
        task_delay_ms = lv_timer_handler(); 
        lv_unlock();
        if (task_delay_ms > LVGL_PORT_TASK_MAX_DELAY_MS) {
            task_delay_ms = LVGL_PORT_TASK_MAX_DELAY_MS;
        } else if (task_delay_ms < LVGL_PORT_TASK_MIN_DELAY_MS) {
            task_delay_ms = LVGL_PORT_TASK_MIN_DELAY_MS;
        }
        vTaskDelay(pdMS_TO_TICKS(task_delay_ms));
    }
}

```

XuNeo, 2024-07-06:
Glad you have solved it out.

Please do note LVGL is NOT thread safe. Which means you have to call ALL LVGL APIs from single thread, unless you do proper protection across threads.

See
https://docs.lvgl.io/master/porting/os.html

Jvsjvs, 2024-07-06:
Sure, LVGL is not thread safe. But protecting it with a mutex (which I did of course) is not enough, then you still have this problem. It is not about thread safety at all, it is about notifying the correct thread.

kisvegabor, 2024-07-11:
We have just faced this problem with the ESP team. It's fantastic that you have found [the solution](https://github.com/lvgl/lvgl/issues/6414#issuecomment-2211741353). :tada: 

We are trying it out! cc @liamHowatt @espzav

Jvsjvs, 2024-07-11:
Thanks Gabor, thanks for the great library. 

Please note that this problem goes beyond ESP32 architecture. The problem occurs for any architecture that uses FreeRTOS. 

I did not look into the implementations of the other OS options, I think it is worthwhile to also look whether they have a similar problem, I don't know how you implemented the synchronisation there.

Regards,

John.

Jvsjvs, 2024-07-12:
Note that the current documentation for version 9.1 (or master) is outdated.
- It still talks about using: "lv_lock()" and: "lv_unlock()" which both do not exist.
- It says that: "lv_timer_handler()" internally uses the same mutex as is used by lv_lock which is not the case.

I think many people will benefit if this whole section on multi-threading protection is made up-to-date including the restriction on using one and the same thread for lv_init and lv_timer_handler etc.

Thanks,

John.

kisvegabor, 2024-07-12:
Hi John,

`lv_lock` and `lv_unlock` exist in `master`. See [here](https://github.com/lvgl/lvgl/blob/master/src/osal/lv_os.c#L45).

Jvsjvs, 2024-07-14:
Hi Gabor,

I see it exists in main indeed, but ......

I downloaded version 9.1 (that is what I am using right now) which does not at all have the file: "src/osal/lv_os.c" and hence these methods are not defined, the prototype declarations are also not in: "src/osal/lv_os.h".

The documentation of version 9.1 on this (https://docs.lvgl.io/9.1/porting/os.html) does talk about lv_lock etc, so there is some inconsistency in version 9.1 on this.

I will move to master now where it is consistent,

Thanks,

John.

kisvegabor, 2024-07-15:
Oh, it's strange! I don't understand how ti made it's way to the v9.1 docs. Especially as it's not in the [docs source file](https://github.com/lvgl/lvgl/blob/release/v9.1/docs/porting/os.rst). 

@kdschlosser do you have any idea? In short [`os.rst` in the `release/v9.1`](https://github.com/lvgl/lvgl/blob/release/v9.1/docs/porting/os.rst) branch doesn't mention `lv_lock` but the compiled [docs for v9.1](https://docs.lvgl.io/9.1/porting/os.html#tasks-and-threads) does. Maybe the docs of `master` (`v9.1.1-dev`) ̇ docs ended up in the v9.1 docs?

kdschlosser, 2024-07-16:
yeah, it's because you guys don't use the patch number... Look at the link. It states version 9.1. So what happens is because there is a version 9.1.1 that is going to override version 9.1.0 due to not using the patch number.

kdschlosser, 2024-07-16:
The only way this has happened is if someone instructed the CI that the build was a release build and the documentation got pushed to the server. Other than that I do not know.. It would be wise to include the patch so this kind of thing would be a non issue.

kisvegabor, 2024-07-25:
Thanks! 

As in master we develop the next minor/major version the the correct current version should be `v9.2-dev`. I'll set it up accordingly in the next cycle and re-trigger v9.1.

Jvsjvs, 2024-08-10:
I assume this is being picked up and hence we can allow this thread to stall right?

liamHowatt, 2024-08-17:
Hi all,

kisvegabor's suggestion to redefine `USE_FREERTOS_TASK_NOTIFY` as `0` works without build issues and eliminates the problem if it is changed in `lv_freertos.h` where the macro is defined, instead of redefining it in `lv_freertos.c`.

To reiterate: in `lv_freertos.h`, change `#define USE_FREERTOS_TASK_NOTIFY 1` to `#define USE_FREERTOS_TASK_NOTIFY 0`.

This means the semaphores work fine as a slower alternative to the task notify system which has the flaw that Jvsjvs explained in detail. The wrong task is being notified.

We could update the docs to mention that `lv_init` needs to be called in the same thread as `lv_timer_handler` but I would like to see the problem fixed. The task notify system is much faster so removing it would not be a great solution. See benchmarks in https://github.com/lvgl/lvgl/pull/4362. The current implementation requires that the thread that calls `lv_thread_sync_init` is the thread that calls `lv_thread_sync_wait`. This requirement is imposed on all uses of thread_sync at the risk of it being the FreeRTOS task notify implementation. See [this code](https://github.com/lvgl/lvgl/blob/d7b82c79668d7b31d52f6535360eff15396eade5/src/draw/renesas/dave2d/lv_draw_dave2d.c#L404-L412) which has to be written that way in order to work, and has a slight race condition as a consequence.

I have opened a PR with a proposed fix. If it affects performance too much we can look at other possibilities or leave it as-is and just update the docs.

PR: #6692

Sidenote: users of the `esp-bsp` component [esp_lvgl_port](https://components.espressif.com/components/espressif/esp_lvgl_port/versions/2.3.1/changelog?language=en), such as [lv_port_espressif_esp32-s3-lcd-ev-board](https://github.com/lvgl/lv_port_espressif_esp32-s3-lcd-ev-board) will find the problem fixed thanks to a change included in this PR https://github.com/espressif/esp-bsp/pull/355 because `lv_init` was moved to the same task as `lv_timer_handler`

cc @nicusorcitu

nicusorcitu, 2024-08-19:
cc @cosmindanielradu19

PR #6692 (2024-08-26): fix(freertos): notifiy-based sync can be initialized from any task
### Description of the feature or fix

Fixes #6414

`lv_thread_sync_wait` can be called from a thread other than the one that called `lv_thread_sync_init`.

@nicusorcitu

### Notes
- Update the [Documentation](https://github.com/lvgl/lvgl/tree/master/docs) if needed.
- Add [Examples](https://github.com/lvgl/lvgl/tree/master/examples) if relevant.
- Add [Tests](https://github.com/lvgl/lvgl/blob/master/tests/README.md) if applicable.
- If you added new options to `lv_conf_template.h` run [lv_conf_internal_gen.py](https://github.c [... cut]