lvgl/lvgl #6415: I think there is an issue  when refreshing the display

kdschlosser, 2024-06-23:
### LVGL version

master

### What happened?

I am having an issue getting an RDB display to display anything.

I have been looking at the code for the refreshing of the display and it appears that if the performance monitor is enabled the refresh timer gets paused but there is a possibility of it not being resumed. The pause occurs here

https://github.com/lvgl/lvgl/blob/21a6a8bb87a89ded4644d8356abafc43e3fa4696/src/core/lv_refr.c#L334

This is where the timer gets resumed

https://github.com/lvgl/lvgl/blob/21a6a8bb87a89ded4644d8356abafc43e3fa4696/src/display/lv_display.c#L1092

You can see that the resume is done using the event `LV_EVENT_REFR_REQUEST`

Now if we go back to `lv_refr.c` we can see the event getting triggered here

https://github.com/lvgl/lvgl/blob/21a6a8bb87a89ded4644d8356abafc43e3fa4696/src/core/lv_refr.c#L281

and also here

https://github.com/lvgl/lvgl/blob/21a6a8bb87a89ded4644d8356abafc43e3fa4696/src/core/lv_refr.c#L303

but not here

https://github.com/lvgl/lvgl/blob/21a6a8bb87a89ded4644d8356abafc43e3fa4696/src/core/lv_refr.c#L286

or here

https://github.com/lvgl/lvgl/blob/21a6a8bb87a89ded4644d8356abafc43e3fa4696/src/core/lv_refr.c#L275

Now I know the display refreshing is a very complex process and I don't understand it fully. But when I see code like that you have to think that it's possible to have a scenario where the refresh timer doesn't get resumed. I am not sure how to go about testing whether or not there is a possibility of it not getting resumed. It seems like there is.

kdschlosser, 2024-06-23:
There is also what appears to be redundant code. 

There is no need to test to see what the event is which is seen here

https://github.com/lvgl/lvgl/blob/21a6a8bb87a89ded4644d8356abafc43e3fa4696/src/display/lv_display.c#L1091

when there is a filter that gets set when the callback gets registered.
https://github.com/lvgl/lvgl/blob/21a6a8bb87a89ded4644d8356abafc43e3fa4696/src/display/lv_display.c#L139

kisvegabor, 2024-06-28:
If I'm not mistaken in these cases it's not required to resume the timer as no new dirty area is added.
1. Around `LV_EVENT_INVALIDATE_AREA` if the event callback return `INVALID` the area will be dropped so nothing new will be rendered
2. Same for the "Out of the screen" part. If it's dropped no need to resume the timer as no new area will be added.

closed 2024-07-23 by lvgl-bot