lvgl/lvgl #6419: Online simulator broken

Viatorus, 2024-06-24:
### LVGL version

any

### What happened?

Going to any simulator page linked here: https://sim.lvgl.io/ shows a page like this (firefox/chrome/mobile):

[image]

kisvegabor, 2024-06-28:
Thank you for reporting this issue!

We have launched our new website a few weeks ago and I thought the issue might be related to it. However I see only this error in the Console which looks like something else.
[image]

@embeddedt do have a little bit of time to take a look at it?

embeddedt, 2024-06-28:
@kisvegabor The simulators relied on https://lvgl.io/assets/bootstrap/css/bootstrap.min.css and https://lvgl.io/assets/bootstrap/js/bootstrap.min.js, which appear to be missing after the site redesign. Can these assets be put back, or do we need to edit the existing simulators to use Bootstrap from another source?

kisvegabor, 2024-06-29:
Ah, thank you! As a hotfix I added bootstrap. However to avoid similar issues in the future, could we use [bootstrap from CDN](https://www.bootstrapcdn.com/)?

Viatorus, 2024-07-08:
The Save Button is also not working (at least on my side):

> ERROR: Network Error
> POST https://forum.lvgl.io:8443/save_file net::ERR_CONNECTION_RESET

kisvegabor, 2024-07-10:
I've fixed it, but saving/loading is very slow on my end.

embeddedt, 2024-09-11:
All of the simulators (including the pre-v8 ones) should now be updated to pull Bootstrap from cdn.js rather than relying on the lvgl.io copy.

closed 2024-09-11 by embeddedt

kisvegabor, 2024-09-11:
Thank you @embeddedt!