lvgl/lvgl #6421: Make Grids and Flex Layout Visible for Debugging

Ju1He1, 2024-06-25:
### Problem to solve

Hey guys,
we are developing our lvgl based UI within the Windows simulator.
When working with more complicated Grids and flex layouts (optionally nested) Debugging becomes pretty hard. Sometimes the UI does not look as what I have expected.

### Success criteria

For this reason It would be nice to visualise the boundaries of each cell within Debug builds.

### Solution outline

Specify an extra Debug Define within lv_config. When this define is set the border lines of flex and grid are visualised. This should help to get a better understanding why the layout does not match the expected layout

### Rabbit holes

Not sure if any

### Testing

create a test application and check if boundaries are visible.

### Teaching

document all defines within the lv_conf.h within a serate section in the docs folder.

### Considerations

none

liamHowatt, 2024-06-26:
Just thinking about how this could be done. I don't think this is the best way but this somewhat works:

In this case the obj edges were already visible. But if column and row padding was zero and the objs in the layout had no border or outline normally, it would effectively show the cell boundaries.

[image]

```diff
diff --git a/examples/layouts/flex/lv_example_flex_2.c b/examples/layouts/flex/lv_example_flex_2.c
index 7c5c3d040..16d4cd849 100644
--- a/examples/layouts/flex/lv_example_flex_2.c
+++ b/examples/layouts/flex/lv_example_flex_2.c
@@ -11,6 +11,7 @@ void lv_example_flex_2(void)
     lv_style_set_flex_flow(&style, LV_FLEX_FLOW_ROW_WRAP);
     lv_style_set_flex_main_place(&style, LV_FLEX_ALIGN_SPACE_EVENLY);
     lv_style_set_layout(&style, LV_LAYOUT_FLEX);
+    lv_style_set_outline_width(&style, 1);
 
     lv_obj_t * cont = lv_obj_create(lv_screen_active());
     lv_obj_set_size(cont, 300, 220);
diff --git a/src/misc/lv_style.c b/src/misc/lv_style.c
index e7278434e..803292434 100644
--- a/src/misc/lv_style.c
+++ b/src/misc/lv_style.c
@@ -88,7 +88,7 @@ const uint8_t _lv_style_builtin_prop_flag_lookup_table[_LV_STYLE_NUM_BUILT_IN_PR
     [LV_STYLE_BORDER_SIDE] = 0,
     [LV_STYLE_BORDER_POST] = 0,
 
-    [LV_STYLE_OUTLINE_WIDTH] =             LV_STYLE_PROP_FLAG_EXT_DRAW_UPDATE,
+    [LV_STYLE_OUTLINE_WIDTH] =             LV_STYLE_PROP_FLAG_EXT_DRAW_UPDATE | LV_STYLE_PROP_FLAG_INHERITABLE,
     [LV_STYLE_OUTLINE_COLOR] = 0,
     [LV_STYLE_OUTLINE_OPA] =               LV_STYLE_PROP_FLAG_EXT_DRAW_UPDATE,
     [LV_STYLE_OUTLINE_PAD] =               LV_STYLE_PROP_FLAG_EXT_DRAW_UPDATE,
```

kisvegabor, 2024-06-27:
It would be really amazing, and it would help me a lot too.  On some places I was already speaking about that we are developing our UI editor and this (and padding/margin highlights) already on our feature list. 

In light of that, I'm more interested in adding these to the editor where we have lot more options to visualize these ind of properties.  

I mean something like this:
[image]
[image]

cc @richardgazdik @coevin

Ju1He1, 2024-06-28:
Nice, looks like I'm not the only one struggling with these issues :)
Maybe some lvgl dev can add this in a later version :)

kisvegabor, 2024-07-04:
This week we had a many editor related meetings and we are confirmed that this is very important feature.

FYI, [this is the announcement](https://forum.lvgl.io/t/work-began-on-lvgls-ui-editor/17091) for the editor.

Ju1He1, 2024-07-05:
ah thx for your feedback. I'm looking forward to this :)

kisvegabor, 2024-07-10:
Me too 😀

Closing this issue now.

closed 2024-07-10 by kisvegabor