lvgl/lvgl #6422: My SSD1619 driver not working, how to fix it?

jecelyin, 2024-06-25:
### Introduce the problem

This my test code

```c
   static void lv_flush_callback(lv_disp_drv_t *disp_drv, const lv_area_t *area, lv_color_t *color_p)
    {
        int32_t w = (area->x2 - area->x1 + 1);
        int32_t h = (area->y2 - area->y1 + 1);

        epd.EPD_init_Part();
        // not working
        epd.writeImage((uint8_t *)color_p,  area->x1, area->y1, w, h);
        // not working
        // epd.EPD_Dis_Part(area->x1, area->x2, area->y1, area->y2, (uint8_t *)color_p, 1);
        epd.EPD_Update();

        Serial.println("refresh ok");
        /* IMPORTANT!!!
         * Inform LVGL that you are ready with the flushing and buf is not used anymore*/
        lv_disp_flush_ready(disp_drv);
    }

    static void init_epaper()
    {

        epd.EPD_init_Full();
        //   epd.EPD_Dis_Full((uint8_t *)gImage_BW, 1);
        epd.EPD_WhiteScreen_ALL_Clean(BUFFER_SIZE); 
        
        epd.EPD_init_Full();
        // below code is work
        epd.writeImage(gImage_BW, 0, 0, SCREEN_WIDTH, SCREEN_HEIGHT);
        epd.EPD_Update();
        epd.deepsleep();
}
```

File: EPaperDrive.cpp

```c
#include "EPaperDrive.h"
#include <lvgl.h>
// #include "EPD_drive_gpio.h"
typedef enum
{
  DOT_PIXEL_1X1 = 1, // 1 x 1
  DOT_PIXEL_2X2,     // 2 X 2
  DOT_PIXEL_3X3,     // 3 X 3
  DOT_PIXEL_4X4,     // 4 X 4
  DOT_PIXEL_5X5,     // 5 X 5
  DOT_PIXEL_6X6,     // 6 X 6
  DOT_PIXEL_7X7,     // 7 X 7
  DOT_PIXEL_8X8,     // 8 X 8
} DOT_PIXEL;
#define WHITE 0xFF // 白
#define BLACK 0x00 // 黑

typedef enum
{
  DOT_FILL_AROUND = 1, // dot pixel 1 x 1//点填充周围
  DOT_FILL_RIGHTUP,    // dot pixel 2 X 2//点填充
} DOT_STYLE;
#define DOT_STYLE_DFT DOT_FILL_AROUND // Default dot pilex//默认点桩

EPaperDrive::EPaperDrive(bool SPIMode, uint8_t CS, uint8_t RST, uint8_t DC, uint8_t BUSY, uint8_t CLK, uint8_t DIN)
{
  _CS = CS;
  _RST = RST;
  _DC = DC;
[... 806 lines cut ...]
  y -= _pw_y;
  // else y = HEIGHT - _pw_y - y - 1;
  // clip to (partial) window
  if ((x < 0) || (x >= int16_t(_pw_w)) || (y < 0) || (y >= int16_t(_pw_h)))
    return;
  // adjust for current page
  // y -= _current_page * _page_height;
  // check if in current page
  if ((y < 0) || (y >= int16_t(HEIGHT)))
    return;
  uint16_t i = x / 8 + y * (_pw_w / 8);
  if (color)
    _buffer[i] = (_buffer[i] | (1 << (7 - x % 8)));
  else
    _buffer[i] = (_buffer[i] & (0xFF ^ (1 << (7 - x % 8))));
}
```

epd.writeImage(gImage_BW, 0, 0, SCREEN_WIDTH, SCREEN_HEIGHT); of init_epaper

[image]

epd.writeImage((uint8_t *)color_p,  area->x1, area->y1, w, h); of lv_flush_callback

[image]

jecelyin, 2024-06-25:
It can't show label: "ABC abc xyz 123456 ght ijk" and button.

jecelyin, 2024-06-25:
my ui code:
```c

        static lv_style_t style_btn;

        lv_style_init(&style_btn);
        lv_style_set_radius(&style_btn, 5);
        lv_style_set_border_color(&style_btn, lv_color_black());
        lv_style_set_border_width(&style_btn, 1);
        lv_style_set_text_color(&style_btn, lv_color_black());

        lv_obj_t *label1 = lv_label_create(lv_scr_act());
        // lv_label_set_long_mode(label1, LV_LABEL_LONG_WRAP);     /*Break the long lines*/
        //  lv_obj_set_width(label1, 127);  /*Set smaller width to make the lines wrap*/
        // lv_obj_set_style_text_align(label1, LV_TEXT_ALIGN_CENTER, 0);
        lv_label_set_text(label1, "ABC abc xyz 123456 ght ijk");
        lv_obj_set_pos(label1, 0, 0);

        // Create button
        lv_obj_t *btn = lv_btn_create(lv_scr_act());
        lv_obj_set_pos(btn, 20, 20);
        lv_obj_set_size(btn, 50, 20);
        lv_obj_t *label = lv_label_create(btn);
        lv_label_set_text(label, "Button");
        lv_obj_center(label);
        lv_obj_add_style(btn, &style_btn, 0);
```

jecelyin, 2024-06-25:
```c
// lv_conf.h
#define LV_COLOR_DEPTH 1

/*Swap the 2 bytes of RGB565 color. Useful if the display has an 8-bit interface (e.g. SPI)*/
#define LV_COLOR_16_SWAP 0

/*Enable features to draw on transparent background.
 *It's required if opa, and transform_* style properties are used.
 *Can be also used if the UI is above another layer, e.g. an OSD menu or video player.*/
#define LV_COLOR_SCREEN_TRANSP 0
```

kisvegabor, 2024-06-26:
Note that with LV_COLOR_DEPTH 1 LVGL still renderers in 1 byte/px and not 1bit/px. Do you might need to convert rendered image.

closed 2024-07-23 by lvgl-bot