lvgl/lvgl #6428: Compile Error with Code Blocks 20.03 (solved)

bartholomaeuz, 2024-06-26:
### LVGL version

v9.1.0

### What happened?

I get an compile error from the following branch:
https://github.com/lvgl/lv_port_win_codeblocks

lvgl\src\drivers\windows\lv_windows_context.c|141|error: 'CREATE_WAITABLE_TIMER_MANUAL_RESET' undeclared (first use in this function)|
lvgl\src\drivers\windows\lv_windows_context.c|141|note: each undeclared identifier is reported only once for each function it appears in|
lvgl\src\drivers\windows\lv_windows_context.c|142|error: 'CREATE_WAITABLE_TIMER_HIGH_RESOLUTION' undeclared (first use in this function)|
||=== Build finished: 2 error(s), 4 warning(s) (2 minute(s), 1 second(s)) ===|

### How to reproduce?

add following lines to this file:
lvgl\src\drivers\windows\lv_windows_context.c

 #define CREATE_WAITABLE_TIMER_MANUAL_RESET 1
 #define CREATE_WAITABLE_TIMER_HIGH_RESOLUTION 2

kisvegabor, 2024-06-27:
Hi,

Thank you for reporting this issue! 

cc @MouriNaruto

MouriNaruto, 2024-06-27:
Got it. I will fix that in the recent days.

Kenji Mouri.

closed 2024-07-08 by FASTSHIFT

bluetur1, 2024-11-30:
It has not be fixed now.Thank you very much for your report. Be lucky to meet !
@heishuini

PR #6425 (2024-07-08): fix(Windows): use global lock
### Description of the feature or fix

Fixes #6059

Use the LVGL global lock in the Windows backend to control access to data and LVGL APIs shared with the Windows backend thread message callback.

Prevent memory faults, especially when the display is deleted.

CC @MouriNaruto

Fixes #6428

### Notes
- Update the [Documentation](https://github.com/lvgl/lvgl/tree/master/docs) if needed.
- Add [Examples](https://github.com/lvgl/lvgl/tree/master/examples) if relevant.
- Add [Tests](https://github.com/lvgl/lvgl/blob/master/tests/README.md) if applicable.
- If you adde [... cut]