lvgl/lvgl #6432: Hardware Rotation in Generic MIPI Display Driver

wasabinz, 2024-06-26:
### LVGL version

v9.1.0

### What happened?

While trying to rotate the display using the integrated display driver for ILI9341 with touch screen.

After calling `lv_display_set_rotation(NULL, LV_DISPLAY_ROTATION_90);`, which leads to 
https://github.com/lvgl/lvgl/blob/75eef0d209dfc96bf87d9d83b30eaee3e21b9f14/src/drivers/display/lcd/lv_lcd_generic_mipi.c#L296-L299

It causes the displayed image to turn 90 degrees clockwise.

At the same time, the input device translation is mapping anti-clockwise, causing mismatch with LCD in 90 and 270 degree settings.
https://github.com/lvgl/lvgl/blob/75eef0d209dfc96bf87d9d83b30eaee3e21b9f14/src/indev/lv_indev.c#L639-L647

Does `LV_DISPLAY_ROTATION_90`  means turning the image clockwise? or the display clockwise?

I couldnt find any mention of the direction of rotation in v9.1.0 documentation.

kisvegabor, 2024-06-28:
Hi, 

It really easy to mess up. Let's say you have a 320x240 display. You make the display controller to rotate it by 90° clockwise. So you will have a 240x320 display (left side rotated to the top). Now LVGL thinks that the display is 240x320 and renders an image accordingly. If you press a the touchpad, the coordinate needs to be rotated the **same way** as the image.  (left side to top).

```c
 if(disp->rotation == LV_DISPLAY_ROTATION_90 || disp->rotation == LV_DISPLAY_ROTATION_270) { 
     int32_t tmp = data->point.y; 
     data->point.y = data->point.x; 
     data->point.x = disp->ver_res - tmp - 1; 
 } 
```

Let's say x=5;y=100 (left middle) is clicked. It will be converted to `x=139;y=5` (top middle). Note that `disp->ver_res` doesn't change on rotation, so it's always 240 in this example.

So 90 means clock wise rotation.

But here comes the twist. With the SDL driver 90° rotation looks like this (needs `LV_SDL_RENDER_MODE      LV_DISPLAY_RENDER_MODE_PARTIAL`):
[image]

So it looks like counter-clock wise. However physically you need to rotate the display clock wise to make it look correct. 

Honestly I would be surprised if we messed it up. It was tested on several hardware, but there might be differences that we haven't considered.

Using the `LV_LCD_FLAG_MIRROR_X` and `LV_LCD_FLAG_MIRROR_Y` flags in `lv_lcd_generic_mipi_create` might help to align what LVGL and display thinks. 

@zjanosy can you joint the discussion?

wasabinz, 2024-06-28:
Hm... I swapped the switch cases for `LV_DISPLAY_ROTATION_90` and `LV_DISPLAY_ROTATION_270` in 
https://github.com/lvgl/lvgl/blob/75eef0d209dfc96bf87d9d83b30eaee3e21b9f14/src/drivers/display/lcd/lv_lcd_generic_mipi.c#L289
And that solves the issue.

Stepping through the code, `LV_DISPLAY_ROTATION_90` sets the MADCTL register to 0x60
[image]

Looking at the datasheet. That rotates the image clockwise 90
[image]

Could this be the problem?

kisvegabor, 2024-07-04:
It makes sense, let's go with your version. Can you open a PR?

kisvegabor, 2024-07-09:
https://github.com/lvgl/lvgl/pull/6470 is merged, closing.

closed 2024-07-09 by kisvegabor