lvgl/lvgl #6438: How can I using a callback function of events handle in a class?

Jack-a-new, 2024-06-27:
### Introduce the problem

while i encapsulate a class of SymbolButton, and than register a callback function. It comes an error as below.
error: cannot convert 'SymbolButton::btn_event_cb' from type 'void (SymbolButton::)(lv_event_t*)' {aka 'void (SymbolButton::)(_lv_event_t*)'} to type 'lv_event_cb_t' {aka 'void (*)(_lv_event_t*)'}

I use LVGL master version and it works on arduino and simulator of codeblocks. Here is the code. I wonder if anyone can help? Thanks!
class Base{
  private:
    lv_obj_t *obj;
  public:
    Base(){
      obj = lv_obj_create(lv_screen_active());
    }

    Base(lv_obj_t * father ){
      obj = father;
      ::lv_obj_set_size(obj, 320, 240);
      ::lv_obj_center(obj);
    }

    lv_obj_t * returnObj(){
      return obj;
    }
};

//  A button of a line of symbol and a line of text in it. It looks like an icon. 
class SymbolButton : public Base{
  private:
    lv_obj_t *btn;
    lv_obj_t *symbol;  //first line of label or symbol in button
    lv_obj_t *text;     //second line of label in button
    void (*event_cb)();
  public:
    SymbolButton(lv_obj_t *scr, const char *symbol, const char* text, void (*func)()):Base(scr){
      this->btn = lv_button_create(scr);
      this->symbol = lv_label_create(this->btn);
      lv_label_set_text(this->symbol, symbol);
      lv_obj_set_size(this->btn,85,55);
      if((text == "" )|| (text ==NULL)){::lv_obj_align(this->symbol, LV_ALIGN_CENTER, 0, 0);}
      else{
        this->text = lv_label_create(this->btn);
        lv_label_set_text(this->text, text);
        lv_obj_align(this->symbol, LV_ALIGN_CENTER, 0, -8 );
        lv_obj_align(this->text, LV_ALIGN_BOTTOM_MID, 0, 0 );
      }
      this->event_cb=func;
      lv_obj_add_event_cb(this->btn, this->btn_event_cb, LV_EVENT_ALL, NULL);
    }

    void btn_event_cb(lv_event_t * e){
        lv_event_code_t code = lv_event_get_code(e);
        if(code == LV_EVENT_CLICKED){
            (*this->event_cb)();
            printf("back func run\n");
        }
    }

    lv_obj_t * returnObj(){
      return this->btn;
    }
}

### Proposal

Any C++ examples could be provided for beginners ?

Jack-a-new, 2024-06-28:
The code in main.cpp is list below. Platform is simulator of codeblocks.

#include <stdlib.h>
#include <unistd.h>

#include "lvgl/lvgl.h"
#include "lvgl/demos/lv_demos.h"
#include "gui.hpp"

void btn_handle(){
    //lv_event_code_t code = lv_event_get_code(e);
    //if(code == LV_EVENT_CLICKED)
        printf("btn_handle run\n");
}
/**********************
 *   GLOBAL FUNCTIONS
 **********************/
int APIENTRY WinMain(HINSTANCE hInstance, HINSTANCE hPrevInstance, LPSTR szCmdLine, int nCmdShow)
{
    /*Initialize LVGL*/
    lv_init();

    /*Initialize the HAL for LVGL*/
    lv_display_t * display = lv_windows_create_display(title, 320, 240, 100, FALSE, FALSE);
    lv_windows_acquire_pointer_indev(display);

    /*Output prompt information to the console, you can also use printf() to print directly*/
    LV_LOG_USER("LVGL initialization completed!");

    lv_obj_t * scr = lv_obj_create(lv_screen_active());

    SymbolButton btn1(backgrd.returnObj(),LV_SYMBOL_AUDIO,"MUSIC",back);
    lv_obj_set_pos(btn1.returnObj(), 0, 0);

    while(1) {
        /* Periodically call the lv_task handler.
         * It could be done in a timer interrupt or an OS task too.*/
        lv_task_handler();
        usleep(5000);       /*Just to let the system breath*/
    }
    return 0;
}

Jack-a-new, 2024-06-28:
The code in main.cpp is list below. Platform is simulator of codeblocks.

#include <stdlib.h>
#include <unistd.h>

#include "lvgl/lvgl.h"
#include "lvgl/demos/lv_demos.h"
#include "gui.hpp"

void btn_handle(){
    //lv_event_code_t code = lv_event_get_code(e);
    //if(code == LV_EVENT_CLICKED)
        printf("btn_handle run\n");
}
/**********************
 *   GLOBAL FUNCTIONS
 **********************/
int APIENTRY WinMain(HINSTANCE hInstance, HINSTANCE hPrevInstance, LPSTR szCmdLine, int nCmdShow)
{
    /*Initialize LVGL*/
    lv_init();

    /*Initialize the HAL for LVGL*/
    lv_display_t * display = lv_windows_create_display(title, 320, 240, 100, FALSE, FALSE);
    lv_windows_acquire_pointer_indev(display);

    /*Output prompt information to the console, you can also use printf() to print directly*/
    LV_LOG_USER("LVGL initialization completed!");

    lv_obj_t * scr = lv_obj_create(lv_screen_active());

    SymbolButton btn1(scr ,LV_SYMBOL_AUDIO,"MUSIC",back);
    lv_obj_set_pos(btn1.returnObj(), 0, 0);

    while(1) {
        /* Periodically call the lv_task handler.
         * It could be done in a timer interrupt or an OS task too.*/
        lv_task_handler();
        usleep(5000);       /*Just to let the system breath*/
    }
    return 0;
}

and-elf, 2024-07-01:
Basically, you can't use a member function as a direct callback. 
I know two work-arounds; use a lambda that takes 'this' as argument i.e like this; lv_obj_add_event_cb(obj,[](lv_event_t* e){ auto o = static_cast<SymbolButton>(lv_event_get_user_data(e));
o->event_cb();
}, static_cast<void*>(this));

or use static objects that free-functions can use. Like this; 
static SymbolButton* s_btn;
void cb(lv_event_t* e){
if(s_btn){
s_btn->event_cb();
}
}
...
lv_obj_event_cb(obj, cb, code, nullptr);

Neither solution is perfect, but this is a problem with using cpp in c API's. 

In my current project, I went from cpp23 to c99, with static callbacks and objects. 

I actually wrote a cpp wrapper of lvgl, to handle this specific problem, but I scrapped it and now just use c instead.

paulh002, 2024-07-05:
You can more or less work around this issue by using a template like:

paulh002, 2024-07-05:
You can use a template to work around this issue:

paulh002, 2024-07-05:
You can use as a workaround a template to pass the class instance pointer to a member function

```
template <typename Class, void (Class::*MemberFunc)(lv_event_t *)>
struct EventHandler
{
	static void staticHandler(lv_event_t *e)
	{
		(static_cast<Class *>(e->user_data)->*MemberFunc)(e);
	}
};
``` 
when adding event pass the this pointer
`lv_obj_add_event_cb(button, buttons_handler, LV_EVENT_CLICKED, (void *)this);`
in the class the class definition you declare with the template the member function 

```
private:
void buttons_handler_class(lv_event_t *e);

public:
static constexpr auto buttonWindowObj_event_handler = EventHandler<guiButtonWindows, &guiButtonWindows::buttonWindowObj_event_handler_class>::staticHandler;
```	
Maybe this helps, there is no real c++ solution that I could find.

Jack-a-new, 2024-07-06:
> You can use as a workaround a template to pass the class instance pointer to a member function
>
> ```
> template <typename Class, void (Class::*MemberFunc)(lv_event_t *)>
> struct EventHandler
> {
> 	static [...]

OK, THANKS!

Jack-a-new, 2024-07-06:
> Basically, you can't use a member function as a direct callback. I know two work-arounds; use a lambda that takes 'this' as argument i.e like this; lv_obj_add_event_cb(obj,[](lv_event_t* e){ auto o = static_cast(lv_event_get_user_data(e)); o->event_cb(); }, static_cast<void*>(this));
> 
> or use static objects that free-functions can use. Like this; static SymbolButton* s_btn; void cb(lv_event_t* e){ if(s_btn){ s_btn->event_cb(); } } ... lv_obj_event_cb(obj, cb, code, nullptr);
> 
> Neither solution is perfect, but this is a problem with using cpp in c API's.
> 
> In my current project, I went from cpp23 to c99, with static callbacks and objects.
> 
> I actually wrote a cpp wrapper of lvgl, to handle this specific problem, but I scrapped it and now just use c instead.

OKay, good!

kisvegabor, 2024-07-11:
If the issue is resolved please close the issue.

closed 2024-08-05 by lvgl-bot