lvgl/lvgl #6442: why there is max height short type var limit?

Staok, 2024-06-28:
### LVGL version

8.3.11

### What happened?

I found that the obj such as a label have a default max height: 8191(by using "lv_obj_get_style_max_height()" discover this)

and my text content is large, put them into a label and then the height is more than 32767 (the max of "short" type...)

so how to handle this situation?

liamHowatt, 2024-07-04:
Can you try using `lv_obj_set_style_max_height` to increase the max height? Is there a `short` somewhere that prevents you from using `32767`?

kisvegabor, 2024-07-04:
You should enable [LV_USE_LARGE_COORD](https://github.com/lvgl/lvgl/blob/release/v8.4/lv_conf_template.h#L359).

Note that in v9 we have only `int32_t` so it shouldn't be an issue.

closed 2024-07-05 by Staok