lvgl/lvgl #6447: Separate the rendering and widgets

AlixANNERAUD, 2024-06-30:
### Introduce the problem

It would be good to be able to get rid of the rendering part when needed (with a `#define` in `lv_conf.h`), somewhat like having a "client" side and a "server" side.
The client side would only be responsible for setting up and configuring the data structure structures, and the server side for the rendering and input device. Currently, LVGL compiled in WASM occupies 568 KB, which is significant. I think we can reduce this by getting rid of the rendering part.

kisvegabor, 2024-07-01:
Hi,

By rendering you mean e.g. software rendering which actually draws a rectangles, text, etc, or even an upper layer which creates `draw_tasks`, dispatches them to `draw_unit`s.

What would be use case for this separation?

Who will do the rendering in the end?

AlixANNERAUD, 2024-07-02:
This can be useful in cases where WASM is used (notably with [WAMR](https://github.com/bytecodealliance/wasm-micro-runtime) on embedded devices) where the client application is sandboxed, preventing direct access to host memory. A naive approach would be to have a buffer per client application (each instance running its own LVGL "rendering engine"), but this is not very memory-efficient.

A better approach would be to compile LVGL as a WASM module without the rendering part (which is likely the heaviest part). The client application would then load this WASM LVGL module, create and set the necessary properties on its objects/widgets, and send references to the parent objects to the host containing the "rendering engine". These references would be attached by the host to one of the screens and then rendered.

Minimizing the size of the WASM part is crucial since the entire code is compiled when building it as a WASM module (similar to a shared library).

kisvegabor, 2024-07-09:
I see now but I'm not sure if it will work. Or have you already tried mixing the widget created by multiple LVGL instances?

My concern is that (I assume) each instance will create the widgets in its own memory pool. However it can happen that an other LVGL instance will delete/free the widgets. In this case `free` need to act on the other instance's memory. If you are using LVGL builtin `malloc` and `free` I believe `free` will crash as it assumes that that memory is allocated in in the current memory pool (and not the other instance's memory pool).

cc @liamHowatt

AlixANNERAUD, 2024-07-09:
Yes, that was also one of the points where I was not sure how to proceed. We would need to have a kind of `lv_set_parent_custom(lv_obj_t* parent, lv_obj_t* obj)` but which does not remove the entry of the object from the old parent (which does not actually exist in the host memory space).
Thus, the whole process would be:
- An object is created in the WASM application (allocated and initialized) : `lv_obj_create`
- The pointer is then passed via FFI to the host
- The latter calls `lv_set_parent_custom`
And normally this should resolve the issues between the different allocators, with the host owning the parent/screen and the WASM application owning the object in question.
On the other hand, the screen/parent object of the host should only be freed after all its children have been deleted (which should be the case in most situations), but I think we should have some sort of protection in place to prevent strange crashes.

kisvegabor, 2024-07-11:
Using a [custom allocator](https://github.com/lvgl/lvgl/blob/master/lv_conf_template.h#L43) which uses the host memory could be an option too. But I guess it beats the purpose of sand boxing.

AlixANNERAUD, 2024-07-11:
> Could you please go into a bit more detail about what exactly `lv_set_parent_custom` would do? Does it prevent `lv_free` from being called on it? What would the `parent` parameter be when called in the host? Thanks.

This function would do the same thing as `lv_set_parent`:
1. Remove the reference from the old parent and decrease the size of the old parent's children array.
2. Add the reference to the new parent and increase the size of the new parent's array.
However, we wouldn't have the first part because `lv_free` would not work/crash in the host memory space since the previous parent is located in wasm application memory space.

AlixANNERAUD, 2024-07-11:
> Using a [custom allocator](https://github.com/lvgl/lvgl/blob/master/lv_conf_template.h#L43) which uses the host memory could be an option too. But I guess it beats the purpose of sand boxing.

Sure, here is the translation:

---

Yes, in a raw manner, this is not a solution for executing unverified code since the passed host pointer could be modified at any point by the guest application.
Another approach I had thought of (and this will be the case if this issue is not accepted) was to have a hashmap for each application instance that bind an identifier to a pointer, and to have a custom allocator that, depending on the instance (as I can know the instance based on the current thread), will allocate memory in the correct instance and return the identifier instead of the pointer. Then instead of using pointer, function would use identifier (e.g. : `lv_set_pos(uint32_t obj, ...)).
However, for each call to LVGL functions, this implies creating a binding where the correspondence between pointer and identifier is made, which represents a lot of boilerplate code.

xlbao123, 2024-07-15:
According to my understanding, do you just need to write a virtual draw driver?

liamHowatt, 2024-07-15:
Yeah, could you write a draw driver that serializes the draw descriptors and coords and the host's LVGL deserializes the commands and calls `lv_draw_(rect|label|image|etc)`. Serialized images could be an exception where a const pointer is sent instead of expensively serializing the data buffers.

In order to get any binary size savings, the serializer would have to be smaller than the rendering code it's replacing.

> I think we can reduce this by getting rid of the rendering part.

It would be good to actually measure the compiled code size somehow so this isn't a wasted effort.

Fyi: #6313

kisvegabor, 2024-07-15:
> However, we wouldn't have the first part because lv_free would not work/crash in the host memory space since the previous parent is located in wasm application memory space. New parent could be any kind of screen / object BUT in host memory space.

How can we handle edge cases like this:
1. The application creates a container with some buttons. If one the buttons are pressed, delete the container
2. The container is attach to a window in the host.
3. The delete button is pressed, the callback from the application is called to delete the container, but it will delete it only from the application. 

As the problem is only the deletion a solution could be this:
1. Create the container as above. 
2. Attach it to a parent in host and also attache a `delete_event_callback` witch will detach it from the host container (just remove the reference).

Now the application has full freedom, but can the host do?
- Attach the application to an parent and let it to work as it's defined by the application
- Call the API function of the application to push some data
- Attach callbacks to be called e.g. when a button is pressed (E.g. "Connect to Bluetooth")
- Use [subjects/observers](https://docs.lvgl.io/master/others/observer.html) to share data.   
- Direct manipulation or creation of application's widget from the host is dangerous as they might involve memory reallocation (e.g. when adding a style) so these cannot be initiated from the host.
 

What do you think?

AlixANNERAUD, 2024-07-15:
> Yeah, could you write a draw driver that serializes the draw descriptors and coords and the host's LVGL deserializes the commands and calls `lv_draw_(rect|label|image|etc)`. Serialized images could [...]

Yes, this approach seems quite good to me as well because it completely eliminates the problem of choosing the allocator. However, we need to ensure that these calls for each rendering to the API via FFI are not too frequent, as FFI calls are quite expensive in terms of CPU time. The same issue applies to serialization/deserialization also (we could also use CBOR, by the way).

AlixANNERAUD, 2024-07-15:
> How can we handle edge cases like this:
>
>     1. The application creates a container with some buttons. If one the buttons are pressed, delete the container
>
>     2. The container is attach to a [...]

If I understand correctly, you want to create a new type of widget (a kind of container dedicated to FFI?). This is a very good idea and quite efficient. The fact that the host cannot do anything on the application's widgets doesn't seem too problematic to me because I don't see many scenarios where this would be useful.

AlixANNERAUD, 2024-07-15:
> Yeah, could you write a draw driver that serializes the draw descriptors and coords and the host's LVGL deserializes the commands and calls `lv_draw_(rect|label|image|etc)`. Serialized images could [...]

But yes, to start with, in order to be sure that all this is worth it, I can proceed with adding flags to exclude the rendering part from the compilation. However, before I start, I would like to know if you think this is feasible without major refactoring (with just the addition of #if here and there).

kisvegabor, 2024-07-17:
Actually rendering quite intertwined with the widgets as all widgets call `lv_draw_*` functions to draw themselves. Regardless to it `#if` can work. We can either
1. Disable the internals of the `lv_draw_*` functions  
2. Disable the rendering and other not necessary parts in all widgets

Option 2) would save much more flash as the extra code for rendering can be quite long. See e.g. [here](https://github.com/lvgl/lvgl/blob/master/src/widgets/chart/lv_chart.c#L688-L762). 

However, I don't think it would work. If the rendering code is no there, when the host asks the applications' widgets to render themselves nothing will happen as the rendering logic is removed from the application. 

This way, it won't be possible for the applications to define custom draw events either. E.g. [this](https://docs.lvgl.io/master/widgets/chart.html#faded-area-line-chart-with-custom-division-lines). In light of that maybe the serialization path that @liamHowatt recommended could be better. 

A simpler approach could be to disable all rendering units (e.g. `LV_USE_DRAW_SW   0`) first after that. During rendering LVGL will create `draw_task`s [here](https://github.com/lvgl/lvgl/blob/master/src/draw/lv_draw.c#L84) and we "just" need to pass the created draw tasks to the host. 

If we pass the application's main container's pointer to the host and the host will see the widgets as they were created there, the `draw_tasks` will be created by the host too in the host's display's layer.

It just came t my mind: can we read the application's memory from the host?

AlixANNERAUD, 2024-07-17:
> Actually rendering quite intertwined with the widgets as all widgets call `lv_draw_*` functions to draw themselves. Regardless to it `#if` can work. We can either
>
>     1. Disable the internals of [...]

I didn't know that one could customize the rendering pipeline to this extent with LVGL. Yes, indeed, it seems like a good solution. However, how do you think we could proceed to add draw_task to host draw_task ? And yes, the host can access the guest application's memory, you just need to perform [address translation](https://bytecodealliance.github.io/wamr.dev/docs/html/wasm__export_8h.html#ac4e4c84342e8c7a3c19f529805abc7ee) beforehand.

kisvegabor, 2024-08-01:
I'm sorry, I missed your reply. :slightly_frowning_face: 

Let's say you have an `lv_display` with an `screen` in the host. You add an application to the screen like:
```c
lv_obj_t * wasm_attach_lvgl_app(lv_obj_t * parent_in_host, lv_obj_t * root_widget_of_the_app);
```

This function manually adds a new child to `parent_in_host` and translates the address if required. From this point when LVGL traverses the widgets it will see `root_widget_of_the_app` as normal child of `parent_in_host`, call its callbacks, handle their drawing etc.  So it will look as it were created in the host and will be handled in transparent way. 

Do we need the address translation if LVGL call the callbacks and uses the children in C? I hope not. :smile: :crossed_fingers:

AlixANNERAUD, 2024-08-11:
Yes, that should be the most appropriate solution (essentially what I meant with my `lv_set_parent_custom` function). However, if `wasm_attach_lvgl_app` uses `lv_obj_set_parent`, it would remove the child pointer from the old parent, which is not possible [because the child is actually a WASM app screen.](https://github.com/lvgl/lvgl/blob/8699a7e86e18fdb09bb2d72e35c75bdedc539b14/src/core/lv_obj_tree.c#L149C1-L149C57)). That's why we need a custom `lv_obj_set_parent` function. In addition, wouldn't the lack of a registered display be a problem on WASM side to create a screen ?

For the callbacks (I assume you mean the event callback), a custom function should be added to handle event callbacks (specifically for address translation). However, the lv_event_t* memory is owned by the host, correct? So, it cannot be read within the WASM application. Perhaps the WASM app should pass a pointer to a memory location it owns when registering the callback, which would then be sent back to the callback when called.

Another issue is that the WASM LVGL part is still quite large in terms of binary size when compiled as a WASM module. This is somewhat expected since all of the drawing code is still included (even though it is completely unnecessary). The compiler treats a WASM module like a dynamic library and does not remove unused code, I believe.

kisvegabor, 2024-08-12:
Regarding `set_parent`: We can leave the app's container in its tree (in the app) because no one will use it. We just need to make it part of the host's tree. Anyway, I think we are mostly on the same page. It would be great to see a proof of concept for that. Just to test if we can really "channel" widget from an app to the host. 

Regarding the events: I think/hope events registered in the app will work in the app, without any further action. However, I see how events added to the app from the host can be tricky. 

Regarding memory size: It would be huge to use LVGL as a library which can be linked to multiple apps. However, as I described above probably the drawing routines cannot be removed to let the users use them in custom draw events.  But maybe it's not so important, because it seems we can have this setup:
- 1 common full LVGL shared library
- 1 host using the LVGL library
- Multiple apps using the same LVGL library

What do you think?

kisvegabor, 2024-10-25:
I'm also sorry for the delay. :slightly_frowning_face: 

> A solution to that could be, instead of having an lv_obj_t* lvgl_obj_create(lv_obj_t* parent) function that takes a parent and returns the allocated and initialized object, to have an alternative void lvgl_obj_initialize(lv_obj_t* obj, lv_obj_t* parent) function that only initializes the already allocated object (allowing custom allocation).

Take a look at [`lv_bar_create`](https://github.com/lvgl/lvgl/blob/master/src/widgets/bar/lv_bar.c#L85-L88). Here we already have some kind of separation:
1. Create and init a basic object on a parent
2. Initialize it according to its class(es).

The main malloc is in `lv_obj_class_create_obj` however nothing prevents the widgets to create new widgets internally (e.g. a tabview) or allocate more memory (e.g. chart). 

---

Maybe we should go back to your original idea: 
- have a full featured version of LVGL in the  host
- have a simpler version of LVGL in each client where a [custom malloc is defined](https://github.com/lvgl/lvgl/blob/master/lv_conf_template.h#L43)

The threading issue is still present, but at least each LVGL would work in its own memory. When the client creates widgets they will be allocated in the clients memory, but when the host does something (e.g. rendering) all the magic will be done in the hosts memory. The only issue I see here is when the host wants to delete a widget created in the client (e.g. when a screen is cleaned by the host). In this case the host will use its `free` function on a memory allocated by the client. Anyway, probably it can be solved by having a custom delete function or so to be used by the host.

AlixANNERAUD, 2024-11-14:
> Maybe we should go back to your original idea:
>
>     * have a full featured version of LVGL in the  host
>
>     * have a simpler version of LVGL in each client where a [custom malloc is [...]

That seems like the most reasonable solution to me.

First, there's the question of code size, which I measured to be around 600 kB. This isn’t a major concern, as if the LVGL client is statically linked to the binary, the compiler will eliminate unused code, significantly reducing the size. Code size only becomes an issue when LVGL is built as a WASI reactor / submodule (where it behaves as a shared library).

However, there’s still the issue of allowing client  third-party code (= unsafe) to put incorrect pointers in object attributes, which are then dereferenced by the host rendering component. This can lead to various memory safety issues. This risk could be mitigated if LVGL used its own memory pool, which is the case by default, allowing for easy memory checks before dereferencing pointers (e.g. "Is the memory address is in the pool ?"). I’m not sure if LVGL currently supports this kind of feature—perhaps with `LV_USE_ASSERT_MEM_INTEGRITY`?

Additionally, the lack of thread safety is a deal breaker for me, especially given the memory safety concerns with "unsafe code." You mentioned that a mutex/rwlock mechanism may not be implemented soon due to code visibility and performance issues (though a read-write lock could mitigate this, as most writes would occur at application startup). I understand the rationale, as LVGL isn’t primarily designed for this use case and adding thread safety would require substantial effort.

I realize this issue goes beyond LVGL's scope, and I completely understand if you prefer not to pursue it further. However, I’d be happy to discuss any potential proposals you might have that I could implement in a pull request!

kisvegabor, 2024-11-17:
Regarding pointer de-referencing: We can add some checks on the most obvious places but it's not reasonable to check all the pointers. It would practically impossible to maintains. 

Regarding thread safety: could you summarize what is the actual issue?

> I realize this issue goes beyond LVGL's scope, and I completely understand if you prefer not to pursue it further. However, I’d be happy to discuss any potential proposals you might have that I could implement in a pull request!

I'll try to help as much as I can. Hopefully we will find a simple solution. :slightly_smiling_face:

AlixANNERAUD, 2024-11-18:
The issue with thread safety arises when an object is modified in a different thread on a multi-threaded platform (such as the ESP32 MCU family). Concurrent rendering in this scenario could lead to undefined behavior.

A potential solution, without significant modifications to LVGL, would be to implement a global mutex or rw lock. However, this approach would require 3rd party "unsafe" code to intentionally lock resources, which is not acceptable since it is, well, unsafe.

Consequently, each LVGL call would need to be protected by a global mutex or rw lock, but this is not currently planned (as indicated in one of the LVGL issues, though I couldn’t find the exact reference). As such, I don’t see any other viable solution at the moment.

kisvegabor, 2024-11-21:
Ah I see. 

> A potential solution, without significant modifications to LVGL, would be to implement a global mutex or rw lock. However, this approach would require 3rd party "unsafe" code to intentionally lock resources, which is not acceptable since it is, well, unsafe.

We are supporting this direction as described [here](https://docs.lvgl.io/master/intro/add-lvgl-to-your-project/threading.html#lvgl-and-threads). :slightly_frowning_face:

AlixANNERAUD, 2024-12-04:
> Ah I see.
>
>
> We are supporting this direction as described [here](https://docs.lvgl.io/master/intro/add-lvgl-to-your-project/threading.html#lvgl-and-threads). 🙁

Yeah, and I totally respect that but I think we are in a dead end...

kisvegabor, 2024-12-22:
> Yeah, and I totally respect that but I think we are in a dead end...

I'm very sorry to hear it. :cry: 

Let's close it for now then, and reopen if anything new comes up.

closed 2024-12-22 by kisvegabor