lvgl/lvgl #6449: Extra styles support for lv_scale widget

lorddevereux, 2024-06-30:
### Problem to solve

I have some projects needing to use lv_scale, and in the process I think there are a few extra style features to be added to make lv_scale as flexible as possible.

These suggestions are for the ROUND scale modes to allow easier styling for the ticks and labels.

(Also it's a small issue but the docs for 9.0/9.1 says [lv_scale_set_draw_ticks_on_top](https://docs.lvgl.io/9.1/API/widgets/scale/lv_scale.html#_CPPv430lv_scale_set_draw_ticks_on_topP8lv_obj_tb) is implemented, but its only in master at the moment)

### Success criteria

The lv_scale widget is more powerful/extendable.

### Solution outline

First, is to use a style attribute on minor/major ticks and labels in order to allow to offset them from the dial. In my example I've created a new one called RADIAL_OFFSET (13), but let me know if there is a better idea.

Second is another style attribute to offset the labels from the major ticks. This would be another attribute, I've created RADIAL_TEXT_OFFSET (14). 

Third, I implemented the standard TRANSLATE_X/TRANSLATE_Y to move the labels up/down/sideways in relation to the ticks. This I could implement for the linear scale also? 

For the ROUND scale, I implemented TRANSFORM_ROTATION which allows to offset the labels by a few degrees from each tick (see illustration). I think this is really a TRANSLATE_ROTATION, so maybe a new style attribute is needed for this?

Finally, I've added lv_scale_set_major_tick_ends which allows to toggle if a major tick is shown at the start/end of the scale.

Changes are in lv_scale.c/h, and also in lv_obj_style_gen.c/h.

At the moment these options apply to the whole scale (including sections) - but I could add support for using section styles to vary these parameters within sections.

### Rabbit holes

I can't think of any issues right now.

### Testing

Here is an example with a few of the options in action
[image]

### Teaching

Add the new fields to the lv_scale documentation

### Considerations

I'm happy to implement this as a PR if you think it's a good idea (it works) but I can tweak the attribute names if needed.

liamHowatt, 2024-07-04:
Sounds cool! So these would be new style properties? With a prefix like `LV_STYLE_SCALE_*`, similar to what there is for arc? My first thought is we should be careful not to add new style properties but maybe these are worthy. Could `TRANSLATE_X` and `TRANSLATE_ROTATION` be merged and the meaning would depend on whether the scale is linear or round?

kisvegabor, 2024-07-04:
I like the idea of `RADIAL_OFFSET`. I believe `RADIAL_TEXT_OFFSET` is meant to replace [LV_SCALE_DEFAULT_LABEL_GAP](https://github.com/lvgl/lvgl/blob/master/src/widgets/scale/lv_scale.c#L649). It's a pity that the major ticks and labels share the `LV_PART_INDICATOR` so we can't use `RADIAL_OFFSET` for both. Any ideas for having a single style property for both somehow? Regarding the name: what about `TRANSLATE_RADIAL`? It can be used for the arc too.

I think `TRANSLATE_X/TRANSLATE_Y` are fine and adding them to the linear scales would be great too. 

IMO `TRANSFORM_ROTATION` should be used to really rotate the label. It's a little bit tricky to implement but possible. The usage for linear scale is trivial, just rotate all labels e.g. by  30°. In case of radial scale we can just enable to rotate all labels to the their current angle. With radial `TRANSFORM_ROTATION` can be used as you have suggested. 

> With a prefix like LV_STYLE_SCALE_*, similar to what there is for arc?

I would keep these generic to allow using them in a more flexible way later.

lorddevereux, 2024-07-20:
> I believe RADIAL_TEXT_OFFSET is meant to replace [LV_SCALE_DEFAULT_LABEL_GAP](https://github.com/lvgl/lvgl/blob/master/src/widgets/scale/lv_scale.c#L649).
You're right, I was writing code in this function and didn't notice this somehow! 

It would be nice if the labels could be a separate part to allow them to be fully controlled separately to the major ticks. However, the only existing one we could possibly use is _maybe_ LV_PART_CURSOR? This might have unintended consequences, I don't know. If we did this, we could then get rid of **lv_scale_set_label_show** and just show/hide the part instead.

I will implement the properties for the linear scale also.

I like TRANSLATE_RADIAL, so we have new properties:
- RADIAL_OFFSET (13) - (radial scale only) offset part from the default radial position
- TRANSLATE_RADIAL (117) - (radial scale only) move the labels around the scale by the given angle
- RADIAL_TEXT_OFFSET (14) - (radial scale only + if we don't assign labels to a new LV_PART) offset text labels from the default radial position

Existing styles implemented for the scale:
- TRANSLATE_X/Y - (linear + radial) adjust label position from the default
- TRANSFORM_ROTATION - (linear + radial) rotate the label. For the linear scale this can be absolute e.g. 30deg. 

For the radial scale, I can see 2 use cases for TRANSFORM_ROTATION. One is to rotate all the labels the same amount (like linear) and the other is rotate them to match the angle of their tick. How about this?
- if TRANSFORM_ROTATION is NOT set, rotate the labels to match the ticks
- if TRANSFORM_ROTATION is 0, don't rotate the labels (they stay horizontal)
- otherwise, rotate the labels to the value of TRANSFORM_ROTATION

hyansuper, 2024-07-20:
large ticks would look better if we can set something like `lv_obj_set_style_line_rounded(scale, LV_RADIUS_CIRCLE, LV_PART_ITEMS)`

kisvegabor, 2024-07-22:
@lorddevereux Your summary sounds good, but the `RADIAL_TEXT_OFFSET` still puzzles me. If we'd map it `LV_PART_CURSOR` it were an API break. So we need to keep using `LV_PART_INDICATOR` for both the label and the major ticks. What if we add a `PAD_RADIAL` style property to define the space between the label the major ticks? It's more universal. 

Regarding `TRANSFORM_ROTATION`, it's harder to get if a value is unset as if you get it the default value will be returned. And we don't use this practice on other places. What about this:
- `0`: do nothing means 0 degrees
- `0..360˚`: just rotate by this angle in case of both radial and linear
-  `0x100000 + 0..360`: rotate to the current degree + the given offset. Of course `0x100000` can be macro or function or so. 

@hyansuper 
> large ticks would look better if we can set something like lv_obj_set_style_line_rounded(scale, LV_RADIUS_CIRCLE, LV_PART_ITEMS)

It can be used achieved like this:
```c
    lv_obj_set_style_line_width(scale_line, 5, LV_PART_INDICATOR);
    lv_obj_set_style_line_rounded(scale_line, 1, LV_PART_INDICATOR);
```

lorddevereux, 2024-08-02:
I've got everything working now except for TRANSFORM_ROTATION, because I don't know how to rotate a label and I'm struggling to work out how it's done for the normal label widget!

Can you point me in the right direction?

liamHowatt, 2024-08-02:
Can you try `lv_obj_set_style_transform_rotation`?

kisvegabor, 2024-08-03:
These are not label widgets just drawn on the fly, so it's a little bit more complicated. I can creat an example on Monday.

kisvegabor, 2024-08-05:
Here, you can see how to rotate a label on the fly: https://github.com/kisvegabor/lvgl_upstream/commit/65e748b32229103a8d4f20a86657c5d2944d91b9

The angle is fixed 30°, but it should work with any number.

lorddevereux, 2024-08-07:
I tweaked it to set the pivot point at the centre of the label, and it works perfectly!

On the left we can see labels translated radially from the major ticks (white)
On the right we can see labels rotated to match the major ticks
[image]

I'll clean this up now and submit a PR for review

lorddevereux, 2024-08-07:
I realised there is an issue with some labels being drawn upside down if the dial goes all the way around. Sometimes this is desired, but sometimes it isn't, so I added another flag **LV_SCALE_LABEL_ROTATE_KEEP_UPRIGHT** which will make the labels always be the right way up.

kisvegabor, 2024-08-23:
Let's continue in #6649

closed 2024-08-23 by kisvegabor