lvgl/lvgl #6451: lv_canvas_draw_polygon stucks when drawing complicated shape

Yangnaan, 2024-07-01:
### LVGL version

v9 or v8

### What happened?

the bug is easily reproduced in simulator.the shape is as follows.
[image]

### How to reproduce?

#define CANVAS_WIDTH  410
#define CANVAS_HEIGHT  410
const lv_point_t point[] ={ {229, 1}, { 228, 3 }, { 229, 4 }, { 229, 5 }, { 227, 4 }, { 226, 5 }, { 228, 7 }, { 226, 10 }, { 224, 9 }, { 223, 11 }, { 223, 13 }, { 221, 13 }, { 241, 81 }, { 244, 86 }, { 251, 90 }, { 262, 92 }, { 267, 94 }, { 272, 99 }, { 276, 104 }, { 277, 105 }, { 275, 110 }, { 283, 115 }, { 297, 114 }, { 306, 114 }, { 325, 123 }, { 329, 112 }, { 328, 104 }, { 336, 99 }, { 331, 84 }, { 315, 57 }, { 316, 47 }, { 300, 4 }, { 309, 0 }, { 228, 0 }, { 229, 1 }};
void lv_example_canvas_1(void)
{
    lv_draw_rect_dsc_t rect_dsc;
    lv_draw_rect_dsc_init(&rect_dsc);
    rect_dsc.bg_color = lv_color_black();
    rect_dsc.border_width = 1;
    rect_dsc.border_color = lv_color_black();

    static uint8_t cbuf[LV_CANVAS_BUF_SIZE_TRUE_COLOR(CANVAS_WIDTH, CANVAS_HEIGHT)];

    lv_obj_t * canvas = lv_canvas_create(lv_scr_act());
    lv_canvas_set_buffer(canvas, cbuf, CANVAS_WIDTH, CANVAS_HEIGHT, LV_COLOR_FORMAT_NATIVE);
    lv_obj_center(canvas);
    lv_canvas_fill_bg(canvas, lv_palette_lighten(LV_PALETTE_GREY, 3), LV_OPA_COVER);

    lv_canvas_draw_polygon(canvas, point, 35, &rect_dsc);
    }

Yangnaan, 2024-07-01:
it stucks in an endless loop wthin lv_draw_sw_polygon(line: 143)

Yangnaan, 2024-07-01:
according to https://forum.lvgl.io/t/lv-draw-polygon-example/13052/5, convex polygons is supportd only, but concave polygons drawing is needed too, with which i attend to draw vector map polygon

liamHowatt, 2024-07-05:
Can you use vector graphics (see `lv_example_canvas_8`) or lines (see `lv_example_canvas_7`)?

Yangnaan, 2024-07-08:
> Can you use vector graphics (see `lv_example_canvas_8`) or lines (see `lv_example_canvas_7`)?

i tried lv_example_cavas_8 in simulator, but got blank result
[image]

liamHowatt, 2024-07-09:
Please make sure `LV_USE_VECTOR_GRAPHIC` is `1` and `LV_USE_THORVG_INTERNAL` is `1` in your lv_conf.h

Sorry for the lack of info. There are different vector graphics systems and ThorVG is the software-based one in LVGL and it needs to be explicitly enabled.

Yangnaan, 2024-07-10:
thanks, with vector feature in v9.1, the polygon could be drawn sucessfully
[image]

closed 2024-08-04 by lvgl-bot