lvgl/lvgl #6455: Artifacts using LV_DISPLAY_RENDER_MODE_DIRECT

LuisCRSousa, 2024-07-01:
### LVGL version

v9.1.0

### What happened?

Below, two results using LV_DISPLAY_RENDER_MODE_PARTIAL and LV_DISPLAY_RENDER_MODE_DIRECT modes respectively.
Everything works using PARTIAL mode but I get artifacts using DIRECT mode.

Platform: STM32H735IGK6U-DK
Configuration: 1 lvgl buffer using complete screen `static __attribute__((aligned(32))) lv_color16_t buf_1[MY_DISP_HOR_RES * MY_DISP_VER_RES];`

Not sure if I need to do anything diferent uwing DIRECT mode. Documentation is not clear and I can't find any example using DIRECT mode.

[image]
[image]

### How to reproduce?

static __attribute__((aligned(32))) lv_color16_t buf_1[MY_DISP_HOR_RES * MY_DISP_VER_RES];
static lv_display_t * disp;
disp = lv_display_create(MY_DISP_HOR_RES, MY_DISP_VER_RES);
lv_display_set_flush_cb(disp, disp_flush);
lv_display_set_buffers(disp, (void*) buf_1, NULL, sizeof(buf_1), LV_DISPLAY_RENDER_MODE_DIRECT);

TimJTi, 2024-07-03:
I think LVGL V9 and NuttX are not playing nicely together yet.  See [this](https://github.com/lvgl/lvgl/issues/6389)

liamHowatt, 2024-07-04:
It's important to know that the flush callback is meant to handle flushing differently when the render mode is `DIRECT` instead of `PARTIAL`.

For `PARTIAL`, the first pixel in `px_map` is the first coordinate in `area`. For `DIRECT`, `area` is still the area that changed, but the first pixel in `px_map` is always the start of your `buf_1`.

TimJTi, 2024-07-22:
@liamHowatt 
> For `PARTIAL`, the first pixel in `px_map` is the first coordinate in `area`. For `DIRECT`, `area` is still the area that changed, but the first pixel in `px_map` is always the start of your `buf_1`.

Where is this documented, please? Or an example?

liamHowatt, 2024-07-23:
Here are the docs for this behavior: https://docs.lvgl.io/master/porting/display.html#draw-buffers
and here is a template: https://github.com/lvgl/lvgl/blob/master/examples/porting/lv_port_disp_template.c

The most confusing part is that for `LV_DISPLAY_RENDER_MODE_DIRECT`, `px_map` points to the start of the whole buffer and not the start of the area to flush, as I mentioned.

The problem would only be partially remedied by making it point to the start of the relevant area in the buffer, because the user would still have to increment by the stride length at the end of each row of pixels.

One option is to not do anything in the flush callback unless `lv_display_flush_is_last` is `true`, and then copy the whole `px_map` to the screen from start to finish.

closed 2024-08-23 by lvgl-bot

woniulive, 2024-09-05:
@liamHowatt :
Hi, if I judge lv_display_flush_is_last, and copy the whole px_map to the screen from start to finish, then what is the difference between LV_DISPLAY_RENDER_MODE_DIRECT and LV_DISPLAY_RENDER_MODE_FULL?

I have external RAM but I only have SPI interface to drive TFT, which means I have enough RAM but the screen refresh speed is too slow. I want to use LV_DISPLAY_RENDER_MODE_DIRECT to improve this. But I don't know how to do this.

As you said, px_map points to the start of the whole buffer and not the start of the area to flush when using LV_DISPLAY_RENDER_MODE_DIRECT, if so, how can I know the area I need to refresh in disp_flush function?

liamHowatt, 2024-09-05:
Hi!

> Hi, if I judge lv_display_flush_is_last, and copy the whole px_map to the screen from start to finish, then what is the difference between LV_DISPLAY_RENDER_MODE_DIRECT and LV_DISPLAY_RENDER_MODE_FULL?

LVGL will only render the parts of the screen that it needs to, from one frame to the next, in `LV_DISPLAY_RENDER_MODE_DIRECT`, whereas with `LV_DISPLAY_RENDER_MODE_FULL` it redraws everything on the screen every frame. In both modes, the `px_map` always has all the pixels in it at the last flush, the only noticeable difference is slightly worse performance with `FULL` because it's spending more time redrawing everything.

`LV_DISPLAY_RENDER_MODE_FULL` is mostly for debugging or some rare cases. You can most likely ignore it, and immediately start using `LV_DISPLAY_RENDER_MODE_DIRECT` with no changes to your flush callback.

> I have external RAM but I only have SPI interface to drive TFT, which means I have enough RAM but the screen refresh speed is too slow. I want to use LV_DISPLAY_RENDER_MODE_DIRECT to improve this. But I don't know how to do this.

The `area` parameter of the flush callback has the area that changed each time the flush callback is called (every call. Not just the final one), so you could send only those sub-rectangles of the framebuffer over SPI and it would be less data on average than sending the whole thing every time. The main challenge for you will be sending a command to the display over SPI which tells it which area you're sending data about. Does your display support that?

> As you said, px_map points to the start of the whole buffer and not the start of the area to flush when using LV_DISPLAY_RENDER_MODE_DIRECT, if so, how can I know the area I need to refresh in disp_flush function?

```c
void your_flush_cb(lv_display_t * display, const lv_area_t * area, uint8_t * px_map)
{
    // ...
    uint32_t pixel_size = lv_color_format_get_size(LV_COLOR_FORMAT_NATIVE);
    uint8_t * first_pixel_of_data_that_changed = px_map + ((DISP_WIDTH * area->y1 + area->x1) * pixel_size);
    // ...
}
```

woniulive, 2024-09-05:
@liamHowatt 
Thank you for your answer.
I use ST7789 with 240 *320 resolution, so I think it should support.
But when I use code below, it will display correctly if area is 0~239 and 0 ~319, but will be wrong in other area. Do you have any advice?
```
static void disp_flush(lv_display_t * disp_drv, const lv_area_t * area, uint8_t * px_map)
{
    if(disp_flush_enabled) {
        /*The most simple case (but also the slowest) to put all pixels to the screen one-by-one*/
#if 0
        int32_t x;
        int32_t y;
        for(y = area->y1; y <= area->y2; y++) {
            for(x = area->x1; x <= area->x2; x++) {
                /*Put a pixel to the display. For example:*/
                /*put_px(x, y, *px_map)*/
                px_map++;
            }
        }
#else
        uint32_t pixel_size = lv_color_format_get_size(LV_COLOR_FORMAT_NATIVE);
        uint8_t * first_pixel_of_data_that_changed = 0;

        if(disp_drv->render_mode == LV_DISPLAY_RENDER_MODE_DIRECT)
        {
            first_pixel_of_data_that_changed = (uint8_t *)(px_map + ((MY_DISP_HOR_RES * area->y1 + area->x1) * pixel_size));

        }
        else
        {
            first_pixel_of_data_that_changed = (uint8_t *)px_map;
        }

        int data_size = (area->y2 - area->y1 + 1) * (area->x2 - area->x1 + 1);
        spitft_write_color_buff(area->x1, area->y1, area->x2, area->y2, (uint16_t *)first_pixel_of_data_that_changed, data_size);
#endif
    }

    /*IMPORTANT!!!
     *Inform the graphics library that you are ready with the flushing*/
    lv_display_flush_ready(disp_drv);
}
```

liamHowatt, 2024-09-09:
That all looks very good. I can't see anything wrong. I assume `spitft_write_color_buff` uses the area parameters to command the display to set the current drawing rectangle.

The only thing I can think of is `spitft_write_color_buff` is trying to find the first pixel by itself and the manual first pixel calculation is unnecessary. You probably already tried without the calculation though.

Is the behavior of `spitft_write_color_buff` documented or do you have its source code?

woniulive, 2024-09-28:
Hi @liamHowatt :
Sorry, I miss this notification. spitft_write_color_buff will send the buff to DMA directly. After I tried serval times, I figured out the reason should be the adress is not continuous. For example, if area->x1 = 5, area->y1 = 5, area->x2 = 10, area->y2 = 10, then the number of changed pixel is 36. if I send the address first_pixel_of_data_that_changed ~ first_pixel_of_data_that_changed+36 to SPI, it will be wrong. Actully, I should send address like first_pixel_of_data_that_changed ~ first_pixel_of_data_that_changed+6, first_pixel_of_data_that_changed + MY_DISP_HOR_RES ~ first_pixel_of_data_that_changed + MY_DISP_HOR_RES + 6...

After I fixed this problem, I run LVGL V9.2 benchmark demo and get result below:
the SPI buadrate is 60M(already max), Do you have any advise to improve FPS?
When I use -O3, the screen will have ripples while it is OK when using -O0. So I can't enable optimization now.
My lv_conf.h is as attachment.
[lv_conf.zip](https://github.com/user-attachments/files/17173795/lv_conf.zip)
Name, Avg. CPU, Avg. FPS, Avg. time, render time, flush time
* Empty screen, 97%, 13, 62, 34, 28
* Moving wallpaper, 98%, 11, 80, 52, 28
* Single rectangle, 96%, 27, 28,11, 17
* Multiple rectangles, 98%, 6, 116, 42, 74
* Multiple RGB images,97%, 22, 33, 9, 24
* Multiple ARGB images, 97%, 22, 33, 8, 25
* Rotated ARGB images, 97%, 13, 57, 15, 42
* Multiple labels, 97%, 18, 42, 19, 23
* Screen sized text, 99%, 5, 169, 141, 28
* Multiple arcs, 98%, 24, 28, 11, 17
* Containers, 98%, 9, 88, 59, 29
* Containers with overlay, 99%, 6, 131, 102, 29
* Containers with opa, 99%, 5, 136, 107, 29
* Containers with opa_layer, 99%, 4, 195, 166, 29
* Containers with scrolling, 98%, 8, 111, 83, 28
* Widgets demo, 99%, 4, 109, 83, 26
* All scenes avg., 97%, 12, 87, 58, 29

kisvegabor, 2024-10-03:
Could you send a video bout the `-O3` flickering. With optimization it will be 2-3 times faster so it's de great to make it work somehow. :slightly_smiling_face:

woniulive, 2024-10-08:
Hi @kisvegabor :
Actually, I am not sure this flickering caused by ST or LVGL or myself. Because I use external QSPI Flash as XIP and external QSPI PSRAM as data memory.
But I will open a new issue and provide a video for this problem.

woniulive, 2024-10-08:
Hi @kisvegabor :
I create issue #7008, If you need anything else, please let me know.

WebDust21, 2024-10-09:
> but I only have SPI interface to drive TFT

> Because I use external QSPI Flash as XIP and external QSPI PSRAM as data memory.

FPS will be poor because the memory access to your processor (and display!) is severely limited due to all of it being over SPI.  There is no software solution to a hardware problem.

> After I fixed this problem, I run LVGL V9.2 benchmark demo and get result below:
> the SPI buadrate is 60M(already max), Do you have any advise to improve FPS?

Let's do some math.  

1. Single-bit SPI @ 60MHz = ~6 megabytes/sec theoretical data rate.  (ST7889 only supports single-bit SPI, not QSPI.)
2. 320*240 pixels * 2 bytes/pixel (16bpp) is 153KB of data.  
3. 6 megabytes/sec / 153KB/frame = 39fps maximum _theoretically_ possible with full-screen redraws and zero processor memory waitstates.  (With LVGL being run out of external QSPI FLASH, and having the framebuffers in external QSPI PSRAM, there most definitely are severe memory waitstate penalties in play!)
4. This means that mathematically, the only way to read higher FPS is if LVGL is only redrawing _part_ of the screen--or if you're running a lower bits/pixel (i.e. 8bpp).

WebDust21, 2024-10-09:
Correct.  My math above was to show that there is a physical hardware limit with the display via SPI.

However, that is not at all anything close to the real issue: the fact that you're running LVGL out of QSPI FLASH.  This is a horrible bottleneck on the processor.

The processor runs at 550MHz, and [runs the THUMB instruction set](https://developer.arm.com/documentation/dui0646/c/The-Cortex-M7-Instruction-Set/Instruction-set-summary), which is a 16-bit instruction.  (ARM instructions are 32-bit.)
Assuming this CPU can run 1 MIPS/MHz (i.e. one clock cycle per instruction), this will require a _minimum_ data rate of 1,100 megabytes/second for the program alone.  (550 million instructions per second * 2 bytes/instruction.)
I don't know how fast the internal FLASH runs (and am well aware it's too small to fit LVGL), but I _do_ know that QSPI @ 60MHz will only give you a maximum theoretical data rate of 30 megabytes/second.  THIRTY.
That means that your processor is bottlenecked to absolutely no more than (30 / 1100) = 27% of its full potential.  Just in the XIP QSPI alone.  

Add to that all of the graphics memory being in external QSPI PSRAM, and the processor bottleneck only gets worse.

Case in point: I have one of these [Waveshare ESP32-S3 7" touchscreen boards](https://www.waveshare.com/esp32-s3-touch-lcd-7.htm).  The ESP32-series of processors have almost identical severe bottlenecks on the (otherwise very capable) processors, resulting in very similar behavior.  In the case of an ESP32-S3, we have a dual-core 32-bit processor running at 400MHz...with all program access over QSPI @ 80MHz (=40MBytes/sec theoretical throughput).  All graphics has to be handled in OPI PSRAM @ 80MHz (octal PSRAM = 8 lines = 80MBytes/sec).
**It literally can only manage 1-2fps full-screen redraws of an 800x480 panel.**

Then on the other hand, I'm currently playing around with an Allwinner V3s processor.  It has 64MBytes of on-chip DRAM, which has a 16-bit bus running at 400MHz (= 800MBytes/sec throughput) to the processor.  In theory, being DDR (double data rate) DRAM, that should provide closer to 1,600MBytes/sec throughput.  (Yeah, sure, the processor runs at 1.08GHz, but that's not significant here.) 
I almost can't get it to do _LESS_ than 30fps with full-screen redraws of an 800x480 panel.

WebDust21, 2024-10-09:
TLDR; doesn't matter how fancy or powerful your bright red fire engine is...if the water source is an IV drip, it's not going to put out a fire!

woniulive, 2024-10-10:
Hi @WebDust21 :
But Empty screen can get 13FPS and I only store assets in external flash, the code run in internal flash. So I think maybe we can improve this 4 FPS by settings something.

kisvegabor, 2024-10-16:
Okay, just to make sure we know all the details well. 
- You have STM32H735IGK6U, 550MHz
- -O0 optimization
- screen sized buffer for partial rendering
- You send the rendered image with SPI

My main question: to run the benchmark, why do you need any external memory. If its really not required, please disable it as an experiment and also try -O3 again.

woniulive, 2024-10-17:
Hi @kisvegabor :
I use LV_DISPLAY_RENDER_MODE_DIRECT with two buffers as below, that is why I use external memory:
```
   LV_ATTRIBUTE_MEM_ALIGN
    __attribute__((section(".ext_ram"))) static uint8_t buf_3_1[MY_DISP_HOR_RES * MY_DISP_VER_RES * BYTE_PER_PIXEL];

    LV_ATTRIBUTE_MEM_ALIGN
    __attribute__((section(".ext_ram"))) static uint8_t buf_3_2[MY_DISP_HOR_RES * MY_DISP_VER_RES * BYTE_PER_PIXEL];
    lv_display_set_buffers(disp, buf_3_1, buf_3_2, sizeof(buf_3_1), LV_DISPLAY_RENDER_MODE_DIRECT);
```
After I enble -O3, I get below result:
Name, Avg. CPU, Avg. FPS, Avg. time, render time, flush time

* Empty screen, 99%, 17, 51, 23, 28
* Moving wallpaper, 99%, 14, 64, 36, 28
* Single rectangle, 97%, 28, 22,5, 17
* Multiple rectangles, 98%, 7, 159, 30, 129
* Multiple RGB images,98%, 25, 33, 5, 28
* Multiple ARGB images, 99%, 20, 44, 15, 29
* Rotated ARGB images, 100%, 8, 95, 48, 47
* Multiple labels, 98%, 26, 29, 7, 22
* Screen sized text, 99%, 8, 107, 79, 28
* Multiple arcs, 97%, 26, 24, 5, 19
* Containers, 99%, 11, 77, 49, 28
* Containers with overlay, 100%, 7, 118, 90, 28
* Containers with opa, 100%, 7, 121, 92, 29
* Containers with opa_layer, 100%, 6, 132, 104, 28
* Containers with scrolling, 100%, 9, 97, 69, 28
* Widgets demo, 99%, 9, 72, 48, 24
All scenes avg., 98%, 14, 77, 44, 33

kisvegabor, 2024-10-17:
Ah, I see. In case of an external display controller a smaller draw buffer in the faster internal RAM should be a better choice. Please try  it out.