lvgl/lvgl #6456: Flag LV_PART_INDICATOR dont work to change the font of indicator checkbox.

3622-1768, 2024-07-01:
### LVGL version

v8.1

### What happened?

I am attempting to change the font of the checkbox because the symbol isn't drawing correctly. Here is what I'm trying:

### How to reproduce?

  cb = lv_checkbox_create(cont);
    lv_checkbox_set_text(cb, "Lemon");
    lv_obj_add_event_cb(cb, event_checkbox, LV_EVENT_ALL, NULL);
    lv_obj_set_style_text_font(cb, &Montserrat_Latin_24, LV_PART_INDICATOR | LV_STATE_DEFAULT);

Yangnaan, 2024-07-02:
try:
lv_obj_set_style_text_font(cb, &Montserrat_Latin_24, LV_PART_MAIN);

it works.

3622-1768, 2024-07-02:
the icon is still not drawn correctly: 
[image]

Yangnaan, 2024-07-03:
try :

lv_obj_set_style_text_font(cb, &Montserrat_Latin_24, LV_PART_INDICATOR | LV_STATE_CHECKED);

3622-1768, 2024-07-03:
it works!
thanks!

C47D, 2024-07-03:
@kisvegabor should we add this note and code sample on the docs?

kisvegabor, 2024-07-17:
In #6521 I've restructured the theme a little bit to make it work with `LV_PART_INDICATOR | LV_STATE_DEFAULT`.

PR #6521 (2024-07-24): fix(theme): make the text styles work on the INDICATOR's DEFAULT state
fixes #6456

### Description of the feature or fix

A clear and concise description of what the bug or new feature is.

### Notes
- Update the [Documentation](https://github.com/lvgl/lvgl/tree/master/docs) if needed.
- Add [Examples](https://github.com/lvgl/lvgl/tree/master/examples) if relevant.
- Add [Tests](https://github.com/lvgl/lvgl/blob/master/tests/README.md) if applicable.
- If you added new options to `lv_conf_template.h` run [lv_conf_internal_gen.py](https://github.com/lvgl/lvgl/blob/master/scripts/lv_conf_intern [... cut]
closed 2024-07-24 by kisvegabor