lvgl/lvgl #6463: Drawing of vector on a canvas layer - the area outside the vector is filled with "white"

minipost, 2024-07-03:
### LVGL version

v9.1.0

### What happened?

The "lv_example_canvas_8" draws a blue triangle on a grey canvas. However contrary to previous canvas examples the area outside of the triangle appears to be filled with white.

### How to reproduce?

Run [example 8](https://github.com/lvgl/lvgl/blob/1e2b1988aa205d096939c7401a3b971e4814af8d/examples/widgets/canvas/lv_example_canvas_8.c)

Observe that the canvas area not covered by the triangle is white:
[image]

comment out l. 38 "lv_draw_vector(dsc);" and observe that the the grey canvas background is now visible.
[image]

liamHowatt, 2024-07-04:
Thanks. There's already a PR that fixes it: #6406

minipost, 2024-07-05:
I just tried https://github.com/kisvegabor/lvgl_upstream.git (revision 7089cb14e923eefb634d81c7fc1e635e07bf1923)
Unfortunately the above behavior appears to be the same.

liamHowatt, 2024-07-05:
You have to checkout the correct branch - `fix/thorvg`, not `master`.
```sh
git fetch kisvegabor
git checkout --track kisvegabor/fix/thorvg
```

minipost, 2024-07-05:
Tried again on ef3c8daeb228128998136d48056e3c62294c2677
It works

Yangnaan, 2024-07-08:
i ran lv_example_canvas_8(), the result is blank, 
[image]

liamHowatt, 2024-07-09:
Thanks @Yangnaan, I was able to reproduce the issue by `LV_USE_VECTOR_GRAPHIC` to `1` but forgetting to set `LV_USE_THORVG_INTERNAL` to `1`. Can you make sure `LV_USE_THORVG_INTERNAL` is set to `1`?

Yangnaan, 2024-07-10:
> Thanks @Yangnaan, I was able to reproduce the issue by `LV_USE_VECTOR_GRAPHIC` to `1` but forgetting to set `LV_USE_THORVG_INTERNAL` to `1`. Can you make sure `LV_USE_THORVG_INTERNAL` is set to `1`?

it works! thanks a lot

closed 2024-08-03 by lvgl-bot