lvgl/lvgl #6465: Configuration does not work when: lv_conf.h is moved using: build_flags -D LV_CONF_PATH="${PROJECT_DIR}/src/lv_conf.h"

Jvsjvs, 2024-07-04:
### LVGL version

9.1

### What happened?

Hi,

I can configure and run LVGL perfectly, I have ported LVGL version 9.1 to my 7 inch Waveshare board and develop using VSCode with the Espressif platform and the Arduino framework.

When I to put the configuration file in my own project directory using: “build_flags = -D LV_CONF_PATH=”${PROJECT_DIR}/src/lv_conf.h" in my platformio.ini file things go wrong:

It still builds fine, but when I make changes in the lv_conf.g file these changes do no longer have any effect. It feels as if the settings are cached somewhere.
Note that if I put something stupid in the local lv_conf.h file to break the build it indeed breaks, so it does read this local file, but does not make the updates to the settings.
The specific settings I wanted to change is e.g.: LV_USE_SYSMON 1

I think it is a bug somewhere in the macros of LVGL.

Any idea?

### How to reproduce?

1) Move the lv_conf.h to a project specific location.
2) Use build_flags -D LV_CONF_PATH="${PROJECT_DIR}/src/lv_conf.h" in the platformio.ini file
3) Build the project

4) Change a setting in the lv_conf.h file, e.g. enable "LV_USE_SYSMON 1"

5) Rebuild the project and observe that the monitor is not enabled.

Note that if you clean and rebuild everything that **sometimes** the system monitor gets enabled.
If you then disable it in the LV_conf.h and rebuild the monitors remains enabled.

liamHowatt, 2024-07-04:
My first thought is that changes to lv_conf.h are not detected by the build system but it should be treated the same as any other include since lv_conf_internal.h includes it using some macro string techniques.

It would be good if someone could test with PlatformIO. I could not produce it with CMake.

You mention that sometimes even a clean build does not reflect the changes to the file. That's even stranger. And you also said that invalid syntax is always detected.

It does sound like the macro values are being cached somehow.

closed 2024-07-31 by lvgl-bot

sgaechter, 2025-05-08:
Hello together.
I faced the same problem with Platformio, but I managed to shift "lv_conf.h" to my "/src"-Folder by using this build_flag:
-DLV_CONF_PATH="${PROJECT_DIR}/src/lv_conf.h" in the platformio.ini file.

Hope this helps someone.