lvgl/lvgl #6466: Roller with different font size

eme-bennytang, 2024-07-04:
### Introduce the problem

Hi all,

I am developing on lvgl v7.11.0 (not sure if it is true throughout all version
Upon some testing, I quickly realize there are some issues in the roller widget. (or maybe limitation)

1. Text will shifted upwards in the selected text
- I found a discussion about this (https://github.com/lvgl/lvgl/issues/6070) and patched it with the v7 fix.
- Somewhat its ok now, but inconsistent, sometimes its perfectly aligned, but sometimes its off by a few pixels.
- Is updating to v9.1.0 going to fix this inconsistency?

2. Selected Row Height is calculated by (line space in BG part) + (line height for BG part font)
- This means if we are having big difference in font size between the selected and BG, the size of the widget created will be very large (in order to display the whole selected font), while most of them are empty in the BG part and wasted.
- Is it possible to set different row height in selected area and BG part individually?  (in latest version?)
- If I try to fine tune the line space in both part BG and Selected, I can get below result
[image]
- Problem is when I start rolling there will be something like this
[image]
Is there a better way (other than finetuning stuff until it is somewhat "acceptable") to display a big selected font with small BG font?

liamHowatt, 2024-07-04:
As you saw in the issue, a lv_roller.c has been changed many times since v7 to get different size fonts working well. The v7 patch was somewhat lacking, as is mentioned in the issue. The most recent roller fixes (I believe) #6082 for v8, #6083 for v9 (they're identical) should have fixed it for good.

Depending on your comfort level you could try applying the fixe(s) since and open a PR targeting the branch `release/v7`. Otherwise, send a code snippet to reproduce your use case and I can try to fix v7. Thanks.

eme-bennytang, 2024-07-05:
Hi @liamHowatt 

Thanks for the reply.

The main question I am having is
> Is it possible to set different row height in selected area and BG part individually? (in latest version?)
Depending on the answer, we might need to consider updating to v9.1

BG font is Arial Font Size 30
Selected Font is Arial Font Size 74

**Roller creation**
`	sGuiRoller.Roller = lv_roller_create(Par, NULL);
	lv_obj_add_style(sGuiRoller.Roller, LV_ROLLER_PART_BG, &sGui.asStyle[kStyleRlrBg]);
	lv_obj_add_style(sGuiRoller.Roller, LV_ROLLER_PART_SELECTED, &sGui.asStyle[kStyleRlrSelected]);
	lv_obj_set_event_cb(sGuiRoller.Roller, event_cb);

    lv_obj_align(sGuiRoller.Roller, NULL, align, x_ofs, y_ofs);
    lv_obj_set_auto_realign(sGuiRoller.Roller, true);`
**Style initialization**
`psStyle = &sGui.asStyle[kStyleRlrBg];
	lv_style_copy(psStyle, &sGui.asStyle[kStyleCommon]);
	// Text: Light Grey
	lv_style_set_text_font(psStyle, LV_STATE_DEFAULT, &FontArial30);
	lv_style_set_text_line_space(psStyle, LV_STATE_DEFAULT, 58-FontArial30.line_height);
	lv_style_set_text_color(psStyle, LV_STATE_DEFAULT, COLOR_LIGHT_GREY);
#if !DEBUG_RLR_AREA
	// Background: Transparent
	lv_style_set_bg_opa(psStyle, LV_STATE_DEFAULT, LV_OPA_TRANSP);
#else
	// Background: Dark Grey (Debug)
	lv_style_set_bg_color(psStyle, LV_STATE_DEFAULT, COLOR_DARK_GREY);
#endif

	// =======================================
	// Init for roller selected
	// =======================================
	psStyle = &sGui.asStyle[kStyleRlrSelected];
	lv_style_copy(psStyle, &sGui.asStyle[kStyleCommon]);
	// Text: Big font, Red/Blue in color
	lv_style_set_text_font(psStyle, LV_STATE_DEFAULT, &FontArial74);
	lv_style_set_text_line_space(psStyle, LV_STATE_DEFAULT, 58-FontArial30.line_height);
	lv_style_set_text_color(psStyle, LV_STATE_DEFAULT, COLOR_RED);`

**Result**
This  is what I get when scrolling to the smallest and largest option.
[image]
[image]

liamHowatt, 2024-07-05:
> > Is it possible to set different row height in selected area and BG part individually? (in latest version?)

Yes, in the latest version (master/v9.1) or even v8.4, different row height in the two parts was solved after some great effort by kisvegabor in #6083. If the row height is different, the text should now always be centered when scrolling to any item in the roller.

> Depending on the answer, we might need to consider updating to v9.1

I think that is the best solution if you are able to update to v9.1. Even v8.4 has the fix. v7 does not typically receive support anymore.

briangshea, 2024-07-10:
I'm still seeing a slight offset in text when using to very different font sizes. I looked at the code in #6083 and it seems to be the version i'm using. I found that this fix makes an adjustment that seems to work.

[image]

lv_roller.c:476
```
            /*Apply a correction with different line heights*/
            const lv_font_t * normal_label_font = lv_obj_get_style_text_font(obj, LV_PART_MAIN);
            int32_t corr = (label_dsc.font->line_height - normal_label_font->line_height) / 2;

            /*Apply the proportional position to the selected text*/
            res_p.y -= corr*2;  // <-- BGS: When using two very different font sizes, the position was off.
            int32_t label_sel_y = roller_h / 2 + obj->coords.y1;
            label_sel_y += (label_y_prop * res_p.y) >> 14;
            label_sel_y -= corr;
```
with fix above

[image]

When two similar font sizes are used, there seems to be little to no effect with the above update and the text seems to be positioned correctly.

liamHowatt, 2024-07-16:
Sorry for the delay. I missed the notification.

Awesome that you were able to come up with a fix.

> I looked at the code in https://github.com/lvgl/lvgl/pull/6083 and it seems to be the version i'm using.

It does not seem to be the same version. That PR was for `master`, and the commit you made which adds the `*2` reflects the current `release/v9.1`.

Can we just backport https://github.com/lvgl/lvgl/pull/6083 to `release/v9.1`?

[I opened a PR for it](https://github.com/lvgl/lvgl/pull/6515). Please test it if you can and see if the alignment is fixed. Thanks.

PR #6515 (2024-07-22): fix(roller): set the position of the selected text correctly (lvgl#6083)
### Description of the feature or fix

Fixes #6466

Backport #6083 to `release/v9.1`.

### Notes
- Update the [Documentation](https://github.com/lvgl/lvgl/tree/master/docs) if needed.
- Add [Examples](https://github.com/lvgl/lvgl/tree/master/examples) if relevant.
- Add [Tests](https://github.com/lvgl/lvgl/blob/master/tests/README.md) if applicable.
- If you added new options to `lv_conf_template.h` run [lv_conf_internal_gen.py](https://github.com/lvgl/lvgl/blob/master/scripts/lv_conf_internal_gen.py) and update [Kconfig] [... cut]
closed 2024-08-10 by lvgl-bot