lvgl/lvgl #6468: Documentation needs more on lv_img_dsc_t

codewitch-honey-crisis, 2024-07-04:
### LVGL version

8.3

### What happened?

There is precious little documentation on using this structure to display images. It has required quite a lot of experimentation and I'm still not successful at it, despite looking and posting on forums, through google in general, the documentation of course, and trying reams of code to get it to function.

Can we get some documentation on using this structure with lv_img_set_src, or maybe some example code?

Hopefully the code I provided describes some of the problem. The issue is actually severalfold and I'm not sure if carries over to newer versions in terms of the errors and warnings, but the docs haven't really improved since that I've seen.

I don't know why this fails, and even looking through the LVGL codebase I can't figure it out. It's probably something stupid, but all the more reason to document this structure more thoroughly.

Adding, if there aren't warnings or errors being logged during the image rendering process where this fails, there should be.

Adding, for background, all I want to do is precreate an image and then background decompress jpgs onto it ahead of time before I go to render it. So I am creating huge raw bitmaps in PSRAM to do this, but it doesn't show up when i set image source to it.

### How to reproduce?

```c
lv_obj_t* ui_screen = lv_obj_create(NULL);
lv_obj_t* ui_img = lv_img_create(ui_screen);
// works fine:
// lv_img_set_src(ui_img,"A:/minou_480.jpg");
// following silently fails:
lv_img_dsc_t img_dsc;
img_dsc.header.always_zero = 0; // needs to be documented
img_dsc.header.cf = LV_IMG_CF_RAW ; // docs not clear assume RAW RGB565 (native)
img_dsc.header.w = 800;
img_dsc.header.h = 480;
img_dsc.data_size = img_dsc.header.w*img_dsc.header.h*LV_COLOR_DEPTH/8;
uint8_t* img_mem = (uint8_t*)ps_malloc(img_dsc.data_size);
assert(img_mem);
// make black
memset(img_mem,0,img_dsc.data_size);
img_dsc.data = img_mem;
// image does not show up. No warning. No error.
lv_scr_load(ui_screen);
```

XuNeo, 2024-07-05:
I can see your confusion comes from `header.cf`, the color format of the `img_dsc->data`.

Besides, the `img_dsc->data_size` is the data size of `data`, which should be obvious.

As for the color format, most normal ARGB8888 or RGB565 is straightforward. However, `LV_IMG_CF_RAW` means the `img_data` is the raw format of JPEG or PNG file. LVGL's builtin decoder will check the cf and ignore this file, so other decoders can continue to try. That includes the JPEG, PNG decoders.

> all I want to do is precreate an image

If you want to pre-convert the image(jpeg, png) to what lvgl natively supported formats, the `"use this website to generate content"` is the correct way. It outputs the pixel format as you instructed, ARGB8888, RGB565 for example. 

The output could be a binary file or C array source file. The later one can be used as an example for you to explore the rules.

codewitch-honey-crisis, 2024-07-05:
Thanks but I guess I wasn't clear enough - i do not want to embed the images into my firmware. They are coming from SD. The trouble is that LVGL renders while displaying, which is not good for an image that is 800x480. What I need to do is pre-decode the image into a raw bitmap, and then get LVGL to render that. I cannot use the website for that. What I want is to know how to use this structure properly, to make my own memory mapped bitmap. I know it must be possible. If not this is a missing feature in LVGL.

XuNeo, 2024-07-05:
> Edit: I found the issue. There's a magic cookie at the beginning of the BLOB. I need to know what it is.

Make sure your are using the correct tool for lvgl v8. The `magic` word you found is more likely from lvgl v9 tool.

>  What I need to do is pre-decode the image into a raw bitmap, and then get LVGL to render that

LVGL will decode the image itself. How would you decode the image? It's file to decode to raw bitmap, you need to set the header to decoded raw bitmap color format.

codewitch-honey-crisis, 2024-07-05:
It's coming from 8.
Specifically these four bytes at the beginning (HEX). The rest is FF as I expect for an all white bitmap.
04 40 00 04

And I'm confused when you say I need to set the header to the raw bitmap color format. I thought was what I did when you corrected me and told me not to do that.

So why don't you just tell what constant to use? Thanks.

XuNeo, 2024-07-05:
You are using LVGL in an unusual way.

You should let LVGL itself calling the decoder library. Cache the result if necessary.

If the image is decoded, you need to set the `header->cf` correspondingly, instead of `LV_IMG_CF_RAW`.

codewitch-honey-crisis, 2024-07-05:
Maybe I should have filed this differently then.

Because LVGL's image handling is ... undesirable.

I don't want the user to watch the jpg load from top to bottom on the screen. It's ridiculous, and it's why I am NOT simply using lv_img_set_src() with a path to the jpg on SD. Because the result is trash. It's trash. There's no nice way to put it.

So yeah, I'm using it in an usual way because the usual way is not acceptable. 

It still needs to work

"If the image is decoded, you need to set the header->cf correspondingly"

What does that mean? You're being vague. The bitmap is RGB565, no header, no compression. What constant is that, because when I try anything other than RAW, the damn thing crashes.

kisvegabor, 2024-07-09:
> Because the result is trash. It's trash. There's no nice way to put it.

I would put it like like "slow".

> I don't want the user to watch the jpg load from top to bottom on the screen.

In v9 [this PR](https://github.com/lvgl/lvgl/pull/6349) adds a new and faster JPEG decoder to v9. It also caches the decoded image in RAM so it should be faster once it's decoded.

closed 2024-08-03 by lvgl-bot