lvgl/lvgl #6469: In multithreaded program, very easy to crash end in obj_valid_child.isra

Staok, 2024-07-05:
### LVGL version

8.3.6

### What happened?

Embedded Linux env, my multithreaded program, already used mutex to protect all "lv_" functions, very easy to crash end in obj_valid_child.isra

[image]

What could be the possible reasons

and also, very easy to crash when use lv_del_obj(), end in lv_event_mark_deleted()

Staok, 2024-07-05:
I may not all all lv_ fun mutex, but the LVGL is so easy crash when in multithreaded env, A slightly larger project may forget to add mutex lock, can LVGL do this for users?? It also makes lvgl easier to use

Staok, 2024-07-08:
My software is quite large, and I add mutexes to all lv_ APIs, the program no longer crashes. Therefore, in future versions of lvgl, can we implement an internal mechanism similar to mutexes so that all APIs can be thread safe?

kisvegabor, 2024-07-11:
> Therefore, in future versions of lvgl, can we implement an internal mechanism similar to mutexes so that all APIs can be thread safe?

I think it would be huge overhead to lock a mutex an each LVGL API call. :slightly_frowning_face:

closed 2024-08-04 by lvgl-bot