lvgl/lvgl #6474: JSON generator missing data

liamHowatt, 2024-07-05:
### LVGL version

v9.1.1

### What happened?

All fields except `"macros"` are empty in the generated JSON file.

```json
{
    "enums": [],
    "functions": [],
    "function_pointers": [],
    "structures": [],
    "unions": [],
    "variables": [],
    "typedefs": [],
    "forward_decls": [],
    "macros": [
        {
            [...]
```

cc @kds

### How to reproduce?

```sh
python3 scripts/gen_json scripts/gen_json/gen_json.py > lvgl.json
```

liamHowatt, 2024-07-05:
cc @kdschlosser

kdschlosser, 2024-07-05:
OK, i'll go over it and see what got goofed up with it.

closed 2024-07-09 by kisvegabor

liamHowatt, 2024-07-09:
I just realized the command I put in "how to reproduce" is flawed. It should be
```sh
python3 scripts/gen_json/gen_json.py > lvgl.json
```

That's not why it failed though, just a typo for the issue.

liamHowatt, 2024-07-09:
No, it works since the PR, thanks. Just clarifying the typo

kdschlosser, 2024-07-09:
Yeah I saw that you said it was working in the PR but after I had already asked here.

PR #6479 (2024-07-09): fix(gen_json): fix bad LVGL header path
I had missed something in the header path when building for the purposes of generating the JSON output. This is the fix for that.

Fixes #6474 

A clear and concise description of what the bug or new feature is.

### Notes
- Update the [Documentation](https://github.com/lvgl/lvgl/tree/master/docs) if needed.
- Add [Examples](https://github.com/lvgl/lvgl/tree/master/examples) if relevant.
- Add [Tests](https://github.com/lvgl/lvgl/blob/master/tests/README.md) if applicable.
- If you added new options to `lv_conf_template.h` run [lv_conf_internal_gen.py](h [... cut]