lvgl/lvgl #6477: Large stride setting for some displays

cristian-stoica, 2024-07-05:
### Introduce the problem

We have a display that requires a stride value of 64 bytes. For the system to work we need to set LV_DRAW_BUF_STRIDE_ALIGN to 64. Because this setting is used by all buffer allocation functions (through width_to_stride) even small buffers (for letter drawing for example) become very large.

What could be a solution to avoid making large stride requirements and thus avoid using unnecessary large buffers for temporary objects when only the display  needs it?

Going further, how should we separate the stride requirements of object buffers (when they are used by accelerators like vglite and are dependent on hw internals) from stride requirements of display frame buffers?

kisvegabor, 2024-07-10:
What is the use case in which only the display's buffer has special stride requirements? Does it mean that all destination buffers shall have 64 byte stride, but the source buffers (e.g. letters) have different requirements?

cristian-stoica, 2024-07-11:
Yes, for example the buffers for letters don't require 64bytes stride alignment. Other buffers may require only 8,16 or no stride alignment.
 If we set the stride alignment for framebuffer to 64 bytes we can't set it differently for other buffers. All buffers will get the same stride alignment whether they need it or not.

kisvegabor, 2024-07-17:
It's not clear to me how can we decide the stride requirements. I know that it depends on the color format but is it different for source and destination buffers too?

cristian-stoica, 2024-07-17:
The stride requirements can be left to the lv_conf.h as before but I think it will need another macro to differentiate the target buffers.
I have a working prototype that I'll submit soon for review and comments.

cristian-stoica, 2024-07-23:
@kisvegabor The patch depends on [6530](https://github.com/lvgl/lvgl/pull/6530) being right. @nicusorcitu has some reservations on it so we need your review there.

Regardless, after looking at the patches for the last days I'm not particular fond of the solution and not going to send it for now. I'll just say that it duplicates width_to_stride (maybe also buf_align) into framebuffer-only versions that are used only during display initialization.

kisvegabor, 2024-07-25:
I don't understand what is the difference between the frame buffer and a layer which created by LVGL internally to snapshot a widget before rotation/scale. From the GPUs point of view the framebuffer and the internal layers are identical render targets. That's why I'm trying to understand if the different constraints apply to other buffer too (besides the frame buffer).

nicusorcitu, 2024-08-10:
@kisvegabor 
The short story about this ticket is that some display controllers requires a very large alignment (128/192 bytes) for both stride and start address.
```
void lv_display_set_buffers(lv_display_t * disp, void * buf1, void * buf2, uint32_t buf_size,
                            lv_display_render_mode_t render_mode)
{
    LV_ASSERT_MSG(buf1 != NULL, "Null buffer");
    lv_color_format_t cf = lv_display_get_color_format(disp);
    uint32_t w = lv_display_get_horizontal_resolution(disp);
    uint32_t h = lv_display_get_vertical_resolution(disp);

    LV_ASSERT_MSG(w != 0 && h != 0, "display resolution is 0");

    /* buf1 or buf2 is not aligned according to LV_DRAW_BUF_ALIGN */
    LV_ASSERT_FORMAT_MSG(buf1 == lv_draw_buf_align(buf1, cf), "buf1 is not aligned: %p", buf1);
    LV_ASSERT_FORMAT_MSG(buf2 == NULL || buf2 == lv_draw_buf_align(buf2, cf), "buf2 is not aligned: %p", buf2);

    uint32_t stride = lv_draw_buf_width_to_stride(w, cf);
```
Just because the LVGL does not make distinguish between the framebuffer alignment (used by display) and internal buffers (like layers or letter buffers) we will end up using the same large alignment all over.

This ticket was to separate those alignments:
- one for framebuffers
- one for all the other internal buffers

kisvegabor, 2024-08-12:
Got it now. So the difference is that the display hardware has some limitation. What about using `lv_display_set_draw_buffers` to this? To this you can pass `draw_buf`s where the stride can be specified.

nicusorcitu, 2024-08-19:
Your suggestion is basically to have a different implementation of `lv_display_set_buffers`. We can do it, but I was thinking on a more generic solution.

1. Like passing the stride argument in the function `lv_display_set_buffers()`. That should fix the stride issue.
2. What is left is how to align with the start address because I see is being used `lv_draw_buf_align` all over. I mean we can pass a buffer already aligned to 192 bytes but the verification will be done on 64 bytes:
E.G.
#define LV_DRAW_BUF_ALIGN 64 - to align LVGL buffers to 64 bytes
If we pass a FB already aligned to 192 bytes, the below check will be basically useless:
[image]

nicusorcitu, 2024-08-19:
@cosmindanielradu19

kisvegabor, 2024-08-21:
We have [lv_display_set_draw_buffers](https://github.com/lvgl/lvgl/blob/master/src/display/lv_display.c#L406-L414) to resolve these kind of issues. It just saves the draw buffers, where you can set the stride and alignment as you wish.

nicusorcitu, 2024-08-21:
I do understand...
What I mean is that we need to implement our own `lv_display_set_buffers()` and have some kind of the below code:
[image]

kisvegabor, 2024-08-23:
We can't modify `lv_display_set_buffers` as it'd be huge breaking change. Instead we can add something like:
```c
lv_display_set_buffers_with_stride(disp, buf1, buf2, buf_size, stride)
```

cc @XuNeo

cosmindanielradu19, 2024-08-23:
> We can't modify `lv_display_set_buffers` as it'd be huge breaking change. Instead we can add something like:
> 
> ```c
> lv_display_set_buffers_with_stride(disp, buf1, buf2, buf_size, stride)
> ```
> 
> cc @XuNeo

What do you think about adding a setter function only for this property?

Something like:

`lv_display_set_buffers_stride(buf1, buf2, stride)`

XuNeo, 2024-08-23:
For display draw buffer, [lv_display_set_draw_buffers](https://github.com/lvgl/lvgl/blob/master/src/display/lv_display.c#L406-L414) should allow you to customize the stride. Is there issue with that?

kisvegabor, 2024-08-24:
> What do you think about adding a setter function only for this property?

We can't set the stride like this as it affects `draw_buf->data_size` too. 

> For display draw buffer, [lv_display_set_draw_buffers](https://github.com/lvgl/lvgl/blob/master/src/display/lv_display.c#L406-L414) should allow you to customize the stride. Is there issue with that?

I think the problem is only with the extra lines of code and complexity on the user's side.

nicusorcitu, 2024-08-26:
>> For display draw buffer, [lv_display_set_draw_buffers](https://github.com/lvgl/lvgl/blob/master/src/display/lv_display.c#L406-L414) should allow you to customize the stride. Is there issue with that?

> I think the problem is only with the extra lines of code and complexity on the user's side.
>

Exactly the point. Hence a solution to this is not urgent and can be addressed later. For now we can leave one single stride for both FB and internal LVGL buffers.

nicusorcitu, 2024-09-11:
Agree, please close it for now. Thanks.

closed 2024-09-12 by kisvegabor