lvgl/lvgl #6478: custom.cmake cannot install lvgl.pc

paulh002, 2024-07-05:
### LVGL version

v8.4

### What happened?

When I issue sudo make install on raspberry pi I get the error 
`CMake Error at _deps/lvgl-build/cmake_install.cmake:70 (file):
  file INSTALL cannot find "/home/pi/sdrberry/build/lvgl.pc": No such file or
  directory.
Call Stack (most recent call first):
  cmake_install.cmake:47 (include)
`
I don't use this file and also don't know why it is necessary, can it be removed? Or issue fixed. Because cmake will stop and don't install the application.

### How to reproduce?

`sudo make install`

kisvegabor, 2024-07-11:
It works for me:
```
cd lvgl 
mkdir build
cd build
cmake ..
make -j
sudo make install
```

Could you test it?

paulh002, 2024-07-11:
This is not the problem. The make install fails when it is part of an application cmake file and lvgl is included using FetchContent_Declare. The path for lvgl.pc is not correct in that case

kisvegabor, 2024-07-17:
Please a minimal CMake file file which uses `FetchContent_Declare` to reproduce the issue.

paulh002, 2024-07-18:
I made a small project which demonstrates the issue.  
You can clone it from https://github.com/paulh002/cmaketest  

to build it:  
git clone https://github.com/paulh002/cmaketest  
cd cmaketest
mkdir build
cd build
cmake ..
make
sudo make install

Result will be:
pi@pi5:~/cmaketest/build $ sudo make install
[ 45%] Built target lvgl
[ 45%] Built target cmaketest
[ 85%] Built target lvgl_examples
[100%] Built target lvgl_demos
Install the project...
-- Install configuration: ""
-- Installing: /usr/local/include/lvgl/../lv_conf.h
CMake Error at _deps/lvgl-build/cmake_install.cmake:70 (file):
  file INSTALL cannot find "/home/pi/cmaketest/build/lvgl.pc": No such file
  or directory.
Call Stack (most recent call first):
  cmake_install.cmake:47 (include)

make: *** [Makefile:100: install] Error 1

paulh002, 2024-07-18:
changing line configure_file("${LVGL_ROOT_DIR}/lvgl.pc.in" lvgl.pc @ONLY) to configure_file("${LVGL_ROOT_DIR}/lvgl.pc.in" ${CMAKE_BINARY_DIR}/lvgl.pc @ONLY) seems to correct the problem.

kisvegabor, 2024-08-08:
Not stale!

@paulh002 can you send a pull request with this fix?

paulh002, 2024-08-08:
ok

kisvegabor, 2024-08-29:
Not stale

PR #6787 (2024-09-11): fix(cmake) fix install
Fixes #6478
Needs to be back ported to v8

### Notes
- Update the [Documentation](https://github.com/lvgl/lvgl/tree/master/docs) if needed.
- Add [Examples](https://github.com/lvgl/lvgl/tree/master/examples) if relevant.
- Add [Tests](https://github.com/lvgl/lvgl/blob/master/tests/README.md) if applicable.
- If you added new options to `lv_conf_template.h` run [lv_conf_internal_gen.py](https://github.com/lvgl/lvgl/blob/master/scripts/lv_conf_internal_gen.py) and update [Kconfig](https://github.com/lvgl/lvgl/blob/master/Kconfig).
- Run `scripts/code-format.py` (`astyle v3 [... cut]
closed 2024-09-11 by kisvegabor