lvgl/lvgl #6480: Fix docs build warnings

kisvegabor, 2024-07-05:
### Introduce the problem

When building the docs with there are a lot of warnings. See e.g. [here](https://github.com/lvgl/lvgl/actions/runs/9811162294/job/27092842848).

Some of them are Doxygen related (missing comments in LVGL's code) other are Sphinx related (syntax error, broken links, etc)

### Proposal

As the warnings are quite diverse we should go through them manually and see how we can fix them. 

In email Victor Wheeler was interested in this task, so I opened this issue to discuss it in a public way.

cc @kdschlosser

vwheeler63, 2024-07-10:
Preliminary findings on the Doxygen part of the docs generation:

1. Doxygen warnings are turned off.

2. Turning on Doxygen warnings makes no difference.  No Doxygen warnings are generated.

3. The only (apparently) unintended thing going on with Doxygen (which doesn't really matter because these files don't appear in the final documentation) is that 3 files are abusing the Doxygen `@file` command by including a filename as an argument (not needed unless it is providing documentation for a DIFFERENT file, so the proper way to guarantee Doxygen generates documentation about THAT file is to only use the bare `@file` command without arguments).  The 3 filenames changed, but the `@file` arguments did not change:

- ./examples/porting/lv_port_disp_template.h
- ./examples/porting/lv_port_fs_template.h
- ./examples/porting/lv_port_indev_template.h

In each case, I suspect these files originally ended as "...templ.h" (which is the case with the `@file` arguments) but now the filenames spell out the word "template" but the "...templ.h" remains in the (unneeded and unwanted) `@file` argument.

These "may" be worth changing for code consistency, but not because they will impact the final documentation.

4.  I see most of the actual LVGL .h files (e.g. below the ./src/ directory) are loaded with Doxygen comments, but Doxygen (in this case of this after-commit docs build) does not actually process these files.  Is it in the plans at all to add raw API documentation to the generated documentation?

5. All the remaining errors come from Sphinx (which I have not looked into deeply yet).

--Vic

kdschlosser, 2024-07-10:
the build system for the documentation generates a lot of stuff so it is hard to tell just by looking at the source files what is actually happening. There is a lot of errors/warnings that are generated because of incorrect documentation in the C files.

The easiest way to deal with this because of the very large number of errors is to go through them one at a time and solve each one of them.

A lot of the sphinx errors and warnings exist because of the headers not being used correctly. It is not hard to fix those but it does take some time to do becaus they need to be done one at a time compiling between each one to see if the correction has caused additional headers down the line to be incorrect. It's tedious.

There is also some syntax errors because of the C primers being used incorrectly as well. This is a matter of just going in and correcting them,.

It is going to take a while to fix the problems because of the sheer number of them. This is the reason why I recommend having the documentation build occur with each and every push to a PR and if there is an issue it should fail the build. The PR should not be merged until the docs build is error/warning free. This will keep the problems from piling up like what has happened. It makes more of a headache having to fix hundreds of errors.

kdschlosser, 2024-07-10:
All of the documentation that is located in the C header files and source files does get added to the documentation via sphinx. This is where some of the complexity comes from because Spinx is not designed to the documentation from C files. So doxygen is used to generate XML output and that XML output then gets read and rst files get generated from them.

The whole parsing of the XML files is not only used for the documentation build. It also gets used for the gen_json script which builds a JSON representation of the LVGL API which also includes the documentation. This is a feature that will be able to be used by bindings to make it easier to write bindings that generate code instead of having to hard code a binding.

kisvegabor, 2024-07-11:
If it doesn't make an issues we can remove these preambles from all files:
```c
/**
 * @file lv_obj.c
 *
 */
```
I thought it's required for Doxygen. (Maybe it was many years ago)

---

Once the docs is error free, I agree with Kevin to build it on all commits in PRs too, so that we can maintain its perfection :slightly_smiling_face:

vwheeler63, 2024-07-11:
@kisvegabor To clarify, the @file command IS REQUIRED in order for Doxygen to process it -- just not the filename argument.

vwheeler63, 2024-07-11:
@kdschlosser Good afternoon, and thank you for your input.  It is clear you have more knowledge about the docs generation than I do.  And I am willing to take the time to fix/compile each error.  From what you described, I think it will be well worth the time in the end.

I see that my Doxygen conclusions were premature, because the `Doxyfile` I was using was the result of

```
doxygen -u
```

(upgrade) which left the INPUT variable empty!  **rolling my eyes at myself**

Please allow me to dig a bit deeper.  I think I am going to like this project.

Also, as an afterthought, between you and Gabor -- I RECENTLY went through the entire LVGL learning curve from "nothing" to having fairly in-depth knowledge of LVGL v8.3 (now v8.4).  This recent experience, and discovering what was difficult about learning it might also have rewarded me with a "gem" of insight about where documentation could be improved....  I'm also a FreeRTOS expert.  I would suggest we consider a separate task later to improve the documentation content, because the time is "ripe" for me to be able to contribute to that.  (The code EXAMPLES are WONDERFUL by the way!  They helped a LOT!)

I'll get back with more findings soon.  (I have my plate fairly full with another motorcycle event during the next 3 days, but I am hopeful I can play a key role in removing all the warnings from the docs compilation!)

kdschlosser, 2024-07-11:
I am the one that wrote most of the current document build system. 

The content does need to be improved i do agree with you there. and as you said that is a separate issue and can be done at a later time.

If you go to the published docs and change the LVGL version to 8.3.x you will see the difference between the old way and the new way the docs were built. It is way more organized and there is linking that is done for functions and data types that are in the documentation where as before there was not. The API portion of the documentation is also formatted a lot better.

Ideally I would like to make some kind of a marker that could be used in the doxygen doc strings that points to the `.rst` files where the documented feature is used. This will tell the person that is making a code change what rst files need to be updated as well. This is where a lot of the issues are coming from. API is getting modified but all of the locations in the `.rst` files are not being updated to align with the changes.

A PR should not be merged if the documentation fails to build. I have already added the mechanics into the build system to detect doxygen warnings and errors and also sphinx warnings and errors which can trigger the build to fail. This is turned off at the moment but it can be turned on very easily. The way the system is written is it will allow the build to complete so it shows all errors when it fails instead of failing when a single error occurs. This enables the user to go and correct everything at one go instead of having to fix one thing at a time which would be exceedingly annoying to do. 

I need to draft up some instructions as to how the C primers work and also how the headings work as those seem to be a sore spot and cause a lot of issues. A basic tutorial of what to use and when and what not to do when using a primer or heading. simple things like the heading marker MUST be the same length as the text written above it (or below in some cases).

kdschlosser, 2024-07-11:
I am dealing with an injury to my right hand.. While it is improving it is still hindering me being able to use a keyboard properly. This makes it hard for dealing with this kind of a thing.

If I had help even by a single person and a location where the both of us would be able to make changes to the code and then submit a PR for the changes... That would be the ideal way to go. IDK how to go about doing that with Github or if it can even be done. I don't think that GitHub has permissions that are refined enough to give write permissions to a single branch or fork. I know with organizations write access can be given to a single repo but I am not sure if it can be done to a specific branch.

vwheeler63, 2024-07-12:
@kdschlosser Hi, Kevin!

Re old vs new documentation, yes, I see new sections:
- Integration and Drivers
- API!
- Coding Style (nice!)

Observation:  this page:  [https://docs.lvgl.io/master/API/core/lv_obj.html](https://docs.lvgl.io/master/API/core/lv_obj.html)   gives the "overview" of the `lv_obj_t` API that I missed so much while I was learning v8.x!   **wheezing sigh of relief -- happy to see this**  I lost count of the number of times I went to `lv_obj.c` and had to read the function definitions to figure out how to use them (and dangers/costs involved, when applicable -- and I have a lot to add about those if I get into the position of enhancing the documentation content at some point).

Re "Ideally I would like to make some kind of a marker...", anchors and web links can be placed anywhere using Doxygen, so (at a glance) that should be easy (hopefully).  (I use anchors and links heavily in my own documentation.  Here is a handy example.  [http://crystal-clear-research.com/for_gabor/tasks/lin/___l_i_n_manager_8c.html#_linmanager_vocabulary](http://crystal-clear-research.com/for_gabor/tasks/lin/___l_i_n_manager_8c.html#_linmanager_vocabulary)   (Please forgive there not being a "main page" because this is just a "subsection" of a larger Doxygen "document" meant to be the whole software project.)  Check out the links that look like [Section 2.3.1].  They link to an anchor in a LIN v2.2A specification document where the terms are used -- much like what you describe.  "LIN" stands for "[Local Interconnect Network](https://en.wikipedia.org/wiki/Local_Interconnect_Network)" -- an automotive network used where speed and reliability are not terribly important, used because it uses very inexpensive hardware and wiring [single wire between network nodes].  I've been working exclusively in the automotive firmware industry for the last 12.5 years.)

Re "I have already added the mechanics into the build system to detect doxygen warnings and errors..." I assume you are referring to the `WARN_FORMAT            = "WARNING: $file:$line: $text` in `Doxyfile`(?).

Re "...exceedingly annoying to do".  Agreed.  The impossible-to-live-with efficiency issue blocking a PR strongly points to the need for a more efficient design, as you have described.

Re "A basic tutorial of what to use and when and what not to do when using a primer or heading. simple things like the heading marker MUST be the same length as the text written above it (or below in some cases)."  I'm not sure what is meant by "primer" and "heading" in this context.  Can you clarify?

Re GitHub features:  understood.  I myself (for the last 4-5 years) have been successfully using the [GitFlow](https://nvie.com/posts/a-successful-git-branching-model/index.html) branching model in my own projects.  So if you and Gabor find it comfortable, we can create a "fix" branch which I can submit PRs to if that helps.

Some other relevant qualifications of mine:

- FreeRTOS expert.
- Qualified as an Editor in English (**extreme** depth understanding and practice with English grammar, making writing clear technical documentation very easy for me).
- I type at ~110 words per minute with very few typos.
- Recently been through full learning curve from zero to proficient with LVGL v8.x.
- Wrote my own LVGL widget based on `lv_scale_t` to represent an automotive dashboard gauge.  The reason I couldn't use `lv_scale_t` was that I needed the drawing to go in sequence from back to front, so I introduced FACE layer, NEEDLE layer and GLASS layer (each layer with its own object Z-list) and kept the 3 Z-lists in drawing order from back to front (letting user introduce back-to-front objects on each layer based on sequence of adding to the gauge).  That lead me to a very deep dive into `lv_scale_t` and is why I submitted several bug fixes to `lv_scale_t` to Gabor (@kisvegabor ) on the v8.x branch several months ago to bring `lv_scale_t` in alignment with the documentation page about it.

Kind regards,
Vic

P.S.  I have today available, but tomorrow will be away from my desk all day at a motorcycle event -- a Poker Run to be specific (I'm the organizer).

vwheeler63, 2024-07-12:
[Here you can see my LVGL widget in action](http://crystal-clear-research.com/for_gabor/20231018_140500_Analog_Rev_Counter.mp4).  Note how the needle is ABOVE the items on the gauge FACE.

kdschlosser, 2024-07-12:
Oh if you have been working in the automotive industry with the electrics in vehicles then we have a lot in common. I retired from doing that work, It's what I did for almost 30 years. I am one of the people that was involved in the design of transponder bypass modules and all kinds of integration pieces that are used in aftermarket upgrades to vehicles. I am full aware of what LIN is and also MOST, CAN, MSCAN, SWCAN, HSCAN, CANFD, etc... I know some of the high level protocols used by the different manufacturers as well like GMLAN, GLOBAL A, ClassII, SCP, etc.... and also the OBDII spec. I actually have a bunch of CAN based stuff on github that I have written. go and take a look at my repos...

I also do vehicle mechanics as well and not just the electrical end of things. So it sounds like we have a huge amount in common...

The fact that you are willing to help out with the docs is an up and up. Always looking for ideas and new ways to do things. I am sure that @kisvegabor would agree here as well. The docs are better than what they were and it is going to keep on improving IF we keep things up to date and don't let the errors go by unfixed. There are just so many of them right now it is a big animal to tackle to fix all of them and even more so with only 1 person doing the fixing.

If @kisvegabor can create a branch dedicated to fixing the documentation and we can submit PR's to that instead of possibly causing problems in the master branch it would be really helpful. Make sure that any PR's merged into the master branch do not have any changes to the documentation in them. changes to the docstrings in the C header files is fine but no changes made to the RST files. submit a separate PR to the branch for fixing the docs for any updates to those. This way we don't have a nightmare of merge conflicts to deal with.

With 2 people working on it it can be knocked out pretty quickly. Just to get the build to work without any errors or warning. Get that merged into the master branch and then we can go in and correct any formatting issues and dead link issues. Then get that merged and after that we can work on adding new things to spruce it up. While I do like the output of sphinx over doxygen there does need to be some customization done for the HTML and CSS to make the design unique to LVGL. I don't like using already made templates for the final output because it ends up looking the same as everyone else.

kdschlosser, 2024-07-12:
I do feel there should be a section dedicated to advanced drawing. Things like this...

https://www.youtube.com/watch?v=NxQ7cLobr_Q

and like this...

https://www.youtube.com/watch?v=Ko-yzwjXRX4

and here is yet another...

https://www.youtube.com/watch?v=gIw8Pf8KtgY

People really need to see what is possible and how versatile LVGL really is. While you do see that with a lot of other GUI frameworks none of them have any documentation that goes over doing the advanced things are seen.

vwheeler63, 2024-07-13:
Hi, Kevin (@kdschlosser ) !

Re us having automotive electronics in common -- how cool!  I bet we'd have a heck of a shop talk discussion in person....

What you describe, including limitations on what gets changed (for this task) sounds perfect.

By the way, I noticed in your GitHub profile you're from "Rocky Mountain, USA".  I'm from north Denver (previously from NE Boulder).  2 weekends ago I took a group of 16 motorcycles on 3 trips into the Rockies in and around Estes Park, and the 3rd group into Rocky Mountain National Park....  I'm wondering if we are close....

--Vic

kdschlosser, 2024-07-13:
I am in Evergreen, Wow that is some really crazy stuff right there!!!. Evergreen borders Golden and Morrison

vwheeler63, 2024-07-13:
@kdschlosser Got it regarding proof of concept.

Re Evergreen:  fascinating!  I've ridden a motorcycle about 30,000 miles all within Colorado, and Evergreen, Kittredge, Conifer, Bergen Park, and the Squaw Pass to/from Mount Evans used to be regular "stomping grounds" for me.  One of my favorite restaurants in Kittredge (2.1 miles downhill from Beau Jo's Pizza in Evergreen) was "Country Road Cafe".  They used to have lines to get in on Sundays (they had outdoor dining under a tent on the east side).  The location is now called "Cabins at Country Road".  Small word, eh?  :-)

kdschlosser, 2024-07-13:
Sorry for not getting back to you sooner. I was dealing with a wife issue and her wanting to go out to eat because she didn't want to prepare food for herself. I was also finishing up working on one of my Lincolns..

Then I was dealing with an issue with dumbass Windows 11 chewing up 32GB of ram for what is called "Standby Memory" which is nothing more than a cache in the event you reopen something. Waste of memory when you have a 32 logical processor machine.  Because that was driving my memory use way up Windows then turned on memory compression which made the entire machine run slow as hell. Mouse studdering and all. There is no way to turn off that standby memory but it can be cleared. I didn't want to have to launch an application and click a bunch of buttons and what have you to clear the memory. 

I wanted to do it programmatically once an hour. So I wrote a python script that interfaces with the Windows DLL's using FFI to make the function calls needed to clear that memory. I have that just running in a loop that waits an hour and then clears it. I will write a service to do this so I don't have to remember to launch the program on boot and I don't have another application open. I could use the task scheduler, but I really do hate that thing because it is where all kinds of malware likes to get itself buried into. 

Microsoft trying to micromanage the memory and ends up making a mess of it, no surprise there....

kdschlosser, 2024-07-13:
If you know where Maxwell Falls lower trail head is that is where I live, across the street at the top of the hill.

vwheeler63, 2024-07-14:
Beautiful!  I've been up and down Hwy 73 between Evergreen and Conifer many times, but never up in that neighborhood to the west.  That explains the tall pines and steep hill in the background of your photo!  Truly lovely area!  I envy your location!

kdschlosser, 2024-07-15:
That restaurant you mentioned, the county road one originally was a bed and breakfast. It actually was the entire time. When COVID shut them down they decided to not open the public side of the restaurant again. It wasn't because of money or anything like that. The bed and breakfast with the cabins still exists and you can still get the breakfasts they make if you rent a cabin. The restaurant came along because of all of the people telling them they should open one because the food was sooo good. When COVID closed it down they probably realized how much nicer life was without having to deal with the restaurant portion of it.

They did have some really good breakfasts though. I have been there many times and I was upset they no longer had it.

I moved to Evergreen because I don't like people too much. Well maybe it's not the people that i don't like it's the stuff they do. IDK.. I am allergic to stupidity is what it is... LOL.

I use Windows because it is almost 80% of the market share. and with the world of virtualization I am able to run any other OS at the same time. This is the reason why I bought a SuperMicro AMD Threadripper Pro workstation. It can hold a monster amount of RAM (2TB) and accepts the 64 core processor. has a 2000 watt power supply and 6 PCIe x16 slots (actual x16 slots) along with 4 PCIe NVMe sockets. The whole machine is manageable with IPMI and hardware level KVM/IP. 

Here is a link to the branch I created for LVGL. You should be able to pull code changes from that branch as well as submit PR's to it. Once we get the documentation building without error I will open a PR to the LVGL master branch.

https://github.com/kdschlosser/lvgl/tree/documentation-fixes

vwheeler63, 2024-07-15:
Re Country Road Cafe -- that makes sense.  Fascinating that it sounds like it was a lot of people's favorite restaurant, and I remember the Mother's Day Line to get in there was enough to motiviate getting reservations....  I did love the breakfasts.  I used to ride up there with no plans, have breakfast, and plan the "route" for the rest of the day at the breakfast table....

I'm with you 100% on being allergic to stupidity!!  The last 4 years have been an amazing eye opener for me regarding the environment (and people) around us....

That is one amazing system.... That's why I use Windows too, but never had the need (yet) for anything fancy hardware-wise.

Re the branch -- sounds good.  I'm heading to bed so will dig in in the morning!

Seeking guidance:  would it be appropriate to clone directly from your fork, or make my own fork from it?  (GitHub PRs are somewhat new to me, so I have learned just enough so far to be able to submit one to Gabor using my own fix/xxxx branch.)

Regards,
Vic

kdschlosser, 2024-07-15:
IDK how the whole thing with github works. I use PyCharm as my IDE and I know I am able to clone a repo using that and then I can add remotes after that. Once a remote is added I am able to checkout a branch on that remote and then create a new branch from that remote. Once the new branch is made I can make code changes and commit them. Then I flip over to github in my browser and I am able to submit a PR using that branch and when doing that I am able to key in the fork and branch where I want the PR to go.

The actual git commands to do this... ??? I would have to look them up. Maybe @kisvegabor might be able to chime in on how to go about doing that.

I have made some code changes to the branch already. I fixed the document build so it works again using Windows and I also added it outputting information to files. Not all of the errors get output to files, only the things that are missing doxygen docstrings get output.

vwheeler63, 2024-07-15:
@kdschlosser Oh, I know how to do clone and fork -- no problem.  My question was merely about if you want me to be able to push directly into YOUR fork of lvgl/lvgl, or if you would like me to push to my own fork (of your fork) and then submit PRs to your fork of lvgl/lvgl?  (It's more steps to do it the latter way, but it allows you to inspect all the changes before committing them to your branch.  Up to you.)  I concur that @kisvegabor might be able to advise....

I welcome the changes that make it work under Windows.

kisvegabor, 2024-07-15:
Here in the `lvgl` repo only the user who opened the PR and the maintainers of LVGL can push to PR. I see two solution:
1. Create a branch in the LVGL repo where we can merge smaller PRs
2. Let's say Kevin has an LVGL fork where Vic opens a PR there. This way both of you can commit to the PR.

I think option 2) is better.

---

I love to see having great conversations and see if like-minded people meet. However to keep the issues easy to follow please use the forum's [General Discussion](https://forum.lvgl.io/c/general-discussion/2) category for the topics not related to the issue.

vwheeler63, 2024-07-15:
@kisvegabor I concur.  Thank you for the tip about General Discussion.  I've been looking for a way to communicate with Kevin (@kdschlosser) directly for things not related to docs generation.  Is there a way to exchange email addresses without making them public?

vwheeler63, 2024-07-15:
@kdschlosser Kevin, I will roll with plan #2 (I will make a fork of your fork, and then push my changes to my fork), which then allows me to submit a PR into your fork on the designated branch).

kdschlosser, 2024-07-15:
we can do PM's if you sign up for LVGL's forum.

vwheeler63, 2024-07-15:
@kdschlosser -- by "primer", am I understanding correctly that you mean "a small introductory book on a subject"?

kdschlosser, 2024-07-15:
primer is this,,,,

```
:cpp:expr:`void lv_lottie_set_buffer(lottie, w, h, buf)`
```

it's the `:cpp:expr:` part. that's the primer or directive whichever you want to call it.

vwheeler63, 2024-07-15:
A Sphinx command?

kisvegabor, 2024-07-17:
> @kisvegabor , can you please confirm that is your understanding (i.e. how you would like us to manage it)?

This forking flow should work. 🙂

kdschlosser, 2024-07-18:
@kisvegabor 

This doesn't look right...

```c
typedef struct {
    lv_prop_id_t id;
    union {
        int32_t num;                /**< Number integer number (opacity, enums, booleans or "normal" numbers)*/
        const void * ptr;           /**< Constant pointers  (font, cone text, etc)*/
        lv_color_t color;           /**< Colors*/
        lv_value_precise_t precise; /**< float or int for precise value*/
        lv_point_t point;           /**< Point*/
        struct {
            /**
             * Note that place struct member `style` at first place is intended.
             * `style` shares same memory with `num`, `ptr`, `color`.
             * So we set the style value directly without using `prop.style.num`.
             *
             * E.g.
             *
             * static const lv_property_t obj_pos_x = {
             *      .id = LV_STYLE_X,
             *      .num = 123,
             *      .selector = LV_STATE_PRESSED,
             * }
             *
             * instead of:
             * static const lv_property_t obj_pos_x = {
             *      .id = LV_STYLE_X,
             *      .style.num = 123, // note this line.
             *      .selector = LV_STATE_PRESSED,
             * }
             */
            lv_style_value_t style; /**< Make sure it's the first element in struct. */
            uint32_t selector;      /**< Style selector, lv_part_t | lv_state_t */
        };
    };
} lv_property_t;
```

and it's causing this error in the doc build...

```
C:\Users\drsch\AppData\Local\Temp\tmpk87bkp6j.lvgl_docs\API\core\lv_obj_property.rst:8: WARNING: Error in declarator or parameters-and-qualifiers
If pointer to member declarator:
  Invalid C++ declaration: Expected identifier in nested name. [error at 19]
    union lv_property_t
    -------------------^
```

It could be because of the nested anonymous union and the nested anonymous structure inside of the union. I am not sure.

kdschlosser, 2024-07-19:
@kisvegabor 

I have almost all of the issues sorted out. It shouldn't be too much longer until it is error free...

kisvegabor, 2024-07-19:
> I have almost all of the issues sorted out. It shouldn't be too much longer until it is error free...

Really? That's huge! Have you added comments to all struct and enum members?

It'd be nice to see together all the non trivial special cases (like the anonymous unions) so that we can discuss them all at once. 

I have no problem with adding the docs build to the CI. Maintaining the docs/comments incrementally it's not big burden.

kdschlosser, 2024-07-19:
No I haven't added the documentation for all the things that are missing. IDK what everything does or how it works. That is going to be better suited for someone like yourself to do. The other reason as to why is I don't know what is going to be moved to being private. If it's private it doesn't necessarily have to be documented because it is not in the public API. 

What I did was I made is so that it captures the current state of the documentation the first go around and that information will be cached in the CI any PR made after that point will use those cached files to see if there are any new things or changes that have been made that have not been documented and it will fail the build. So the things that are currently not documented would still remain that way until someone with more knowledge on the inner workings has the time to document them. It will stop new undocumented things from being added to the code. It's a step in the right direction. 

I have almost all of the errors created by Sphinx all sorted out. @vwheeler63 is fixing the errors caused by doxygen.

kisvegabor, 2024-07-22:
Okay, clear! I thought missing function or field docs are causing Sphinx errors too.

vwheeler63, 2024-07-22:
@kdschlosser Kevin, thank you for the build fixes thus far!  I have your latest as of 22:13 last night (commit 465df94 on `documentation-fixes` branch).  Let me know (preferably by phone text so that it is timely) if there are any more coming. 

--Vic

kdschlosser, 2024-07-22:
OK so now we have an even larger issue at hand.

There is no development taking place for Breathe which is the library that is used to convert the doxygen xml output into something sphinx is able to use... I have come across several bugs in Breathe and it would appear as tho these are not going to be fixed due to Breathe no longer being maintained because of not being paid to do so.

https://github.com/breathe-doc/breathe/issues/990

This is going to make things a whole lot more difficult. It is going to keep up from doing things like updating sphinx versions in order to get the latest bug fixes and updates. So now we are going to have to either write our own code that does what Breathe does or we are going to have to fork Breathe and maintain that fork so it will work properly.. OYE!!! This really sucks....

vwheeler63, 2024-07-22:
@kdschlosser @kisvegabor   Let's put our heads together.  I have little knowledge of what is in `breathe`.  Is it big and complicated because it is trying to deal with so many features of Doxygen?  Do we use a mere subset of those features that we could do a better job of?  Perhaps with better tools than were available to Michael Jones 15 years ago?

kdschlosser, 2024-07-22:
Sphinx is designed to build the documentation for python. By using it's ability to be extended breathe reads the doxygen xml output and is able to inject it into sphinx. The reason why sphinx is used because it provides a nicer looking documentation that is easier to use.

It does more than just the API documentation. we are able to link to functions, structures, enumerations by using directives. This is a nice feature to have. This allows the documentation that is written in the ReST files to connect with the API portions.

If you click on the link below and then read the next line, at the very end there is a function that is mentioned. Now click on that function.

https://docs.lvgl.io/master/porting/log.html#custom-log-function

Those are the kinds of things that sphinx does. However it is designed to do that for python code only. The breathe addon extends that functionality so it works with c code as well.

kdschlosser, 2024-07-22:
To my knowledge breathe is the only thing made that does this. There are other extensions for sphinx that will generate documentation from c code using doxygen but none of them function to the extent that breathe does.

vwheeler63, 2024-07-22:
After following the link, and then the link to the function -- that is nice.  You'd think it would be wildly popular... enough to keep it financed.

vwheeler63, 2024-07-22:
@kdschlosser I notice the Linux Kernel uses Sphinx for its documentation.  Do you happen to know its data flow?  (I don't know if any of the documentation "lives" in the source code or not.  If so, how they do it might be of interest.)

kdschlosser, 2024-07-23:
I have been really looking at breathe and it would not be a big deal to fork it and make updated to it to fix bugs and to add new features to it if we wanted to.

I did figure out that the XML files generated when doxygen gets run the first time is what gets used in breathe. So that is kind of nice that it does that.

vwheeler63, 2024-07-23:
@kdschlosser Re `breathe` and new features (and bug fixes):  Excellent!

Re XML files -- that makes a lot of sense (since I observed it generates that whole directory of XML files on that first pass).  Oh yea, we didn't see any text output on that run where you added `print()` calls did we...?

vwheeler63, 2024-07-23:
@kdschlosser  Good morning, Kevin!

I was going to gather information from the actual documentation to show Gabor @kisvegabor  how a large handful (134 plus a number of the "is not documented" errors) of the 3181 warnings Doxygen is currently resulting in elements being OMITTED from the current documentation.  (List of generated Doxygen warnings in attached file.)  While doing so, I found something concerning -- not sure if it's `breathe` or `sphinx` or perhaps a parameter(?):  The following is from ./src/misc/lv_style.h:

```
/**
 * Possible options how to blend opaque drawings
 */
enum _lv_blend_mode_t {
    LV_BLEND_MODE_NORMAL,     /**< Simply mix according to the opacity value*/
    LV_BLEND_MODE_ADDITIVE,   /**< Add the respective color channels*/
    LV_BLEND_MODE_SUBTRACTIVE,/**< Subtract the foreground from the background*/
    LV_BLEND_MODE_MULTIPLY,   /**< Multiply the foreground and background*/
};
```

This is correctly documented for Doxygen.  These strings are not making it into the final documentation.  They ARE making it into the <tmpdir>/xml/lv__style_8h.xml file generated by Doxygen:

```
        <enumvalue id="lv__style_8h_1ac7a9af626799558a27cae4315c5148ceaad57a0fd5e04f5691d8f417ee5434b7e" prot="public">
          <name>LV_BLEND_MODE_NORMAL</name>
          <briefdescription>
          </briefdescription>
          <detaileddescription>
<para>Simply mix according to the opacity value </para>
          </detaileddescription>
        </enumvalue>
```

But they are not here in the resulting documentation:

[https://docs.lvgl.io/master/API/misc/lv_style.html#_CPPv415lv_blend_mode_t](https://docs.lvgl.io/master/API/misc/lv_style.html#_CPPv415lv_blend_mode_t)

If you click the little arrow to "open" the list of enumeration values with their docstrings, the list is empty!

What I was going to show Gabor was the fact that the docstrings are missing from the INCORRECTLY-documented under the various

```
#define LV_STYLE_PROP_FLAG_...
```
macros.  The documentation error causes the docstrings to be invisible to Doxygen, which then cannot pass them on to `breathe` and subsequently to `sphinx`.

[_doxy_warnings.txt](https://github.com/user-attachments/files/16348993/_doxy_warnings.txt)

vwheeler63, 2024-07-23:
@kdschlosser Kevin, if you haven't already done so, would you please update our `documentation-fixes` branch and let me know when it is ready!  (Reason:  I'd like to minimize any conflicts that may arise from the changes I'm about to make.)

kdschlosser, 2024-07-23:
This is the problem I am having...

```
API\misc\lv_style_gen.rst:8: WARNING: Invalid C++ declaration: Expected end of definition. [error at 47]
  LV_ATTRIBUTE_EXTERN_DATA const lv_style_prop_t _lv_style_const_prop_id_ROTARY_SENSITIVITY
```

It is something with LV_ATTRIBUTE_EXTERN_DATA because when I remove that the error doesn't happen. That macro should be evaluating to nothing and I am not sure why the macro is even showing up. The error is being generated by Sphinx but for some reason Doxygen is not expanding the macros when it should be.

vwheeler63, 2024-07-24:
@kdschlosser I found something that MIGHT be pertinent.  A week ago, this item in the edited `Doxyfile` (in the temp directory)

```
PREDEFINED             = DOXYGEN LV_CONF_PATH="R:\lvgl\docs\lv_conf.h"
```

used to look like this:

```
PREDEFINED             = DOXYGEN
                  LV_CONF_PATH="R:\lvgl\docs\lv_conf.h"
```

Did something change that?  It might be causing Doxygen to miss including `lv_conf.h`!

kdschlosser, 2024-07-24:
That shouldn't cause a problem because the `\` is to escape the newline. so long as there is a space between them then it will work fine.

I know for a fact that lv_conf.h is not missed because if it was there would be an error that is generated for it not being available....

kdschlosser, 2024-07-24:
I know where the problem is... It's because c variables are not supported using sphinx. That's why the errors come about.... I am going to not document that file is all. Not a huge deal as there isn't anything in there that I believe is public anyhow.

kdschlosser, 2024-07-24:
@kisvegabor 

The errors that were generated because of the styles were because sphinx doesn't have any way to deal with variables in C code. All of the constant variables that were declared in `misc/lv_style_gen.h` I have moved into misc/lv_style_gen_private.h. Since private header files are omitted from the document build and also because all of the variable names are prefixed with an underscore this seemed like the nest approach to do away with the errors.

I am going to push a commit to the PR shortly,  I am just making sure the build is in fact 100% error free at this point... Well it's not 100% error free but none of the errors that are happening are going to cause the build to fail and none of the errors actually get output. The remainder of the errors reside in the doxygen docstrings and can be handled through a separate PR.

kdschlosser, 2024-07-24:
@vwheeler63 

I am not sure why you said that `lv_blend_mode_t` doesn't show up in the generated documentation. It does...

[image]

kdschlosser, 2024-07-24:
OK so I have managed to work some magic and I am able to monkey path breathe. So if we encounter any behavior that we want to alter or bugs that need to be fixed we are able to do it without having to maintain a separate repository. YAY!!!..

One of the things I did notice is that if there are functions that have not been documented it appears that they don't show up on the API section of the documentation. I am in the process of fixing this problem. I am going to make sure that all of the API shows up properly with or without there being doxygen docstrings.

vwheeler63, 2024-07-24:
> That shouldn't cause a problem because the `\` is to escape the newline. so long as there is a space between them then it will work fine.
> 
> I know for a fact that lv_conf.h is not missed because if it was there would be an error that is generated for it not being available....

Got it -- glad to hear this.

> I know where the problem is... It's because c variables are not supported using sphinx. That's why the errors come about.... I am going to not document that file is all. Not a huge deal as there isn't anything in there that I believe is public anyhow.

You probably know this, so forgive me if I am preaching to the choir:  It can also be done by:

```
#ifdef DOXYGEN
modified line
#else
original line
#endif
```

(Will discuss the other posts in different replies.)

vwheeler63, 2024-07-24:
> @vwheeler63
> 
> I am not sure why you said that `lv_blend_mode_t` doesn't show up in the generated documentation. It does...

Wow -- my apologies.  I know what happened:  I did the "drop-down" on the `lv_blend_mode_t` enumeration and the other links (for the enumeration values below it) had their "triangles" already in the "down" position, which made me think they were empty.  I can confirm the strings to the right of each enumeration value are present.

And that gets me to the point I wanted to show Gabor @kisvegabor that the docstrings to the right of the `LV_STYLE_PROP_FLAG_...` macros are missing due to Doxygen documentation errors (which I am fixing -- generating a before-and-after example now).

Before:

[image]

After:

[image]

--Vic

vwheeler63, 2024-07-24:
2 Questions for Gabor @kisvegabor 

Above the list of `LV_STYLE_PROP_FLAG_...` macros is an over-list comment:  "Flags for style behavior".  This is good for human readability of course in the source code, but because it is not a `struct` or `enum` or variable or `typedef`, Doxygen thinks it is for labeling the first macro in the list, and parses it accordingly.

[image]

There is a possible handling for this (Doxygen has a way to "group" things together under a label), but I need to study it for feasibility (i.e. if `breathe` [the bridge to `sphinx`] will preserve it as a group label) so I have added this to the list of things that might be worth pursuing (the list I mentioned in #4 in my proposal).

However, just below that comment is this promise about LVGL version 9 that LOOKS to me like a TODO that has been fulfilled:

```
The rest of the flags will have _FLAG added to their name in v9.
```

Q1:  It seems like something *this obvious* is something I could go ahead and delete.  But I thought I should confirm with you -- would you like me to edit such things that are really obvious?  Or add it to my list of things that might be addressed in the future?

Q2:  Do you use Trello at all for collaboration on projects?

Kind regards,
Vic

kisvegabor, 2024-07-24:
I'm catching up :slightly_smiling_face: 

> Above the list of LV_STYLE_PROP_FLAG_... macros is an over-list comment: "Flags for style behavior".

I think it should look like this:
```c
/*
 * Flags for style behavior. (Comments here for human in /* ... */)
 */
#define LV_STYLE_PROP_FLAG_NONE                     (0) /**< No special behavior */
#define LV_STYLE_PROP_FLAG_INHERITABLE              (1 << 0)  /**< Inherited*/
#define LV_STYLE_PROP_FLAG_EXT_DRAW_UPDATE          (1 << 1)  /**< Requires ext. draw size update when changed*/
#define LV_STYLE_PROP_FLAG_LAYOUT_UPDATE            (1 << 2)  /**< Requires layout update when changed*/
#define LV_STYLE_PROP_FLAG_PARENT_LAYOUT_UPDATE     (1 << 3)  /**< Requires layout update on parent when changed*/
#define LV_STYLE_PROP_FLAG_LAYER_UPDATE             (1 << 4)  /**< Affects layer handling*/
#define LV_STYLE_PROP_FLAG_TRANSFORM                (1 << 5)  /**< Affects the object's transformation*/
#define LV_STYLE_PROP_FLAG_ALL 
```

> Q1: It seems like something this obvious is something I could go ahead and delete. But I thought I should confirm with you -- would you like me to edit such things that are really obvious? Or add it to my list of things that might be addressed in the future?

Honestly I don't even understand what "The rest of the flags will have _FLAG added to their name in v9" means :smile: Probably it's a leftover from some earlier features.  Please, add/remove/edit obvious things. It'd help a lot. After that we can have a more targeted round for the more complex ones.

> Q2: Do you use Trello at all for collaboration on projects?

We are on Slack using Slack Canvases ans Lists for project management. 

> In the files that got renamed, but the @file <filename> did not -- I am also finding that some of these same files .H files have the "guard macro" #ifndef LV_DRAW_VG_LITE_H and/or the comment after the #endif at the end, do not match the filename. I'm going ahead and correcting these.

Great, please fix these as well! Are these detected by Doxygen?

vwheeler63, 2024-07-24:
Hi @kisvegabor !

> I think it should look like this:

Agreed.  Done!

> Please, add/remove/edit obvious things. It'd help a lot. After that we can have a more targeted round for the more complex ones.

Thank you.  An example:  I found the entire contents of the file `lv_templ.c` in the top of `lv_gridnav.h` causing a `@file`  command error, before the actual contents of `lv_gridnav.h` began.  Clearly this was a copy/paste casualty from some time in the past.  That's been the only unusual one thus far.  Thank you for the encouragement to handle things that are obvious.

> We are on Slack using Slack Canvases ans Lists for project management.

Got it!  I haven't used Slack yet, but it looks amazing!

> Great, please fix these as well! Are these detected by Doxygen?

Done!  These are NOT detected by Doxygen, but what I have done is add this to the list of things that might be worth pursuing in future tasks.  It would be easy to set up a quick automated tool to find all of them.  And it looks like the rule of thumb for the guard-macro names is like I use with my own projects:  filename in all upper case with any '.' replaced with an underscore.  Thus:  `lv_style.h` translates to LV_STYLE_H as the guard macro.  In my own projects, I use these in .H files to prevent .H files from attempting to include .H files that were already included (advised by the excellent book *Large Scale C++ Software Design* by Lakos) -- and this has been instrumental to sorting out where C/C++ compilers get tangled (or bogged down and slowed down) by recursive `#include` directives (typically A.H includes B.H includes C.H includes D.H which includes A.H again....).  So the `#include` directives I use in .H files ALWAYS look like this:

```
#ifndef LV_STYLE_H
    #include "lv_style.h"                       /* For style API and related tools. */
#endif
```

Q3:  Please let me know if any of these types of directives are included in the LVGL EXAMPLES repository (`lv_port_pc_visual_studio`) and I can scan/fix them there too -- again, for a future task.

Kind regards,
Vic

vwheeler63, 2024-07-24:
For all to see:  @kdschlosser @kisvegabor 

Here are some emerging examples of the API documentation before and after:

Before (Doxygen ignores docstrings that don't parse correctly):

[image]

After:

[image]

--Vic

kdschlosser, 2024-07-24:
Groups should be supported by Breathe. so go ahead and give that a try if you want to see if it works. I am still messing about with the ability to monkey patch Breathe to be able to patch it so we can tweak it to provide the output we want and/or expand the functionality of it.

The thing I am trying to figure out is why the API for functions that are not documented are not being populated. If you open up the API section and go to the misc folder and then to lv_style_gen.h there should be a bunch of functions in there. Not sure why it's not populating. There are other functions in other files that do not have documentation that are showing up properly. It's very strange behavior.

Breathe has a large amount of voodoo magic code in it so it's not the easiest thing to follow a data path through.  I can tell it was written but a C programmer and not someone that was really proficient in Python. It was also written some 15 years ago so it was written to work with python 2.x which is a whole different animal than something written for 3.6+..

The more I look at it the more I am thinking that forking the repo and updating the code might be the better thing to do. It sucks that it is no longer being maintained as there really is no replacement that is available. Everything I have come across is written for a specific use case or is overly complicated to make work.

@kisvegabor 

would you entertain the idea of creating a repository for the documentation build system for LVGL? This can be attached to the LVGL repo via a submodule. The ReST files would still be located in the LVGL repo it would only be the build system that is moved to a different repo. The reason why I am suggesting to do this is so that the build system is able to be worked on and not effect the LVGL repo. the LVGL repo would need to have the submodule to the doc build system updated in order to have any changes made in the doc build system used. This would allow for better testing to ensure everything is working properly without breaking anything in LVGL. 

If we decide to add the breathe code to the build system it would actually makes things a bit easier to do and it would also make the code footprint smaller. This would be because I would be able to use the XML parsing that Breathe does to handle being able to access the information externally (gen_json). I wouldn't need to have code written to generate ReST files for the API section. This would all be handled using the same code. all that we would need feed in would be a single path to a single header file... It eliminates code that for all intents and purposes does basically the same thing. Breathe has a more complete parsing of the XML files that doxygen generates. It is able to handle all the CPP code as well.

[125 more replies]

closed 2024-10-03 by lvgl-bot