lvgl/lvgl #6489: ESP LCD Panel API Driver support

codewitch-honey-crisis, 2024-07-08:
### Problem to solve

Since the LVGL project has now taken it upon itself to provide some last mile driver/display integration support, the lack of ESP LCD Panel API support is a real missed opportunity to cover like 80% of the displays and interconnects for an ESP32 that already exist with a DMA capable high performance subsystem exposed by Espressif's chips. This is particularly desirable since the ESP32 line has enumerable kits with integrated screens available for the consumer IoT market.

### Success criteria

This will be successful when it can seamlessly integrate Espressif's ESP LCD Panel API into LVGL 9+'s driver API, and thus take advantage of all of the available support for displays in that arena.

### Solution outline

I'm not sure how to implement the driver portion directly from the LVGL standpoint. I need help there.

I did produce this, using 9.1, which integrates the ESP LCD Panel API with LVGL in Arduino and ESP-IDF on over a dozen devices.

[https://github.com/codewitch-honey-crisis/esp_lvgl](https://github.com/codewitch-honey-crisis/esp_lvgl)

Basically I just need to integrate that code, and I'd be willing to do so with a bit of guidance. I don't know where to begin to write a new driver.

### Rabbit holes

On the surface, not many, as Espressif seems to have designed their API with LVGL in mind.

Under the covers, there's the fine print. The ESP LCD Panel API can drive RGB interface screens (20+ pin interconnect), but not without some significant kludging of their API, which seems kinda new at the RGB interface thing and so it's maybe not as mature as it should be, even though it's used routinely by boards with those screens, usually by way of Arduino_GFX or Lovyan_GFX libraries.

What I did, is take all the necessary spaghetti from lovyan without anything i didn't need (like line drawing and stuff), to hack RGB screens, and integrated it into include/lcd_init.h in that project. That file is a big ball of mud, given the code i took it from was muddy to begin with, and is hacky besides, but it works great, even if it does hijack the i8080 api to drive RGB in some cases.

### Testing

We'd test this the same we test any other drivers - by getting a kit, and running LVGL on it, to see if the display works.

### Teaching

This feature would be documented alongside the other drivers, maybe with a section on "integrating for ESP32s" like you have for STM32 already.

### Considerations

n/a

kisvegabor, 2024-07-10:
Hey,

We had [a PR](https://github.com/lvgl/lvgl/pull/4507) for that but it was closed as Espressif already maintains an [LVGL init wrapper](https://github.com/espressif/esp-bsp/tree/master/components/esp_lvgl_port/src/lvgl9).

It should have a better documentation on our side and will work on that after fixing some FreeRTOS ralated issues with ESP team.  

@codewitch-honey-crisis Are there any use cases when ESP's LVGL port cannot be used and we should have our own solution? 

cc @espzav

codewitch-honey-crisis, 2024-07-10:
Without testing the LVGL Init wrapper, I can't say for sure. I'll look into it. If I had to wager right now I'd say it will not work with several RGB interface screens, due to the necessary hacks I've had to make (lifted from LovyanGFX) to get many RGB interface screens to work. I also need to see if they have touch. I'm working on touch wrappers right now..

espzav, 2024-07-10:
Hi @codewitch-honey-crisis, the RGB screens are supported in `esp_lvgl_port` (https://github.com/espressif/esp-bsp/tree/master/components/esp_lvgl_port) component, which should be easier to use (initialization and some other functions for using LVGL). If there is something missing for you, you can add intu issues on GitHub or make PR. 

Second (recommended) solution is to use `ESP BSP` (https://github.com/espressif/esp-bsp/tree/master). We have component for each Espressif's board, which is prepared for display initialization, touch and other peripherals. The API is same for all boards and there is guide, how to make own BSP for another board. 

And yes, we have solution for screen touch too - it is in the same repository.

codewitch-honey-crisis, 2024-07-10:
Have yo uever actually used the RGB interface support in the ESP LCD Panel API? Long story short, it doesn't work. You have to hack the crap out of it for several displays, including in some cases hijacking the i8080 interface and using that.

codewitch-honey-crisis, 2024-07-10:
Yeah, some of them work. It's been a long time since I made my driver header, so I forget which displays it doesn't work with offhand - i think some of the makerfabs ESP Parallel Display boards, like the 4" but that's from my techy memory. I'll need to check to be sure. I'll report back when I find out some more information.

codewitch-honey-crisis, 2024-07-10:
As an aside, the other issue with the ESP support as it sits, is it's ESP-IDF only, while the ESP32 supports ESP-IDF *and* Arduino. I think it's kind of a bummer to lock people into using the ESP-IDF.

espzav, 2024-07-11:
> makerfabs ESP Parallel Display boards

Do you mean Parallel i8080 or RGB? There is big difference. RGB is usually without any other interfaces inside display. Parallel have some chip inside and you should use right driver. We have supported only some of them now. All supported LCD drivers are in this table: https://github.com/espressif/esp-bsp/blob/master/LCD.md

About IDF/Arduino, I am sorry I haven't got enought information about that. I am responsible for LCD/Touch and BSP only for IDF.

codewitch-honey-crisis, 2024-07-11:
I mean RGB interface, like 1 pin for each color bit, and then HSYNC/VSYNC control pins, etc. I only mentioned i8080 because of lovyan GFX using it off book to hack the RGB panel functionality.

codewitch-honey-crisis, 2024-07-11:
@espzav This is an aside, but pertinent to me adding touch support to this stuff. I noticed the ESP Panel API now seems to have a companion touch API, although I didn't see if it was integrated into the latest IDF bits directly, or if it was a component. Do you know anything about this? Is Espressif planning on integrating touch support into the ESP-IDF the way they have LCD panels?

espzav, 2024-07-11:
@codewitch-honey-crisis We have component `esp_lcd_touch` and lot of touch drivers which are using this main component: https://components.espressif.com/components?q=esp_lcd_touch 
There is no plan to add it into IDF, because IDF is too big and the plan is remove features from IDF and make components from them.

codewitch-honey-crisis, 2024-07-11:
Thanks. That's good to know. I think what I'll do is peruse that code, and evaluate whether or not I want to wrap that for the IDF or not. My inclination is not to do it since I already have drivers for most touch devices that are dual Arduino and IDF, but still I'll give it a look. I appreciate it.

codewitch-honey-crisis, 2024-07-11:
Okay, I've investigated a little further and now I know the issue with the makerfabs boards. It's Arduino related. Arduino rides alongside and half on top of the ESP-IDF, but it uses an older version of the ESP-IDF, so if you're still on like 2.0.x of Arduino you're on 4.x of ESP-IDF which means the RGB stuff is incomplete or non-existent. It's for this reason that all those hacks I mentioned exist, and why LovyanGFX hijacks the i8080 facilities. The technique works under Arduino and the ESP-IDF, but appears to no longer be necessary with newer versions.

That said, PlatformIO is still a major player, and its packages are still affected by this.

I can continue this effort, which would bring Arduino for the ESP-IDF panels, but now I'm not sure how much value that would add since Arduino 3.x i think is on the later ESP-IDF API anyway. It's just that PlatformIO hasn't caught up yet, and I'm not sure when that will happen.

@kisvegabor , your thoughts?

kisvegabor, 2024-07-12:
@codewitch-honey-crisis if you can also confirm that the ESP drivers are working well, I think all we need are
- a good documentation
- some example projects

Re-implementing what Espressif already developed wouldn't be that useful.

closed 2024-08-04 by lvgl-bot